WouterJD/FortiusANT #309: Zwift receives no data from FortiusANT on macOS via BLE

albtaeler, 2021-05-19:
Hi all, 
I'm trying to connect my iFlow to Zwift, but so far without success.  FortiusANT is running on a MacBook (macOS 10.15), Zwift is running on a iPad, or iPhone with BLE  connection.
I can connect it successfully, but Zwift shows always 0 for power and HRM.
What can that be? 

Full log:
[FortiusAnt.2021-05-19 09-21-02.log](https://github.com/WouterJD/FortiusANT/files/6506964/FortiusAnt.2021-05-19.09-21-02.log)
Zwift:
[image]

WouterJD, 2021-05-19:
Hi @albtaeler 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

The Zwift image looks good, I have read more issues on Mac not being compatible with BLE but I'm not a specialist in that area AND do not have a Mac.
Could you try with an iPhone?

albtaeler, 2021-05-21:
Hi @WouterJD 
Thanks for your reply.

I got the same result with an iPhone. 
Yeah, maybe it's a BLE compatibility issue. I took a closer look at it. LightBlue shows me two services but it changes constantly between "2 services" and "no services", as if it is running unstable and the service is constantly restarting.
Is the a possibility to get more debug info from the bleno service? 

Are there Mac users who successfully use the BLE interface on a Macbook Pro Retina (mid 2014) with Bluetooth 4.0 or a comparable hardware generation?

I bought two ANT + dongles and will try it out with them tonight.

And yes, I'll leave a comment under issue #14...

WouterJD, 2021-06-07:
Hi @albtaeler; any results?

Since there is no communication here, I assume can be closed.
If not happy to reopen

closed 2021-06-07 by WouterJD