WouterJD/FortiusANT #310: Tacx Desktop App does not connect to FortiusAnt using BLE

HeathB7, 2021-05-19:
Tacx wants to remain closed to BLE. ANT connection is unstable.
If somebody could provide a log with TTS/TacxNeo, maybe you could modify FortiusAnt to properly emulate a Tacx Neo.

Scissors200, 2021-05-26:
The first weeks I didn't have any problems using Tacx Desktop App. Since two weeks I am experiencing problems especially with Training Plans. When I need to push a certain wattage it seems to be losing power. I end up having to pedal myself to death at 120rpm to reach the watt. I was so frustrated that I even looked for a new trainer. But ofcourse for the Tacx App you need a Tacx. And I refuse to give them any more of my money. I don't like Zwift. And I think it's to much money for what they offer.

Anyway, how can I produce the logs?

WouterJD, 2021-06-07:
Hi @Scissors200 Sorry for late response, been busy lately - also outside :-)

Producing the logs is easy: add -d255 to the command-line and logs will be created in the start-up folder.
Add the logs to your response on this issue (using a browser, not replying using email)

WouterJD, 2021-12-03:
Hi all; is this issue still relevant?
Please update me...

closed 2021-12-28 by WouterJD