WouterJD/FortiusANT #311: Run FortiusAnt on a 3.5" TFT LCD screen on the raspberry Pi using raspi2fb

WouterJD, 2021-05-19:
Hi Wouter, I just made some modifications to run FortiusAnt on a 3.5" TFT LCD screen on the raspberry Pi. Works like a charm. 

I didn't have to make any modifications on your software. The keything is to have raspi2fb installed to make sure your X desktop is copied to the framebuffer (fb1) that the LCD is using.

[](url)
[image]

_Originally posted by @GewoonGijs in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-832754050_

WouterJD, 2021-05-19:
_copied from [here](https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-843889079)_

@WouterJD, Thanx.

> Looks really nice @GewoonGijs I assume this is a Raspberry 3 or 4 solution, since wxPython does not run (easily) on the rpi0.

Yes, it is a raspberry 3.

> Using [raspi2fb](raspi2fb), I assume that the target TFT must support the window-size from FortiusAnt, which is 900x600 by default. Correct?

Actually, that ain't necessary. What I use is a 3,5" cheap LCD-TFT screen from AliExpress with a resolution of 480x320. The nice thing about [raspi2fb](https://github.com/AndrewFromMelbourne/raspi2fb) is that it automatically scales your desktop to fit in the target frame buffer. 

So the trick is to configure the RPi Desktop to be 1024 x 768, just a little bigger than the FortiusANT window size. The setup of the driver for the LCD screen is a little tricky. The way it is documented on the site of the supplier is not completely right. 

If more people are interested I can invest some time to make a step-by-step description of this setup.

WouterJD, 2021-05-19:
_copied from [here](https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-843918458)_

> @WouterJD, Thanx.
>
>
> Yes, it is a raspberry 3.
>
>
> Actually, that ain't necessary. What I use is a 3,5" cheap LCD-TFT screen from AliExpress with a resolution of 480x320. The nice thing about [...]

I would be very interested in a step-by-step instruction to use a Raspberry Pi wiith LCD for FortiusANT.

MeanHat, 2021-05-24:
Looks great @GewoonGijs

kwouters1, 2021-08-22:
I've got the same display running at the 480x320 resolution.
But I don't understand how to implement raspi2fb to show a larger resolution.
@GewoonGijs Could you please assist me?

GewoonGijs, 2021-10-02:
Hi, sorry it took some time, but I was on holiday.

First of all, you have to make sure that the 3.5"display driver software is installed the proper way. The cheap 3.5" Displays froam AliExpress and the like, mostly use  the drivers [https://github.com/goodtft/LCD-show.git](url). Th thing is that the scripts form this reposotory tend to make a lot more changes then necessary, as is explained by Avik Das [https://medium.com/@avik.das/setting-up-an-lcd-screen-on-the-raspberry-pi-2018-edition-d0b2792783cd](url).

If you follow Avik's instructions first you will have a clean install of the required software for your display.

After that you have to install raspi2fb from [https://github.com/AndrewFromMelbourne/raspi2fb](url).

And last but not least you have to configure the RaspberryPi to setup the screen with a resolution of 1024 x 768 on the HDMI port.

**Step 1** is to let X11 use the HDMI screen again:
in `/usr/share/X11/xorg.conf.d/99-fbturbo.conf` switch from `/dev/fb1` to `/dev/fb0` 

`Option          "fbdev" "/dev/fb1" # <-- change this to fb0`

**Step 2** is to change the resolution  to 1024 x 768 with the standard RPi tooling.

**Step 3** is to configure raspi2fb to clone fb0 to fb1 (fb0 is HDMI and fb1 is the 3.5" LCD/Touchscreen. To do this it is enough to follow the install instructions on [https://github.com/AndrewFromMelbourne/raspi2fb](url) in the section **Raspbian Jessie**

closed 2023-12-15 by WouterJD