WouterJD/FortiusANT #312: Installation instructions to run FortiusAnt on a Mac running Zwift

davxbeau, 2021-05-21:
From scratch - how do I actually set it up and het it running?

WouterJD, 2021-06-07:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

Hi @davxbeau if installation instructions on github are not enough, there is a great alternative:
order a FA-Link from www.fortiusant.eu, which comes 100% preconfigured.

MeanHat, 2021-06-07:
Hi @davxbeau - I can endorse and recommend **FA-Link** if that helps.

davxbeau, 2021-06-07:
Yes sorry Wouter I’m not techy at all and unfortunately GitHub just looks Like a bunch of folders I don’t know what to do with.
I don’t even know where to find the installation instructions.

I tried for ages to get my Fortius working with the old Tacx software - I couldn’t do it because fo lack of keys and weird errors.
I want to ditch the old PC laptop altogether however and use Zwift or Rouvy or RGT on my modern Mac and if I can do that with FortiusANT then great.
I don’t want to use an emulator though - I know it will crawl and there's no space on my laptop anyway - so I assume you don’t need any emulator on the Mac to run this?
So can I do that?

My other option is ditch the Fortius and buy a new trainer but it seems a waste - the Fortius is I assume still a reasonable quality chunk of metal? It cost me enough at the time.

I have to say though I got it working on a limited version of the old Tacx software (couldn’t access the rest of it due to no codes) and it was impossible to get it going fast - the wheel felt all wrong and bumpy and could go faster than about 10MPH

I’ll try the link as you say.

Thanks
David

WouterJD, 2021-06-07:
Hi @davxbeau 

- Bunch of folders: that is correct and you can copy them to local and then work with it. But don't be intimidated!
- Installation instructions: start at [wiki](https://github.com/WouterJD/FortiusANT/wiki)
- Mac: the software runs natively on mac, so you can ditch windows emulator (from point of view of FortiusAnt)
- Ditch the Fortius: that's exactly what Tacx suggests. And yes it provides a newer trainer and yes that trainer "talks" ANT or BLE. FortiusAnt adds this ability to your trainer.
- Keep the Fortius: of course; the hardware is good.

Succes David!

davxbeau, 2021-06-24:
Hi Wouter

How about a half way house?
I don’t need a raspberry pie  - I have an iMac - it’ll work fine on that or do you think the pie solution is better?
If I had an easy install though for a Mac without having to wade through those folders and files trying to work out what to do then I would pay for that.

I looked at the installation guide - there is one line on Mac installation - and that one line I didn’t understand.
Basically you need to be techy to do this or spend the time working it out.
It would take me many hours to do that.
Why can’t installation be straightforward?

My other problem with the pie solution is its too expensive - its too close to the discounted price I would get from Tacx for a new smart trainer  - about £200 - so for about £80 more I could get a brand new solution without the extra bits and pieces cluttering the place up.

Best
David

WouterJD, 2021-06-24:
Hi @davxbeau 
Did you already introduce yourself?
As you have seen, we always try to help people forward.

The answer on "Why can't installation be straight forward?" is that there are many trainers (n) and many systems (m) causing n*m options. FortiusAnt has grown from "A solution for my own trainer (FortiusAnt/Windows)" into what it is now, free of charge for everybody who likes it.

A pre-packaged Raspberry Pi Zero is exactly what you ask for: power on and go - no issues. Many happy customers.

I think you are right; don't worry about using your current Fortius and buy a new Tacx. Then you have all advantages of new hardware. You can sell your Fortius for additional discount.

I assume we can close this ticket?

davxbeau, 2021-06-24:
Sorry Wouter - yes we emailed before.
I was just analysing not complaining.
I think what you have done is really very good.
Yes you can close this ticket.

Best 
David

closed 2021-06-24 by WouterJD

WouterJD, 2024-01-15:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

There is no reason to install windows, on a mac/linux/raspberry system, the python version is used. 
The manual gives step by step instructions, if you have specific questions please tel us.