WouterJD/FortiusANT #315: Installation on Raspberry Pi not succeeding

kwouters1, 2021-06-04:
After using FortiusANT succesfully this winter using an old laptop I was planning on building something more permanent.
I've seen what it looks like on a 3.5" LCD in issue 311, so that's what i'm trying to achieve.
Also I would like to keep using my ANT-dongle to send it to my Windows laptop with Zwift.
Reading the manual that should be possible.

Before installing raspi2fb I've just tried to install FortiusANT following the instructions in section 3.12.5.
Unfortunately I'm stuck at the file manager.
When I double click the scripts as indicated nothing happens when I select 'open in terminal'.
So the script doesn't get executed.

Am I overlooking something?

WouterJD, 2021-06-07:
Hi @kwouters1 I would say:
- open terminal
- go to raspberry folder
- execute the script by entering name+<ENTER>
- check what happens

kwouters1, 2021-06-07:
I've opened LXTerminal from Raspberry Pi Buster.
Navigated to the raspberry folder within FortiusANT
When I execute the first script by entering ./1_UpgradeSystem.sh nothing happens. I only get a new line in terminal.
I tried changing rights with chmod +x 1_UpgradeSystem.sh and tried to start the script then, but it gives the same result.

WouterJD, 2021-06-07:
@kwouters1 you're drilling down...

First check with ls whether the size is 0
The next step would be to "cat" the file and see whether the shell can read it.

```
pi@rpiErica:~ $ cd ./FortiusANT/
pi@rpiErica:~/FortiusANT $ cd raspberry/
pi@rpiErica:~/FortiusANT/raspberry $ ls 1*
1_UpgradeSystem.sh
pi@rpiErica:~/FortiusANT/raspberry $ ls -lia 1*
259554 -rwxr-xr-x 1 pi pi 359 apr 25 16:17 1_UpgradeSystem.sh
pi@rpiErica:~/FortiusANT/raspberry $ cat 1_UpgradeSystem.sh 
#!/bin/bash

# ----------------------------------------------------------
# Update your system
# ----------------------------------------------------------
sudo apt update
sudo apt full-upgrade

# ----------------------------------------------------- Done

Raspberry='\033[0;35m'
printf "${Raspberry} System is upgraded, press Enter to continue: "
read reply
pi@rpiErica:~/FortiusANT/raspberry $ 
```

should be like above

kwouters1, 2021-06-07:
We seem to have found the issue.
For some reason the filesize off all scriptfiles is 0.
I tried an install on one of my virtual raspberry's which succeeded perfect.
Going to retry the whole process.

kwouters1, 2021-06-08:
Well. I've got Oracle Virtualbox installed on my laptop.
Then it's possible to make a virtual machine, and the Raspberry pi organisation has made an x86 port of Raspian.
See https://www.raspberrypi.org/software/raspberry-pi-desktop/
This I have installed, so I am able to experiment a little without flashing SD-cards all the time.

WouterJD, 2021-06-08:
Fascinating