WouterJD/FortiusANT #316: Is it possible for me to grab what you’ve done here and apply it to my Tacx?

WouterJD, 2021-06-07:
Hi all, 
I am very new to GitHub and was actually just searching for a way to connect my Tacx Genius to Zwift etc when I came across this thread! Looks like you’ve found a workaround to get this happening - my question is, without ANY experience in programming, is it possible for me to grab what you’ve done here and apply it to my Tacx? Would REALLY appreciate any assistance you can give me...oh, and part of the backstory, I bought the trainer off gumtree and when I went to install the Tacx software, it had expired! So do I need to actually install the software before I can use the workaround you have been discussing here?
Thanks in anticipation 
Jonathon

_Originally posted by @jonojpsg in https://github.com/WouterJD/FortiusANT/issues/101#issuecomment-855820540_

WouterJD, 2021-06-07:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
Thanks any way for joining the sponsoring!
- - -

Of course you can take a computer, install FortiusAnt bridging between Zwift (or others) and your Tacx.
You can also ordering a fa-link, being completely pre-installed and ready-to-run.
Visit www.fortiusant.eu for more info.

jonojpsg, 2021-06-07:
Using a Tacx Genius 2021 - have installed FortiusAnt and have it running but haven’t been able to get it to acknowledge the trainer yet...i think it registered the ANT USB dongle I’m using, but will keep trying

WouterJD, 2021-06-07:
@jonojpsg "but haven’t been able to get it to acknowledge yet" ... did you uninstall the Tacx drivers and install libusb as described in the manual?

Sigfreid74, 2021-06-07:
Hi Jonathon, No you don't need to install the TACX software anymore. Just follow the instructions on the Github install, and see if FortiusAnt finds your trainer. 

PS have you got your 2x ANT+ sticks ready? 
Once you get it recognising your Genius, a Zwift connection should be easy.

Sigfreid (also in Melbourne Lockdown)

WouterJD, 2021-06-07:
Great to see that the Melbourners find each other here!

jonojpsg, 2021-06-08:
> Hi Jonathon, No you don't need to install the TACX software anymore. Just follow the instructions on the Github install, and see if FortiusAnt finds your trainer.
>
> PS have you got your 2x ANT+ [...]

Hi Sigfried, why two dongles? I’ve got one...

WouterJD, 2021-06-08:
Note that the Tacx Genius is a proprietary Tacx-ANT trainer.
Since Tacx uses various non-standard protocols, you have to specify what Tacx trainer you have with the "-t Genius" command-line parameter. FortiusAnt uses the ANT-dongle to communicate with the trainer.
FortiusAnt uses the same dongle to transmit data to communicate with Zwift.

Zwift needs the second dongle.

Hope this helps, sorry for overlooking the "Genius" side of the story, which you mentioned in the first post.

jonojpsg, 2021-06-08:
Still trying with no luck yet...

WouterJD, 2021-06-08:
please show console-output

in addition to -tGenius also specify -d255 to produce logfiles and attach (use browser, not email)

jonojpsg, 2021-06-08:
Dumb questions...what do you mean by console output, and in the setup, I’ve specified 255 in the input box for developer arguments, but then how do I get the log files?

WouterJD, 2021-06-09:
Not so dumb, just lack of info.

I assume you use a Windows computer
There is a folder Startup
This folder contains a lot of sample .bat files which can be used to start FortiusAnt.exe

"FortiusAnt (gui, Genius, autostart).bat" would be applicable for you, specifying to expect a Genius trainer.
The file can be edited (using notepad, rightclick Edit) 
Add -d255 to the command-line and execute

In the folder you will get the logfiles

jonojpsg, 2021-06-09:
[FortiusAntGUI.2021-06-10 09-12-11.log](https://github.com/WouterJD/FortiusANT/files/6627393/FortiusAntGUI.2021-06-10.09-12-11.log)
This is what I got...

Sigfreid74, 2021-06-09:
Sometimes my trainer takes a little while to be picked up. Sometimes I have
to wait for a minute... Also try making sure the ant dongle is fairly close
to the trainer, also try USB extension cables and other usb ports. All part
of the TACX experience for many years (not really a FortiusAnt issue).
FortiusAnt is very straightforward compared with getting the Tacx Trainer
Software to work about 10 years ago!

Good luck Jonathon! End of Lockdown tomorrow!

jonojpsg, 2021-06-10:
Ok, so I've finally got somewhere! Found my Genius, but now the GUI keeps flickering while calibrating then says calibration failed :( It goes back to runoff and start though, and is registering speed but way too low on the power reading...6434 Uncalibrated brake error.

When it's calibrating, should the brake unit be moving the wheel? Nothing seems to happen...

[FortiusAntGUI.2021-06-10 11-29-14.log](https://github.com/WouterJD/FortiusANT/files/6627784/FortiusAntGUI.2021-06-10.11-29-14.log)

Sigfreid74, 2021-06-10:
The Calibration should automatically start soon after pairing (can't
remember the command line switch to force it to calibrate - Wouter?), then
in blue writing, the instruction comes "Nudge Wheel forward". This starts
the wheel turning, and if you have nudge it enough, the genius takes over
the calibration routine from there, it will speed up to about 30km/h and
then take a reading as it coasts to a stop. Then it will say "calibration
complete" and a number (like +1 or something). The further away from Zero
the number is, the worse the calibration and might mean you need to
increase the pressure from the wheel to the roller, or reduce it. (Use the
blue knob under the roller to adjust the pressure.

Good luck - keep at it.

WouterJD, 2021-06-10:
HI all Good work!

@Sigfreid74 to my knowledge your description is correct.
Waiting for minutes to be detected is normal and takes longer than usual ANT-devices. (I do not know why)
There is no command to force or disable calibration; it's always done.

@jonojpsg 
You have attached one logfile (FortiusAntGUI*.log); usually the other one FortiusAnt*.log is required to debug the main program.
But it appears you are getting close!

jonojpsg, 2021-06-10:
Thanks Wouter and Sigfreid,
Have calibrated successfully now (though I got a big -51 result...does that
mean I have to tighten up the screw a lot?) - just having trouble getting
it to talk to Rouvy - Rouvy saying that all ANT+ connections are busy but I
have two ANT+ plugs - the original TACx antenna and another one I bought.
Also reading the manual and trying to get my head around how to actually
ride while on the trainer! You’ve said to ride in top gear and adjust the
trainer power output rather than change gears on the bike - is
that correct? ride in 53-11 the whole time? Also, the GUI speed seemed to
get WAY too high to easily - was ramping up to 50+ kph in the first minute!
Is that something to do with not enough resistance as per calibration
reading?

Definitely closer to achieving connection though 😀👍
Thanks!

Sigfreid74, 2021-06-10:
OK: some good progress. Yes definitely way out calibrating the tyre. Try
winding it in and see how it goes. If the number goes more negative - then
wind it out. Needs to be between -9 and +9. (I always aim for 0!) Put the
tyre air pressure up to 90-100.
I don't know about Rouvy. Someone else is going to have to help you there.

I use my old Tacx videos (Spent a fortune on them back in the day) on
Golden Cheetah (Free Trainer Software).
I ride just how I would outside, using the gears etc. The Genius changes
the resistance depending on the road on the video.

Keep going!

WouterJD, 2021-06-10:
Calibration; as @Sigfreid74 says.

ANT-dongle in use: each application needs it's owner dongle and can then communicate with multiple devices.
So take care you do not start more applications than there are dongles.

Garmin Express is a notories dongle-grabber (and you do not see it).
So take care not  to have Golden Cheetah (is reported frequently being used) and Rouvy!

Sigfreid74, 2021-06-10:
Dongle-grabber!

closed 2021-11-22 by WouterJD