WouterJD/FortiusANT #318: Raspberry PI3: Import Error: No module named 'lib_programname'

RoheHero, 2021-07-17:
Tacx T1902, Raspberry PI3, 16GB-SD-Card: FrontiusAnt won't start. At settings.py in line 18 at 
import lib_programname
an error occurs: ImportError: No module named 'lib_programname'

Could you please help me?

Thx Roman

RoheHero, 2021-07-18:
Here the problem as screenshot:
[image]
or as text - sorry, it's the first time i do this!
------------------------------------------------------
Trashcan emptied - keep 1 month
find: ‘/home/pi/.local/share/Trash/’: Datei oder Verzeichnis nicht gefunden
find: ‘/home/pi/.local/share/Trash/’: Datei oder Verzeichnis nicht gefunden
find: ‘/home/pi/.local/share/Trash/’: Datei oder Verzeichnis nicht gefunden
Log-files from previous session are deleted - keep two days
Stop standard Bluetooth Service
Enable Bluetooth for FortiusAnt
Traceback (most recent call last):
  File "../pythoncode/FortiusAnt.py", line 45, in <module>
    import antDongle            as ant
  File "/home/pi/FortiusANT/pythoncode/antDongle.py", line 112, in <module>
    import FortiusAntCommand    as cmd
  File "/home/pi/FortiusANT/pythoncode/FortiusAntCommand.py", line 73, in <module>
    import settings
  File "/home/pi/FortiusANT/pythoncode/settings.py", line 18, in <module>
    import lib_programname
ImportError: No module named 'lib_programname'
Start standard Bluetooth Service
FortiusAnt finished
  Press Enter to continue:  v
--------------------------------------------------

Maybe this error occors, because i didn't have an ANT+ USB-STICK at the raspberry pi3 connected?
But in the user manual is written (under 3.2.4): 

" 4) If the computer is a Raspberry Pi, the BLE-interface is onboard and no dongle is required. "

So I only have the Tacx Trainer connected and USB-Mouse and USB-Keyboard and HDMI Monitor

I also tried:
python -m pip install lib_programname
Error displayed: Could not find any version that satisfies the requirement lib_programname. No matching distribution found.

npm install bleno
npm install express

Installed Python Version: 3.5.3 and 2.7.13

THX Roman

RoheHero, 2021-07-26:
I solved the problem after three days of studying the error messages.
Unfortunately, the installation instructions forgot to mention the version of the Raspberry PI operating system. I still had the STRETCH version on my PI3B and not the BUSTER version.
After reinstalling BUSTER, the installation of FortiusANT also worked without any problems.
I didn't get FortiusANT to run on Windows 10 yet. But that's another story :-)

closed 2021-07-26 by RoheHero

WouterJD, 2021-08-04:
Sorry for not responding earlier (holidays) but I would not have known either.
I have added your STRETCH/BUSTYER remark to the manual (it will be on github on next publication)

WouterJD, 2021-08-04:
Thanks for feedback