WouterJD/FortiusANT #319: Bushido T2790 on Rouvy app?

Yvank7, 2021-08-02:
Hello all, just acquired a 2nd hand TACX Bushido T2790 and wanted to know if I can connect to Rouvy with it.... I successfully connected my old Tacs Flow using the executable from GITHUB (thanks Wouter!) but do not know how to setup this other TCX model... My older one has a cadence sensor that I attach to the bike, but the Bushido has a built in cadence sensor.

Any help getting this Bushido on Rouvy would be greatly appreciated

WouterJD, 2021-08-04:
Hi @Yvank7 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -
The Bushido is an ANT-trainer and the manual describes how to deal with that. Use the -t flag to tell FortiusAnt that a bushido must be connected. See section 5.15 and let me know your experience.

WouterJD, 2021-11-10:
Since there is no communication here, I assume can be closed.
If not happy to reopen

Yvank7, 2021-11-10:
Hello Wouter,
Strange coïncidence…. I as trying to set up my Bushido last night… I did not understand how to modify the prgm so it sees I am running a Bushido…

Yvan Cossette

WouterJD, 2021-11-11:
Hi @Yvank7 

Did you find howto? 
•	Run FortiusAnt with '-a -g -H0 -t Bushido'
Succes!

switchabl, 2021-11-11:
If this is indeed a T2780/T2790, it is the newer Bushido Smart and should have native Rouvy support over both ANT+ and BLE. It is possible it needs a firmware update first though (https://support.garmin.com/en-US/?faq=UKSoZn0kn16TsOSWYyJ7Y5&productID=696562).

Yvank7, 2021-11-11:
> 
> 
> Hi @Yvank7
> 
> Did you find howto? • Run FortiusAnt with '-a -g -H0 -t Bushido' Succes!

Hello Wouter, is this "howto" located in a spefic folder ? And is that correct to say that we need to edit a Python code prgm (.py) to change the parameter to :  '-a -g -H0 -t Bushido' Succes! (do I need to write this exact syntax ?)

switchabl, 2021-11-11:
@Yvank7 Sorry if this was not clear before: with the Bushido Smart T2780/T2790 you should NOT need FortiusANT. It can connect to Rouvy directly. If Rouvy does not see it, you need to install the firmware update using the "Tacx Utility" app (see link above).

Bushido support in FortiusANT is intended for (non-Smart) Bushido T1980. It may or may not work with the T2780 (untested), but there is no good reason to try it.

WouterJD, 2021-11-15:
I've added the note to the manual, will be distributed on next upload.
@Yvank7 are you operational now?

Yvank7, 2021-11-22:
Hello Wouter
Sorry for my late reply… I was very active recently…. But I have tested the Tacx apps and it Works fine… I did not tested it with Rouvy yet… You can close it for now…. If I have other questions CAN I ping you again?
Thanks!!

Yvan Cossette

WouterJD, 2021-11-22:
Thanks. @switchabl is doing a compatibility test, please comment on #336