WouterJD/FortiusANT #321: Convert trainer to do drivetrain testing...

cognoquest, 2021-08-13:
Hello All/Developers,

I might be posting this request at the wrong place, if so I am open to move the discussion elsewhere. My request for help with the use of FortiusANT is for a different purpose than its original intent. My son is looking for a flywheel/brake solution for an efficiency custom rig that he wishes to build to test a new mountain bike drivetrain design. The mtb drivetrain combines the advantages of gearboxes and derailleur drivetrain. The drivetrain is not released in the public domain yet but if you wish to be kept informed on its release you can register on my son’s website here: [https://lalbikes.com/](https://lalbikes.com/). We were given by a friend a Tacx T1901-T1902 and we thought it might be a good fit/solution for the resistance/brake portion of the data acquisition rig. I fired-up FortiusANT using the following commands: FortiusANT.exe –g –m & FortiusANT.exe –g –e with no CTP as stated in the documentation, it seems to work as per the documentation. Note that I do not have the original frame for the above Tacx brake the spinning test are currently limited to me spinning by hand the flywheel. 

I suspect that it is not possible from the command line to startup FortiusANT.exe with a different resistance power of 100 W and also modify the granularity of the increments. Can someone knowledgeable with the code direct me to where in the code the variables: startup resistance 100W and –m flag modifying the power from 10...250 in 10 Watt steps are set?

Thank You in advance. 
Philippe

WouterJD, 2021-08-14:
I will assist next week since I'm off-site nowMet vriendelijke groetWouter Dubbeldam

WouterJD, 2021-08-23:
Hi @cognoquest 

Step 1. Use the python-version (not the .exe) so you can modify the code.
Step 2. modify FortiusAntBody.py
```
L 976 TacxTrainer.SetPower(100) ----- sets the initial power
L1201...L1203 -10, 100, 10 ----- sets the decrement/resets iinitial/increments
```

Succes!

cognoquest, 2021-08-24:
@WouterJD Thank you so much. I have cloned the code to identify and confirm the lines that require the changes. Now I have no reason to not start building the custom rig :) Your help is much appreciated.

All the best,  Philippe

WouterJD, 2021-08-25:
Hi Philippe

Please keep me informed; quite interested in what's going on
If you like 1:1 by email wouter.dubbeldam@xs4ll.nl, or messenger "Wouter Dubbeldam"

Succes!