WouterJD/FortiusANT #322: Use FortiusAnt with a Vortex T1961 with T1962 display

WouterJD, 2021-08-24:
Hi Wouter
I plan to use FortiusAnt with a Vortex T1961 with T1962 display. Are these supported? The trainer was initially not found (Ant+ dongle is identified as Dynastream Innovations dongle). Trainer was only found after I tried Garmin dongle and switched back to the Tacx T2018 dongle. I am still busy setting the rest up ("second PC") part. 

[image]

PS thanks for all the great work you are doing!

Johann Kistner, Stellenbosch, South Africa

_Originally posted by @johnit-learn in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-903167421_

WouterJD, 2021-08-24:
Originally posted by @johnit-learn in [#14 (comment)](https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-904414926)

My setup is the 230V T2160 kit/configuration described here:
https://activesport.co/WebRoot/Store5/Shops/80c85f8f-7a95-4b1c-9c30-e64b314f3f2e/MediaGallery/tacx/T2160_T2165_i-Vortex_Tech_Info_1310.pdf. It was purchased in 2013. The core components are the T1960 230V resistance unit, T1962 i-Vortex computer unit and T2018 Ant II USB stick.

I managed to get the FortiusAnt working in a two computer setup (3.2.4 setup 2) and will provide more details when I have done a workout. The most difficult part was getting the two Ant+ USB adapters working, for which I followed your instructions on both PCs. The Switchable t19xx_usb drivers did the trick. Both computers are running Windows 10. The FortiusAnt PC uses the Tacx Ant II USB stick and the trainer app PC uses a Garmin Ant II stick.

I would prefer to get the system working with Bluetooth, on a Raspberry Pi Zero W, but this item is currently out of stock locally.

For a trainer app I use the Tacx app and Rouvy.

WouterJD, 2021-08-24:
This is the text output:
10:05:41,648: FortiusANT started
10:05:41,649: -a
10:05:41,649: -B False
10:05:41,650: -c False
10:05:41,650: -C 0/0
10:05:41,650: -d 7 (0b111)
10:05:41,650: -D None
10:05:41,651: -g
10:05:41,651: -H 0
10:05:41,651: -p 1.0
10:05:41,651: -O False
10:05:41,651: -R defines Runoff: maxSpeed=40 dip=2 minSpeed=1 targetTime=7 power=100
10:05:41,652: -t Vortex
10:05:41,652: -T [34, 50] x [34, 30, 27, 25, 23, 21, 19, 17, 15, 13, 11] (start=50x19)
10:05:41,652: ------------------
10:05:42,906: FortiusAnt GUI started in child-process
10:05:43,931: GetDongle - Try dongle: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)
10:05:45,725: FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
10:05:45,945: FortiusANT receives data from an ANT+ Heart Rate Monitor (HRM display), any device
10:05:46,206: FortiusANT receives data from an ANT Tacx Vortex (VTX Controller), any device
10:05:46,481: FortiusANT receives data from an ANT Tacx Vortex Headunit (VHU Controller), any device
10:05:46,762: FortiusANT broadcasts data as an ANT+ Bicycle Power Sensor (PWR), id=57596
10:05:46,985: FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595
10:05:47,240: FortiusANT receives commands from an ANT+ Generic Remote Control
10:05:47,536: ---------- ANT-devices are activated ----------

WouterJD, 2021-08-24:
> 10:05:46,206: FortiusANT receives data from an ANT Tacx Vortex (VTX Controller), any device
> 10:05:46,481: FortiusANT receives data from an ANT Tacx Vortex Headunit (VHU Controller), any device

This means that FortiusANT is listening for two Vortex devices; the brake and the headunit.
Now it is interesting to see what will come on the screen.

The very first post shows the screen listening, but no device connected yet,
Please be patient; it can take multiple minutes for pairing to work.

Note I do not have a Vortex, neither your combination of brake and headunit.

If nothing happens at all; create a logfile using the -d255 option and attach those in a next comment

WouterJD, 2021-08-30:
Hi John

You probably responded by email and attached photo's are then not uploaded to github.
The vortex is an ANT+ enabled trainer and will not work with bluetooth; so unless I'm wrongly informed this is a nogo (does Tacx Trainer Software communicate to the vortex through bluetooth?).

"hciconfig failed, press enter to continue:" sometimes appears without clear reason why.
I have the impression the BT interface is not 100% stables, especially when the RPi400 is warm.
Restart and try....

WouterJD, 2021-09-02:
Hi @johnit-learn congratulations!

So I understand that the Pi/4 is connected to the Tacx through ANT.
Where is Rouvy running?

johnit-learn, 2021-09-02:
Hi Wouter
Rouvy is running on my Samsung Galaxy S21 or on my iPad. The iPad is my
preferred option. My Apple TV is too old.
I also had the entire setup running, without Bluetooth, but with FortiusAnt
and Rouvy on an older Windows 10 core i7 laptop.

johnit-learn, 2021-11-03:
Hi Wouter
A big thankyou from my side: I have FortiusAnt working on a Raspberry Pi Zero W and am very happy with the result. I have simply mounted LEDs with built-in resistors and a small switch on a header. My next project might be to move to a Pi Zero 2 W with a small LCD display.
regards
Johann
[image]

WouterJD, 2021-11-03:
Congratulations and happy to see FortiusANT working!

closed 2022-01-05 by WouterJD