WouterJD/FortiusANT #327: Vortex T1961+ T1962 head unit, dongle tacx

Rybsonzst, 2021-10-30:
Hi!
Firstable great job of Wouter, you're amazing.

Now my issue.
Vortex T1961+ display T1962.
I did everything from manual, but unfortunately i can't connect my trainer to app.
Im using tacx dongle T2028, app finding dongle, but then i get a info from attachment.
What i can do more to fix it?
[image]

WouterJD, 2021-10-30:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

Please provide command line you use

Rybsonzst, 2021-10-30:
Im trying to start it on windows with executable file.
I hope this is what You mean.

WouterJD, 2021-10-30:
You can use the executable, but specify -t Vortex, otherwise FortiusAnt does not know what type of trainer to look for.
please refer to manual section 4.2 page 39

WouterJD, 2021-11-10:
Hi @Rybsonzst how do you proceed?

Since there is no communication here, I assume can be closed.
If not happy to reopen

closed 2021-11-10 by WouterJD