WouterJD/FortiusANT #328: FortiusAnt does not start

LandminePete, 2021-11-02:
I realize this may be a big ask, but I'll try anyway.

I'm using a Windows 10 platform with a Fortius MP. TTS4 is installed.
I've installed the libusb driver package and downloaded the Windows executable. I've created the directory structure as explained in the manual.
When I run the executable, a command box opens, something happens and it closes again, really quickly. The FortiusAnt GUI does not appear.
How do I troubleshoot this ?
Thanks,

Pete

WouterJD, 2021-11-02:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -
Most likely you start FortiusAnt by double-clicking. If you use "FortiusAnt (exe).bat" in the StartUp folder, you can see the messages in the command-box before it closes (pause command).

LandminePete, 2021-11-02:
> Welcome to the FortiusANT community
>
> I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used. Please tell me what bundle did you buy, and what [...]

Dag Wouter,

Introduction done.
I'm using a Fortius MP with a T1932 head unit (white/blue). TT4 is installed, and a bunch of films. I'm trying to get it online to try out the many newfangled online training apps. Get me through winter.
There are messages in the command window, but they go by really fast. I'll try executing from a pre-openend command box and see what sticks.

Pete

LandminePete, 2021-11-02:
I’m pretty sure I’m screwing up the dance with the usb  driver and the replacement procedure. FortiusANT -b sees rage Bluetooth adapter, but not the trainer.

WouterJD, 2021-11-03:
You'r on the way; your trainer is USB-connected so -b is not the option. But good to gather experience.
TTS and FortiusAnt together requires some work; refer to the manual searching for TTS. See paragraphs 5.6 and 5.7

LandminePete, 2021-11-03:
How long is the ‘calibrating’ cycle supposed to last?

LandminePete, 2021-11-03:
Unfortunately I have a wife, a life and a business. Enough faffing about, it’s go ing in the bin.

LandminePete, 2021-11-04:
yes. Goodbye;

closed 2021-11-04 by WouterJD