WouterJD/FortiusANT #329: Show ANT+ devices nearby

kwouters1, 2021-11-04:
My question is not completly  FortiusANT related, but has a lot to do with it.
I'm using FortiusANT in the two computer setup (one sending, one receiving).
But my setup has more ANT+ devices (e.g. my bike has its own speed and cadence sensor).
I'm also using my Garmin Edge 530.
What I'm looking for is a piece of software that could show me all available ANT+ devices.
I'm asking this because my Garmin doesn't see every device, and Zwift also misses some.
Does anyone know of a tool for this?

WouterJD, 2021-11-05:
This is not a trivial issue

ANT opens a channel, filtering on specific parameters and can detect a limitted set of devices on such a channel, not all.
It's also possible to open a specific channel, but usually an ANTdongle allows for 8 channels max.

There is standard tooling, such as simulant, which can be found on www.thisisant.com
I have a simple support tool ExplorANT which could be a starter to build what you want to achieve.

Your question is quite open "my garmin does not see every device", being more specific might help.
If you have a heartrate monitor and your garmin does not see it, most likely other tools would not show it either.

kwouters1, 2021-11-08:
Thanks for the response. ExploreANT helps me on my way. 
I've found out my Garmin had some disabled sensors. Thats why it didn't see the speed sensor.
Removing them and searching again made it find the sensor.
Now to see if I can get it to work right. It drops a lot of the time on my Garmin resulting in no speed (or distance) counting.
The battery is fresh, so that wouldn't be the issue. 
I think i will connect it to Zwift or RGT en see how it performs then. 
I'm aware there are some issues with dropping sensors on the Edge 530 and Garmin is trying to solve that.

WouterJD, 2021-11-10:
Since you  can go forward now, I assume can be closed.
If not happy to reopen

closed 2021-11-10 by WouterJD