WouterJD/FortiusANT #330: Tacx genius 2021

josipmikulic168, 2021-11-08:
Hello.I have problem with tacx treiner genius t2021.This treiner uses an old ant tehnology so i cant conectet to any cycling app such as zwift.So i there eny fix solution for this problem.

WouterJD, 2021-11-09:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -
You are at the right place, use FortiusAnt with - t genius option and happily Zwift the rest of the season😊

josipmikulic168, 2021-11-09:
My name is Josip.So i use tacx genius t2021 model.The only app i can connect my treiner is tacx traning app.The only device i can conectet my treiner is samsung galaxi s9 plus.Eny othere device cant see my treiner.I was sending and asking tacx company what is problem and they sad that my treiner uses an old ant+ tehnology that is not longer suportet from zwift and othere cyclin platforms.

WouterJD, 2021-11-09:
FortiusAnt is the solution, tacx does not provide

josipmikulic168, 2021-11-09:
I know that it is a smart trainer because in the tacx application I can ride movies and the trainer simulates all ups and downs

josipmikulic168, 2021-11-09:
So what is solution?😅

WouterJD, 2021-11-09:
Go to wiki, download an read manual
FortiusAnt connects to your genius using the proprietary tacx-protocol and communicates to Zwift using the open Ant-protocol (or Bluetooth).

You need one ANTdongle for FortiusAnt and
One ANTdongle for Zwift.

WouterJD, 2021-11-09:
You can also buy  a FAlink, see www.FAlink.eu

WouterJD, 2021-11-10:
@josipmikulic168 anything I can help with?

closed 2021-11-22 by WouterJD