WouterJD/FortiusANT #332: Problem with running FortiusANT (old Vista 32-bit PC) - message about KERNEL32.dll problem appeared.

KDrB, 2021-11-15:
Hi Wouter
I dug up my old Tacx i-magic (green 32-bit) after several years
I found your great idea to return it to life with FortiusANT software.
First I ran Tacx i-magic with Tacx Fortius USB software downloaded by following  https://lunicus.org/ according to Michael suggestion.
I launched it on my old 32-bit laptop with Vista Home system. It works perfect!
Then I found your idea to run an old Tacx i-magic with Zwift or other by ANT+ dongles.
Tacx i-magic ("host-transmiter") on Vista 32-bit PC with ANT+
and Zwift ("slave-receiver") on Win10 64-bit PC with ANT+.
Zwift machine works well
I wanted to run FortiusANT on this 32-bit Vista PC, but strange info appeared:
(translated) "Procedure GetActiveProcessorCount entry point not found in library KERNEL32.dll".
There is no information in your comprehensive manual what to do in this case?
Please help!
Your fan Chris
/After I sent e-mail I found that you suggest to write to you on GitHub, so it is copy of my mail massage, sorry!/

WouterJD, 2021-11-16:
Hi Krzysztof Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

Vista ... if you like to keep using that PC, try to use FortiusAnt without the userinterface (-g not specified), which will disable multiprocessing. FortiusAnt does not use GetActiveProcessorCount  directly but it might be implicit.

You can also run FortiusdAnt on the same PC as Zwift, Windows 10 will not provide this error.

switchabl, 2021-11-16:
The Windows binary bundles Python 3.8 which requires Windows 7 or higher. Turning off the GUI will not help. There is a decent chance that it will work on Visa if you install it manually (sections 3.4-3.6 in the manual). When it comes to installing Python, use version 3.7.9 (https://www.python.org/ftp/python/3.7.9/python-3.7.9.exe). But, as @WouterJD suggested, installing it on the Windows 10 PC is probably your best bet.

In general, it might make some sense to build the binaries with 3.7 in order to maximize compatibility. This is also the version installed on almost all Raspberry Pi systems more than two weeks old. So we will need to keep support for that for a while still.

Alternatively, a note should probably be added, stating that the Windows binary works on Windows 7 or higher.

WouterJD, 2021-11-16:
Agree, gui-less might work using the python version; then no binary needs to be created

switchabl, 2021-11-16:
@WouterJD Even GUI should work with Python 3.7. Python 3.8 just doesn't run at all on Vista, it does not matter if you use multiprocessing or not.

It is still possible that some dependency does not work with Vista (or will stop working at some point), so there is no guarantee. Vista is supposedly now just 0.1% of Windows users. So compatibility will be low priority for most projects.

Also, Bluetooth support may be tricky to get working on anything older than Windows 8.1.

KDrB, 2021-11-16:
Hi everybody.
My hard-soft status quo:
Tacx i-magic hardware (green for 32-bit):
Head Serial: 02004787 Break: Basic Excel 1901 and steering interface.
An old Fujitsu-Siemens laptop with 32-bit architecture Intel T2130 1,86 GHz 1,87 GHz, 1 GB RAM.
(earlier Vista was installed, but today I've installed new Win 7 Pro 32-bit). 
I supossed Win 7 will suits better then older systems.
As transmitters: USB ANT+ dongle (ANT310 model) 
Above hard+soft as master-host - connected to trainer.
For Zwift (as slave-receiver) I use nowdays Win10  64-bit on PC i5-2500 3,30 GHz, 16GB RAM.
- 
On Vista Tacx Fortius worked well.
Now I changed system to Win 7 Pro 32-bit, reinstalled Tacx Fortius and... 
Tacx Fortius soft doesn't see Tacx hardware! - usblib_19xx drivers are properly installed.
What to do?
FortiusANT (earlier on Vista) stopped on kernel32.dll problem as I mentioned earlier,
but now on Win7 32-bit - FortiusANT stops with message:
"To run program FortiusANT is impossible because on this PC can't find:
api-ms-win-crt-runtime-|1-1-0.dll
Try install program again to fix this problem"
I really not deep in computers :(
What to do? Please help!

switchabl, 2021-11-16:
@KDrB Download and install https://aka.ms/vs/17/release/vc_redist.x86.exe

KDrB, 2021-11-16:
Hi switchabl :)
I've installed what do you suggested.
In fact, previous message did not appeared again, but now...
black window appeares for moment (less than part of 1 second),
appeares text and window dissapeared. Nothing happen later :(
I make a short movie and grabbed one screen with text in this black window (attached).
Any suggestions to make FortiusANT working?
[image]

switchabl, 2021-11-16:
@KDrB This I have not seen before, I will investigate and get back to you. I believe no one ever tested on 32bit before...

Again, just to be clear: you probably do not need the old PC. You can run FortiusANT and Zwift together on the Windows 10 PC. You can put _both_ ANT dongles in the Windows 10 PC. And run FortiusANT first and then Zwift.

WouterJD, 2021-11-16:
@KDrB to start FortiusAnt without "flashing window" do not double click FortiusAnt.py but use a.bat command-file in the startup folder

OR start FortiusAnt from the commandline

KDrB, 2021-11-16:
@switchabl 
So now I am completely mish-mashed ;)
In FortiusANT manual on page 10 we can see (attached) simply and clear configuration I wanted to try.
You wrote some notunderstandable for me suggestions. Please explain:

a) how can I run FortiusANT and Zwift together on Windows 10 PC (64-bit for sure) when everywhere they say that Tacx imagic green (which I mentioned everywhere about my configuration) needs 32-bit system! 

b) FortiusANT (if I understand well - is needed to convert USB signal from Tacx imagic to ANT+ to transmit to another (or the same PC) with ANT+ and Zwift as "receivers".

I have two PC's (one 32-bit and another 64-bit) and I don't need to force to put all hardware and software on one PC - this way is easier to fix any problems.
-
Why all my configuration doesn't work properly if I put all parts together 
clear and precisely according to FortiusANT manual? 
Where I made any mistake(s)?

KDrB, 2021-11-16:
Sorry for big and small fonts - I really don't know why text changes height?

KDrB, 2021-11-16:
Page 10 of FortiusANT manual I've mentioned above:
[image]

switchabl, 2021-11-16:
@KDrB I am sorry I have confused you. I will try to explain. Do not worry, we will make it work.

a) When people say "iMagic green unit needs 32-bit" they talk about old Tacx software. With FortiusANT, the green unit works on a new 64-bit PC. FortiusANT works _better_ on a new 64-bit PC. Probably no one else has tried it on old 32-bit Windows. That is why you have problems.

b) Yes, this is correct. You can transmit to a second PC or it can be the same PC (then you put Tacx USB and 2x ANT USB in the same PC).

If you prefer two PCs, we can try to make it work. But it will be more difficult because of old 32-bit Windows. If you try Section 3.3 Quickstart on Windows 10 64-bits, like most people, it will probably "just work".

KDrB, 2021-11-16:
@switchabl uffff :)
Now I understand much better :) They say "the devil is in the details" ;)
Ad a) OK - now I understand :)
Ad b) OK, I will try to run all training system on one Win 10 PC but... tomorrow ;)
I will inform you here about results.
Generally... thank you very very much for your determination to help me :) Thats really great!!!
I have another question:
I ran succesfully Tacx Fortius with imagic on Vista (32-bit)
but from the moment I installed Win7 (32-bit) on my old 32-bit laptop,
mentioned above configuration stopped working (I mean Tacx Fortius can't see Tacx imagic interface)
despite of this that I've installed everything the same like before on Vista. 
What is the reason?

switchabl, 2021-11-16:
@KDrB It is probably because you have replaced the driver when you installed FortiusANT. FortiusANT uses a different driver than the Tacx software (that is how it works on 64-bit). If you want to use the Tacx software again, you have to remove the FortiusANT driver. If you go to the Control Panel/Uninstall a Program, it should be listed as "libusb-win32 Tacx T19xx" driver package.

switchabl, 2021-11-17:
I have now tried FortiusANT with with Windows 7 32bit and it works fine. I believe the problem is that you have not installed Windows Updates. At least this is needed:
* Service Pack 1 (if not already installed): http://download.windowsupdate.com/msdownload/update/software/svpk/2011/02/windows6.1-kb976932-x86_c3516bc5c9e69fee6d9ac4f981f5b95977a8a2fa.exe
* KB3063858: https://www.microsoft.com/en-US/download/details.aspx?id=47409

Installing full updates is probably best, but it seems it is now a bit complicated: https://forums.techguy.org/threads/updated-steps-for-installing-and-updating-windows-7-sp1.1222250/

WouterJD, 2021-11-17:
> Sorry for big and small fonts - I really don't know why text changes height?

A line with a single - makes github show previous text enlarged. I have replaced with..
I was curious myself😊

KDrB, 2021-11-17:
@switchabl @WouterJD 
Thanks for yours attention :)
I will try do uninstall/reinstall some soft, drivers etc. today according to your's suggestions.
I will inform you about results!
-
Important question about green head LED:
Now I see that in the Tacx head there is no green LED lighting at all now.
I don't know in what situation it will light. I remember it light earlier, but now it does not light.
Is this green LED light depends on software/drivers etc. or hardware brake greem LED is undependent on software? 
Now it never light and I don't know if it is normal in the case of software malfunction (and this is the reason and it will light again after i reinstall Tacx Fortius) 
or it should light all the time after plug main Tacx T1901 brake power supply plug into power AC 230V socket?

switchabl, 2021-11-17:
@KDrB The green light will only light up after FortiusANT is running. This is normal.

It is possible that with the old Fortius software it is different (because of the different driver). But I cannot check this right now.

KDrB, 2021-11-17:
So... I've uninstalled software, drivers and then reinstalled Tacx Fortius /without FortiusANT/ on my 32-bit laptop.
Unfortunately Tacx Fortius can't see Tacx imagic hardware :(
Maybe the reason is that something wrong happend with hardware /I mentioned in previous message that green LED /brake LED/ does not light in the Tacx imagic head.
(Maybe hardware is OK - but why software+hardware doesn't work? 
I can't see reasons, but it is teoretically possible - unknown hardware malfunction.)
What do you think about it?

KDrB, 2021-11-17:
OK @switchabl, now I've read your message - so hardware probably is OK :)

KDrB, 2021-11-17:
1) I have no FortiusANT on laptop, I only need to run "classic" Tacx imagic with Tacx Fortius software again.
But on previous Vista - Tacx Fortius worked well and green LED light too! I remember.

KDrB, 2021-11-17:
2. Now I need to do:
- as I mentioned above - to run "classic" Tacx imagic with Tacx Fortius software on my 32-bit laptop
and independently I will try also "new" configuration, according to your suggestions, I mean:
- on PC with Win10 64-bit - FortiusAnt with two ANT+ dongles
but first I have to be sure that Tacx imagic hardware work well and there is no any malfunction in training hardware.
How can I try if Tacx imagic hardware is OK?

switchabl, 2021-11-17:
@KDrB I do not think there is a problem with your Tacx hardware. It is very unlikely.

I was able to run the Fortius 2.0 software on Vista 32bit. But I had to install a newer Tacx driver (https://www.filemail.com/d/isgqchacdlfaacz). The driver on the CD did not work.

WouterJD, 2021-11-17:
> So... I've uninstalled software, drivers and then reinstalled Tacx Fortius /without FortiusANT/ on my 32-bit laptop.

@KDrB I agree with @switchabl. The manual has some sections on this subject 5.6 and 5.7 (TTS and FortiusAnt). Installation of USB is in section 3.6

Coexistance of "Tacx software" and "FortiusAnt" is difficult because of driver issues; in fact you should understand the subject otherwise the non-connectivity as you experience may result. PS. I have used windows 7/64 bit, but that should not make a difference.
I do not know how the old Tacx-software works, for the major reason that I said good-by to that software because I want to use the moderns CTP's, Tacx does not (want to) support.

You might chose for the following:
- use the OLD PC for Tacx software (USB cable)
- use the NEW PC for FortiusAnt + Zwift (USB cable + 2 dongles)

Switch cable when you switch application,

KDrB, 2021-11-18:
Hi @switchabl @WouterJD 
Thanks for suggestions. Step by step I approch to (maybe) the good finish.
In teh mean time I was so furious that I cant run anything and probably Tacx hardware is out of order, 
that I decided to make as you've explained a little bit earlier and also finally in the previous message.
I realized that it is good idea. 
There is so big problem with Win system versions, drivers etc. that it is Gordian knot. 
I left my Win7 and Tacx Fortius soft and laptop for a moment.

I followed @WouterJD FortiusANT manual's suggestions.
Simply I've run FortiusANT on Win10 64-bit PC with two ANT+ dongles and USB Tacx imagic interface.
And... it launched... very well :)
Conclusion from this step:
- my Tacx imagic hardware (USB interface) is OK and works well (green LED turned ON! "again")
- probably, as you've suggested, problem with laptop and Win7 is in Tacx drivers
- I've run Zwift (on Win 10 64-bit PC with FortiusANT) and everything paired well and works well 
- so this setup is OK :)

I've tried to install Tacx drivers according to @switchabl suggestions, but Win7 had problem with installation NET Framework 4.5 and 4.8 which are neccessary to Tacx driver installation, so I left laptop. To much problem, to much effort...
Zwift works on Win10 64-bit with FortiusANT and Tacx imagic and it is OK.
Only problem which appeared:
Unstable ANT+ transmission. I mean very often Zwift shows that ANT+ devices are unpaired!
What do you think about it?

Are those ANT+ named "ANT310" wrong? I have possibility to return them and to buy ANT+ dongles 
by GARMIN or ELITE. Price are almost the same. Which one will be better (best)?

KDrB, 2021-11-18:
You can see my ANT310 ANT+ dongles. Are OK or not? Price was about 20 Euro.
Price of GARMIN and ELITE    ANT+ dongles are about twice higher (40 Euro).
Should I buy new ANT+ dongles?
GARMIN or ELITE?

[image]

WouterJD, 2021-11-18:
Refer to issue #61; there are ANTdongle that do not work well. But that caused visible errors, no unpairing if I recall well.
The two dongles are best close to each other, no laptop in between. I would give it a try. Do you have two monitors so you can "see" Zwift And FortiusAnt simultaneously?

KDrB, 2021-11-18:
Hi @WouterJD, @switchabl 
I left my laptop 32-bit with Win7 now - as I mentioned earlier - there is no idea to tyre with it. 
Tacx i-magic hardware has no malfunction, so with hope I'm look in to the future :)
I've returned today those two USB ANT+ named ANT310 to Decathlon where I've bought them. They looked suspicious ;)
Instead of them I've ordered two well known brand dongles - USB ANT+ ELITE /looking the same like GARMIN dongles/. I hope it is a good choice. Tomorrow them will come to me and I'll try again FortiusANT + Tack i-magic system on Win10 64-bit PC. I'll try it on my best machine i7-7800X 64GB RAM with NVIDIA GeForce GTX1070.
I hope entire system will works perfect on Zwift :)
I will inform you about progress.

I'm very happy that both of you - @WouterJD and @switchabl are so helpful! 
It is very kind of you. Without you I would do NOTHING :( 
You are GREAT!!!

WouterJD, 2021-11-18:
Succes!

KDrB, 2021-11-19:
Hi @WouterJD 
So... 
Still something new works wrong.
I can't stand it :(
I've tried to run my set on Win10 PC.
I made everything like you've described in manual.
Drivers for Tacx i-magic and USB ANT+ ELITE dongles works well - I means system Win10 see them as "installed devices".
Yellow and green LEDs lights in imagic head unit.
When I've run FortiusANT (after standard messages) 
in black window appeared message "can't see ANT+, retry..." 
or something similiar every 1 second again and again.
Then I plugged out one ANT+ dongle, plug it again... made some restarts and so on...
Now: 
- both ANT+ dongles are in USB sockets close to each other
- after I run FortiusANT appeares black screen as attached
and Fortius Antifier v6.2 window as attached
but when I start pedalling - nothing happends :(
No values changes.

Please help... what happend this time?

[image]

KDrB, 2021-11-19:
Previous...
[image]
Repeated again :(

KDrB, 2021-11-19:
I've bought and would like (just made it) to connect to training system also:
a) WAHOO HRM "dual" - by ANT+ (this WAHOO has both ANT+ and BLE)  
b) ELITE STERZO SMART - steering device - they wrote that only BLE is working nowdays (device hardware has both ANT+ and BLE). I've updated STERZO according to message from Zwift to "Update ver. 21" - it is actual version.
I don't know if those devices causing any problem or not.
I have plugged and installed properly in Win10
two ANT+ dongles and one Bluetooth dongle.
Logically they are independent and should not disturb FortiusANT and Tacx imagic.

KDrB, 2021-11-19:
It looks like everything is OK, but does not work :(
I mean Tacx imagic does not generate any data (during pedalling).
Another screens for help to solve the problem:
[image]
[image]

WouterJD, 2021-11-19:
Hi @KDrB what a struggle to get things working.

It's always good to do things one-at-a-time.
The iMagic with loadable firmware seems to work OK, so that step is done.
You do not use bluetooth, so no need to look at that

The Dynastream dongle worked (one image) and then gives the unexpected answer message.

My advise:
- stop Zwift, FortiusAnt and all other applications
- insert one dongle
- reboot
- start FortiusAnt
- show what happens
That should be good before proceeding

KDrB, 2021-11-19:
Hi @WouterJD 
I left plugged in only USB from Tacx and one dongle ANT+ 
(the second ANT+ and Bluetooth I removed from USB hub).

First time happend this:
[image]

I've restarted with the same hardware setup and happend this:
[image]

WouterJD, 2021-11-19:
Refer to #61
If I recall well, the reconnect only occurred with the CYCPLUS dongles.

The seconds screen looks good. Is that the current situation now? If the reconnect reoccur, please provide logfile and I will check

KDrB, 2021-11-19:
No, no I never used CYCPLUS! 
I returned yesterday ANT310 USB ANT+ dongles which I tried to use the first.
Today I've used new ones: ELITE USB ANT+ dongles (which are se same like GARMIN) and described as working good with FortiusANT.

Both screens was displayed one hour ago. 
I dn't know why after restart and in the case of exactly she same setup displayed another messages (behaviour) is another.

How to achieve "log file"?

KDrB, 2021-11-19:
Now, twice after restart appeared messages: ... try to reconnect after 1 second".
This message appeares every 1 second till I stop FortiusANT running.

KDrB, 2021-11-19:
Additional info: screenshot from ELITE USB ANT+ driver (both dongles are the same)
(now, only one of two is plugged in):

[image]

KDrB, 2021-11-19:
Pictures of my ELITE USB ANT+ dongles:
It is DYNASTREAM manufacturing (as you described in #61 it is OK)  and sold as ELITE or GARMIN.
My reseller sold them as ELITE.

[image]
[image]
[image]

KDrB, 2021-11-19:
Comment from #61 :
===
As a general guide (whether buying new or used):
GOOD:
    ANTUSB-m: Most reliable and full-featured option (but also most expensive). Made by Dynastream and sold under a variety of brand names. They are all identical so get whichever is cheaper. They can be ordered with custom branding so labeling may vary slightly. By far the most common are "Garmin" and neutral "ANT+":
    antusb-m
    
    Known brands: BKOOL, Dynastream, Elite, Garmin, hLine, LifeLine, Rotor, Saris, Quarq, Wahoo, XAND, Zycle

KDrB, 2021-11-19:
Picture from #61 (text description in the message above):
[image]

So in my opinion (based on #61) now I use the best USB ANT+ dongles available.
and described as working with FortiusANT without any problems.

KDrB, 2021-11-19:
I found following log files. Are those files OK for your's investigations?
[FortiusAnt.2021-11-19 23-51-18.log](https://github.com/WouterJD/FortiusANT/files/7573310/FortiusAnt.2021-11-19.23-51-18.log)
[FortiusAntGUI.2021-11-19 23-51-19.log](https://github.com/WouterJD/FortiusANT/files/7573311/FortiusAntGUI.2021-11-19.23-51-19.log)

KDrB, 2021-11-20:
Hi @WouterJD 
I had a dream... ;)
I made the next experiment.
As you know I have some PC machines with different hardware and operating system.
Simply, I've returned with Fortius ANT and Tacx hardware with new ELITE ANT+ dongles to i5/Win10 machine /where an old ANT310 dongles (which I've returned to Decathlon the day before) seemed to work unstable/.

So... with new hardware setup on an old i5/Win10 /plugged in USB hub with: 
2x ELITE ANT+ dongles, Tacx USB interface and USB Bluetooth adapter (for ELITE STERZO steering interface)/ 
works with Fortius ANT and Zwift... PERFECTLY!!! 
Stable and without any problems.

But I'm very disapointed of Zwift graphic movement on this i5 
with graphic card not designed to graphic operations which is installed inside. 

Conclusions:
Entire hardware Tacx/FortiusANT with USB transmission dongles are OK -
 not here we have to find solution of disfunction on i7 machine.
 
I've sent you yesterday (above) some messages with information which I though (maybe) are important.

What software problem for FortiusANT appeared on the i7/Win10 machine, 
which is much faster with very well graphic card there?

KDrB, 2021-11-20:
After described above experiments, I've returned to i7 machine
(which is intended as destination for my Tacx cycle training system).
it is still not working, but maybe situation is a little bit clearer.

Which I observed:
Maybe I'm wrong, but in my opinion FortiusANT has a problem with finding ANT+ dongle for transmitting data (which it can find without any problem on i5 machine).
There is a clear message on the second box FortiusANT GUI:
> No (free) ANT-dongle found < 

Difference is also that FortiusANT working well on i5 machine can generate virtual gearbox - here there is no virtual gearbox displayed (I think - it is consequence of earlier ANT+ finding problem during starting procedure.

How to run properly FortiusANT on i7 machine?
Please read carefully all my mesages above, maybe something is important?
What another information about i7 machine do you need?

KDrB, 2021-11-20:
The latest approach:

[image]

KDrB, 2021-11-20:
**U N B E L I E V A B L E !!!**
-
My TACX i-magic hardware start to work with Fortius ANT on my i7/Win10 machine!!!
What happend?
I found on one of electronic forum quite another suggestion about problems with dongles, maybe stupid, too simple, too obvious... plug in your USB hub into another socket.
I've done it for peace, I didn't expect any changes.
**BUT IT MADE IT WORKS!!!!!!!!!!!!!!!!!!!!!!!
UNBELIEVABLE!!!**
Uffff.....  
Now I have to try if this solution is stable?

@WouterJD 
I have some other questions about FortiusANT software, but I will start another topic for them.
It is about calibration, virtual gearbox, turing off HRM from Tacx etc.
The first I'll try to find answers in your manual :)

Ohhhh so much nervous days gone ;)
So simple solution was needed!
UNBELIEVABLE...

KDrB, 2021-11-20:
THANKS A LOT @WouterJD - your idea is REALLY GREAT!!!
THANKS A LOT @switchabl

WouterJD, 2021-11-20:
Hey @KDrB good work, thanks to your dedication and persistence.
I will add the experience to the manual. Conclusion: USBhubs MAY be too slow and this also sheds a light on the CYCPLUS issue: somewhere the device is too slow

WouterJD, 2021-11-20:
For the "device in use" I have no explanation, it occurs on my system and I see no pattern.
Remove/insert the ANTdongle resolves the issue; no system/application restart required. After reset, dongle is found. 
Suggestions welcome

KDrB, 2021-11-20:
Wooow! I do was so determined! ;)
Everytime when somethin' is not workin as I need I'm so nervous, furious etc. :)))
Simply I am sure that everytime there is solution - problem is to find it! :)
-
My knowledge about computers isn't very deep.
I thought that USB hubs works similiar to net switches and does not add any delays or decreases speed.
I used hub, because I want to need to be as close as possible with Bluetooth and ANT+ to me on trainer, and thought it is a good, elegant solution that every signals come to one point (hub - concentrator). 
In fact maybe better will be to plug in all dongles to other USB sockets in PC?
Everything works now some hours, so I don't want to chane anything now when it is working perfect! :D
But in fact it is worth to add a cue to manual that socket position of dongles, hubs etc. could make problems and in this case is worth to experiment with place where they are plugged :)

WouterJD, 2021-11-20:
> similiar to net switches and does not add any delays or decreases speed.

Well... They do
Every component causes delay (latency) and has a maximum throughput (bandwidth).

WouterJD, 2021-11-20:
> My knowledge about computers isn't very deep.

Well you have done very good, we all went through this kind of experiences.
Ping me on Strava, so I can follow👍🏃🚴

KDrB, 2021-11-21:
Hi @WouterJD 
it is wery kind of you that you've invited me on Strava! 👍 
Maybe it looks strange, but I don't use Strava ;)

At early 2000's, about twenty years ago, when I was very involved in amatour cycling, I used this my Tacx i-magic, followed my every training session with Polar equipment and software, I used Polar power meter in those times etc. 
I took part in amatour competitions on the road and MTB. I was very focused on training process.
Now, I ride for fun and fitness. Loosing weight is still my challange ;) With success as result of consequent bicycle activity!

I like chalenges, but I am not focused on achiving strict sport results.
So, I ride almost everyday, but don't follow and analysing typical cycling parameters for the best training results, don't save my training etc. ;)

I wanted to launch Tacx i-magic and on-line riding for fun, personal and family fitness and mainly for fall-winter activity, when outdoor weather is very often poor, cold, rainy etc. After worktime there is also dark outdoor, so riding bike on the road could be dangerous because of cars traffic.

So I don't use cycling software which gives riding documentation or cycling data analysing ;)
Simply, I don't use Strava.

KDrB, 2021-11-21:
But you can follow me on Zwift under nick: K. Dr_B 
👍

WouterJD, 2021-11-21:
Right you are! For me, the fun of Strava is to see where FortiusAnt ers are zwifting. All measurements and statistics I leave for what it is.
Let's close this issue, happy to see other questions.

closed 2021-11-21 by WouterJD