WouterJD/FortiusANT #333: FortiusANT different behaviour with different CTP training apps.

KDrB, 2021-11-21:
Hi,
I've just launched Tacx i-magic with FortiusANT software working with Zwift on i7/Win10 64-bit PC.
This setup works perfect. Every functions was easy to connect and Zwift see Tacx as controllable elements.
I also needed to check other training apps - to choose the best for my needs,
so I've seen on YT some reviews and I've read some text reviews.

I've started to try install and then to check some of known training applications.
And I've realised that not every applications behave the same as I expect!
Please help to solve that problem. Where is the cue? Maybe there is solution?
I've tested (divided in two groups):

3D ANIMATION CTP APPLICATIONS:
Zwift - https://www.zwift.com/ - can see sensors - sensors controling CTP and CTP controlling resistance - OK
RGT - https://www.rgtcycling.com/ - can see sensors - sensors controling CTP and CTP controlling resistance - OK

REAL VIDEO CTP APPLICATIONS:
Rouvy - https://rouvy.com/ -  can see sensors - sensors controling CTP and CTP controlling resistance - OK
FULGAZ - https://fulgaz.com/ -  can see sensors - BUT sensors NOT controling CTP and CTP NOT controlling resistance!
BigRingVR - https://fulgaz.com/ - can't see sensors and there is not possible to start properly CTP.
BKOOL - https://www.bkool.com/ - can't see sensors and there is not possible to start properly CTP.

Could you explain why setup: Tacx i-magic (old fashioned USB trainer) with FortiusANT behaves so different? 
Is there any solution or it is normal?

WouterJD, 2021-11-21:
> And I've realised that not every applications behave the same as I expect!

AND:

> Could you explain why setup: Tacx i-magic (old fashioned USB trainer) with FortiusANT behaves so different?

Seem two different questions.

To start with the last, I think FortiusAnt behaves equally for all CTP's (even though I think you reformulate the first question).

And then the first question: I don't know why the CTP's behave differently.

To suggest for investigation:
- the Ant-protocol is defined publically, see thisisant.com
- FortiusAnt can log all activitity in a logfile and the source is open

Comparing the two may reveal flaws in the implementation of the protocol, in that case please let me know and I will work together to resolve and improve the code.

KDrB, 2021-11-21:
For this issue documentation, also:
VeloReality - http://veloreality.com/ - can see sensors - sensors controling CTP and CTP controlling resistance - OK

Returning to your answer @WouterJD. I also expect that it is "problem" on the CTP applications side, not FortiusANT :)
I'm sure that mentioned above not working CTP apps with FortiusANT were probably designed 
with some shortcuts which makes problems in some cases.

Simply I thought it is known problem and there is an easy solution or simply answer:
"some applications are not working because they use not standard data exchange procedures"
- or something like this ;)

For me personally working applications are enough :)
I will use Zwift and I will choose one of video based - which one - I'm not sure yet. 

I am not so strong in programing/coding to analyze log files or/and ANT protocol :(
So sorry, I would help for sure if I can, but I have no enough knowledge which will help to solve this problem.

For me personally working applications are enough :)
I will use Zwift and I will choose one of video based - which one - I'm not sure yet. 

I am not so strong in programing/coding to analyze log files or/and ANT protocol :(
So sorry, I would help for sure if I can, but I have no enough knowledge which will help to solve this problem.

switchabl, 2021-11-21:
@KDrB It is definitely helpful to know when Training Apps do not work, even if you don't know enough to solve the problem. I will look into it when I have the time.

@WouterJD Can you keep this open or should I make a new "Compatibility problems" issue to remind me?

KDrB, 2021-11-22:
Great @switchabl !
I will follow new issue :)

I also realised that maybe it could be general feature (appearing everywhere on every setups with particular CTP applications) or particular problem only on my setup (depending rather on particular hardware or software PC configuration) - it is interesting.

Of course the destination is rather objective knowledge about source of that problem which could help to improve FortiusANT or conclusion that it is not necessary in this case, because problem is on particular CTP side.
Not forcing to work FortiusANT in my particular case :) 

Zwift (animation) works OK and VeloReality (video with real routes) looks nice and works also OK.
In fact FULGAZ also could be nice CTP - but does not work on my setup :(

Every application gives a little bit another features needed for another athlete - possibility of choice is a target.
It will be nice if FortiusANT idea will be working on every setup correct and without problems, but in fact it is not necessary for every users because it is not sence to use many CTP application by one athlete.
As I mentioned above - a potential possibility to choice from more of CTP is a destination, the sense.

It will be also interesting - the volume of users distribition of different CTP application using with FortiusANT? :)
I am sure not only Zwift is used ;)

WouterJD, 2021-11-22:
Let's follow #336 and close this issue.
FortiusAnt implements the protocol according the standard and it's up to the CTP's to connect; unless errors are found

closed 2021-11-22 by WouterJD