WouterJD/FortiusANT #334: Right calibration procedure for FE (Tacx I-magic) + FortiusANT with CTE.

KDrB, 2021-11-21:
Hi,
I've just launched Tacx i-magic with FortiusANT software working with Zwift on i7/Win10 64-bit PC.
This setup works perfect. Every functions was easy to connect and Zwift see Tacx as controllable elements.

I have question about right calibration procedure to achive correct resistance results.
Smart trainers works with CTE right - I suppose - not additional callibration is needed.

I've realised that PROBABLY? my setup doesn't reflects real resistance and simulated speed.

There is a part in FortiusANT manual about calibration. 
Is 7-sec stopping from 40 km/h to 0 km/h calibration procedure only and enough?

I suspect different CTE works in another way with the same setap and FortiusANT output values.
Precisely:
I can see that data (power, speed, RPM) on FortiusANT screen are similiar (the same) like in CTE display,
but I feel that CTE works a little bit in another way one to other. 

There is no special calibration procedure in CTEs, so how to be sure, 
that all displayed values are adequate to real?

WouterJD, 2021-11-22:
Just to avoid confusion: what do you mean with CTE? [I assume Cycling Training Equipment -> your iMagic?]

I have a Fortius myself and do not use the runoff-procedure.
Most important is, that the resistance of your bike is equal every trainer, so you have comparable results.
The procedure is based upon what Tacx did in the old days and basically unchanged from the ancestor for FortiusAnt, antifier.

Note that the iMagic and Fortius are from the era before the power crancks, so accurate calibration was not possible.

Therefore if you ask "... right calibration procedure to achive correct resistance results...." how do you define correct, other than the runoff procedure described.

KDrB, 2021-11-22:
@WouterJD 
Sorry, I made mistake in the one letter ;) 
So you misunderstood me. 
I mean instead of CTE should be written CTP /Cycling Training Program = particular application = software only/.

I am new in the field of modern trainers equipment and software.
Twenty years ago it was really great to ride on trainer kind Tacx in the virtual world :) and feel real resistance during uphill!
I remember I had also paralell Polar power meter on my Tacx and results of power metering was quite the same (not precisely) but not more than 5% - 10% difference. 
Of course both metod are estimated - not precisely mesured by dedicated equipment.

I mean my question is that it looks "too easy" that without any callibration (more or not complicated) resistance in wheel is more-less correct. It looks like miracle :)
So I expexted in any place of data chain that we can calibrate this chain in any way.
Could you belive - I put my bike on Tacx I-magic with Fortius ANT, pumped wheel, set the brake and... calibration procedure according to your description was not needed. After I stopped pedaling on 40 km/h wheel stopped after about 7 sek!!!
UNEBLIEVABLE! So I am not belive it ;)

In addition there is no calibration procedure in Zwift,
but some CTP gives some possibilities to tune parameters.

Coffe riding gives much fun and precise callibration isn't needed, but...
I am really suprised and mish-mashed - do all system works well?
Do power values shows the real power? 
Can I took part in fair competition in Zwift with true resistance in my wheel? 
Will be my FTP test reliable?

How to check it without objective calibration?

WouterJD, 2021-11-22:
All very good questions, I cannot answer not having iMagic&power meter.
I believe formulas are not exact, but noone has come up with suggestions how to improve.

Zwift believes the figures as communicated by the FE-C interface, so now it's up to "us" to produce the correct ones. Formulas are in the open source.
Adjustment can be made using the command line parameters

Please do not forget you judge "old" equipment with newer, not in age but especially in development and architecture

As always: improvement suggestions welcome

KDrB, 2021-11-22:
Of course - you are absolutely right Wouter!
I understand everything.

Simply:
1) I need be most reilable I can 

2) I understand - development makes better equipment and nowdays training stations calibrate themselves and apps "belive" that they provide correct data

3) I don't know what is known and what not - I learn about everything in this area now,
so your explanations and knowledge are very important to me, 
because after your answers I better know where is way with possibility to go and where is the wall and it is impossible to go further.

closed 2021-12-03 by WouterJD