WouterJD/FortiusANT #335: FortiusANT - starting with input parameters? How to do it?

KDrB, 2021-11-21:
Hi,
I've just launched Tacx i-magic with FortiusANT software working with Zwift on i7/Win10 64-bit PC.
This setup works perfect. Every functions was easy to connect and Zwift see Tacx as controllable elements.

As you know FortiusANT could be launched with some input parameters.

1) I've tried to run FortiusANT with additional parameters (manual #29 and #30) 
and I can't achieve expected results.
For example: 
I need to launch FortiusANT without HRM monitor sending data to FortiusANT
which is connected via Bluetooth directly to CTE.
During launching FortiusANT.exe -H-1 - screen appeares for a moment and disappeared immediately!

2) How to enter correct value of -H parameter in FortiusANT setup screen?
It is not possible to enter -H-1 value (any value) in the setup box on FortiusANT setup screen.

Please help questions 1) and 2)

KDrB, 2021-11-21:
Hi,
I've just launched Tacx i-magic with FortiusANT software working with Zwift on i7/Win10 64-bit PC.
This setup works perfect. Every functions was easy to connect and Zwift see Tacx as controllable elements.

As you know FortiusANT could be launched with some input parameters.

1) I've tried to run FortiusANT with additional parameters (manual  page 29 and page 30) 
and I can't achieve expected results.
For example: 
I need to launch FortiusANT without HRM monitor sending data to FortiusANT
which is connected via Bluetooth directly to CTE.
During launching FortiusANT.exe -H-1 - screen appeares for a moment and disappeared immediately!

2) How to enter correct value of -H parameter in FortiusANT setup screen?
It is not possible to enter -H-1 value (any value) in the setup box on FortiusANT setup screen.

Please help questions 1) and 2)

WouterJD, 2021-11-22:
> 1. During launching FortiusANT.exe -H-1 - screen appeares for a moment and disappeared immediately!

Start from command-line OR put pause-command in your script and check the messages

WouterJD, 2021-11-22:
> 2\. How to enter correct value of -H parameter in FortiusANT setup screen?

See manual 4.3

KDrB, 2021-11-22:
Hi, @WouterJD 
of course, the first I've tried to enter values according to described in the manual.
But here is problem, I've described above.
FortiusANT setup menu on my PC does not let input any values in the input boxes. It's impossible to type a letters!

It is possible to turn on/off every options, but entering values 
into the boxes (particulary HRM) doesn't work :(

So I wrote my question.

WouterJD, 2021-11-22:
You don't need to enter letters, but values only.
Please explain what you do/type/click exactly - if possible with screen prints and responses.

KDrB, 2021-11-22:
For example in the HRM case input box.
You wrote in the description: 0: any, -1: none
It is impossible to enter "-" sign and then "1" to achieve "-1" value - which I need to enter, 
because I use independent Bluetooth HRM monitor which transmit bpm straight to training app.

KDrB, 2021-11-23:
>>        During launching FortiusANT.exe -H-1 - screen appeares for a moment and disappeared immediately!
> Start from command-line OR put pause-command in your script and check the messages
So I've done it: I start FortiusANT from command line in Windows\Run as on picture:

[image]

And after that (in the opposite to "regular" launching without parameters which works normal = well), 
FortiusANT (console black window with information screen) appeares only for a moment and disappeares and no FortiusANT main screen appeares.

WouterJD, 2021-11-23:
Indeed, negative values are not allowed, so -H-1 can only be specified on the command-line.
That an error, which can be resolved at a later stage

The command-line:
- type cmd in the dialogue as you show and press enter
- then type the command and you will see why it ends immediatly

Alternatively, use the scripts in the startup folder

KDrB, 2021-11-23:
1)
> Indeed, negative values are not allowed, so -H-1 can only be specified on the command-line.
> That an error, which can be resolved at a later stage
So, if negative values are not allowed, why do you wrote in manual that "-1" value entered in this box makes FortiusANT - "HRM not sensitive" - I mean HRM monitor signal is not expected by FortiusANT.

2)
I've used command line as you've suggested and results are:

a) only FortiusANT.exe - WITHOUT any parameters in the line:
[image]
and FortiusANT launched normal with typical window - it's OK.
/I have not connected interface, but it is not important at that moment/

b) FortiusANT.exe WITH parameter "-H-1" in the line:
[image]
and FortiusANT DOES NOT launched normal
I mean it doesn't open grafic window and closed cmd running (here you can see, that cmd procedure was finished and cmd window closed).

I expected that in the second case FortiusANT will launch graphic window too, and inform us, that it doesn't wait for HRM monitor signal /does not use HRM/.

As you can see... -H-1 parameter doesn't work properly.
What do you think about it.
How turn off HRM in FortiusANT?

WouterJD, 2021-11-24:
If you specify NO command-line parameters, a default set is assumed as printed on the console screen.

If you Do specify, you need to provide each parameter you need/want, for example -g for the GUI.

Now you start without GUI and it appears there is no free ANTdongle, so FortiusAnt ends.

So reset dongle and retry please

WouterJD, 2021-11-24:
PS. What is the reason you want to disable HRM?

WouterJD, 2021-11-24:
> Maybe I omitted above rules in manual or if not - it will be worth to write them clear in manual :)

Read the console-output, which is there in case you do not find the manual right away😉

WouterJD, 2021-11-24:
> it will be “elegant” from engineering point of view to deactivate HRM data

Agree

KDrB, 2021-11-24:
👍 :)

WouterJD, 2021-11-29:
I think you're OK now and assume can be closed.
If not happy to reopen

closed 2021-11-29 by WouterJD

KDrB, 2021-11-29:
It's OK @WouterJD 
All setup works perfect!
For 3-4 past days I focused on training apps ;)
Generally, now I have knowledge about input line parameters.
I have to experiment a little bit now.
In the case something will be unclear I'll contact you again :)