WouterJD/FortiusANT #336: Training App compatibility (status and problems)

switchabl, 2021-11-22:
Goal of this issue: test and document compatibilty with CTPs, try and fix any issues (see #333).

| CTP                      | ANT HR | ANT PWR | ANT FE-C data | ANT FE-C control | BLE HR | BLE FTMS | Notes |
|--------------------|----------|------------|------------------|--------------------|----------|------------|--------|
| Zwift (Win)          | +           | +              | +                      | +                         | +           | +              |           |
| Rouvy (Win)        |              |                  |                         |                             |              |                  |           |
| RGT (Win)           |              |                  |                         |                             |              |                  |           |
| FulGaz (Win)       | ?           |  ?               |  ?                      | -                          | ?            | ?               | #333  |
| BigRingVR (Win) | ?           | ?                | -                       | -                          | o           | o               | #333  |
| BKOOL (Win)      | ?           | ?                | -                       | -                          | ?            | ?               | #333  |
| SYSTM (Win)      |              |                  |                         |                             |              |                  |           |
| TrainerRoad (Win) |          |                  |                         |                             |              |                  |           |
| Tacx Training (Win) |         |                  |                         |                             |              |                  |           |
| Kinomap (Android) |        |                  |                         |                             |              |                  |           |
| Kinetic Fit(Android) | o     |  o              | o                      | o                          |              |                  |           |
| GoldenCheetah (Win) |     |                  |                         |                             |              |                  |           |

`+` working
`-` not working
`?` unkown
`o` not supported/not applicable

KDrB, 2021-11-22:
Hi, 
also with FortiusANT setup works well on my PC machine (i7):
1) Tacx Training (Garmin) on Win10 - https://tacx.com/
2) VeloReality on Win10 - https://veloreality.com/
3) Rouvy on Win10 - https://rouvy.com/
4) RGT on Win10 - https://rgtcycling.com/

WouterJD, 2021-11-22:
I've updated TrainerRoad; that's my primary CTP.
Why do you make difference between ANT FE-C data/control; I would consider it one device.

WouterJD, 2021-11-22:
As you see I was extending the table, but perhaps an easier format is:

What CTP's have been proven to work with FortiusAnt ?
| CTP | Windows ANT | Windows BLE | Android BLE | iOS BLE | Notes |
|---|---|---|---|---|---|
| Zwift            | + | + | + | +  | |
| TrainerRoad | + | + | + | + | |
| Rouvy          | + | + | + |  + | |
| BigRingVR |   |   |   |   |
| BKOOL      |   |   |   |   |
| FulGaz       |   |   |   |   |
| GoldenCheetah |   |   |   |   |
| Kinomap |   |   |   |   |
| Kinetic Fit |   |   |   |   |
| RGT             |   |   |   |   |
| SYSTM      |   |   |   |   |
| Tacx Training |   |   |   |   |

The above table shows what CTP's are known to be working with FortiusAnt, sending commands/receiving data through ANT FE-C or Bluetooth FTMS. For a full compatibility list what devices are supported by these CTP's, pleasse refer to their websites.

switchabl, 2021-11-22:
@WouterJD I think the simplified version definitely makes sense as a reference for users. I admit I was primarily thinking of it as a TODO-list for myself (what to test/fix) where I would like to have some more detail. Maybe it makes some sense to keep the detailed list here and the simplified one in the README/Manual/Wiki? Not sure.

Also, I think some of the new entries may not be accurate. Zwift definitely has ANT-support on Android (though there have been problems in the past) and Trainerroad claims ANT-support on both Android and iOS. Of course this may not work on all phones/only with additional hardware.

WouterJD, 2021-11-22:
I understand.
- imho if Zwift, TrainerRoad, Rouvy work, I would wait for a solid problem statement what other CTP does not work and why - - > issue number in notes (before taking action) 
- a reference list is handy for users asking "should it work or am I the first one trying" [and for us having pointer to open issue under notes] 
- ANT (on smartphone) seems to be bypassed due to availability of Bluetooth, some apps release BLE before ANT
- if ANT/Smartphone is considered to be documented, it may be required to add manufacturers, further extending the table

WouterJD, 2021-11-22:
But of course @switchabl if you maintain the detailed list, please feel free to do so.

switchabl, 2021-11-22:
@WouterJD Yes, I have lately decided to take a more proactive approach. I feel that there are a lot of small quirks and incompatibilites everywhere and they lead to unnecessary support cases. The answer is usually "yes, X works, but you have to do Y first" or "with X, yes, unless maybe you have Y". I want to get rid of as many IFs and BUTs as possible. Part of that will be trying to test as many reasonable platforms as possible. If something works with most regular smart trainers, it should just work with FortiusANT.

- Zwift: https://support.zwift.com/en_us/ant-and-ble-overview-B1S1BVgr
   * "limited ANT+ support for Androids"
   * ANT+ supported on iOS with ANT dongle (+ Lightning adapter)
- TrainerRoad: https://support.trainerroad.com/hc/en-us/articles/201618914-ANT-Connectivity:
   * ANT+ supported on Android with built-in (mostly Samsung phones) or USB ANT dongle
   * ANT+ supported on iOS with ANT dongle (+ Lightning adapter)

WouterJD, 2021-11-22:
Cool great work!

KDrB, 2021-11-22:
Hi,
I don't know if incompatibility is connected with specific application or with particular kind of machine (PC, Mac, Android etc.). 
It is possible that the same application works with FortiusANT very well with training application on PC and is not working on Android for example.
You have much, much better knowledge about possible reasons.

It's most easy to understand, that particular training application can't recognise FE-C (ANT+) 
and as a result can't exchange data, can't show variables and can't control resistance, but...
very interesting is the case of FULGAZ on PC, 
when this application recognise FE-C (Tacx I-magic with FortiusANT) but doesn't show any data from it 
and doesn't control resistance in wheel.
I will check it more, then I'll maybe write more about it.

KDrB, 2021-11-22:
I've checked FULGAZ once more. It works :) 
Before, I didn't found one important option: Steady or Reactive - which changes modes.
Steady - we can call "free run" without depending on external control (it was turned ON before),
Reactive - full control from athlete activity (it should be turned ON in this case).
FULGAZ - https://fulgaz.com/

Also checked ERGVideo - it works well too :)
ERGVideo - https://ergvideo.com/

So FortiusANT setup works well on my PC machine (i7)/Win10 with FortiusANT and ANT+ dongles:
    Zwift - https://www.zwift.com/
    Rouvy - https://rouvy.com/
    Tacx Training (Garmin) - https://tacx.com/
    VeloReality - https://veloreality.com/
    RGT - https://rgtcycling.com/
    FULGAZ - https://fulgaz.com/
    ERGVideo - https://ergvideo.com/
    
Does not work (training App does not recognize ANT+):
    BKOOL - https://www.bkool.com/
    BigRingVR - https://www.bigringvr.com/

Of course it could be my fault and maybe it is easy to solve this problem?

WouterJD, 2021-11-23:
I've moved the two tables to the top and added @KDrB findings to the end-user table

jr111, 2021-12-04:
BigRing just has to be started first. Veloreality is iffy once in a while and have to mess around with starting order. Both will say ANT not found or stick occupied. I figure something about which ANT stick they grab first then they want the other one. 

Oh and you can add the Grand Theft Auto bike mod to the list I actually use it almost exclusively now it's way better than Zwift or Onelap. Onelap could be listed as well.

Thanks sincerely for all the time and work put in.

decodeais, 2022-01-24:
Whats the Problem with Golden Cheetah? I found no problems using Golden Cheetah with Bluetooth

WouterJD, 2022-01-24:
> Whats the Problem with Golden Cheetah? I found no problems using Golden Cheetah with Bluetooth

Well, it's as far as we know; I've updated the table.
Thanks for info

To all: please let us know!

decodeais, 2022-01-25:
Golden Cheetah works on Linux too.

decodeais, 2022-01-25:
The old Android app Rouvy Workout is still working perfectly with interal Ant+. If you want to use  the internal ant+ you can search for a *.apk file and install it without google play. The problem is  that you must find the right version which works on your android version.

TacxBiker, 2022-02-04:
Hi,

I can confirm that wahoo system is working as wel on iOS as windows 10 /11

CTP | ANT HR | ANT PWR | ANT FE-C data | ANT FE-C   control | BLE HR | BLE FTMS 
Wahoo   Systm (windows) | + | + | + | + | + | + |  
Wahoo   System (iOS) | o | o | o | o | + | + |

WouterJD, 2022-03-23:
Since there is no communication here, I assume can be closed.
If not happy to reopen

Although a compatibility list is useful, it requires maintenance for which I have no time.
When needed we can reopen

closed 2022-03-23 by WouterJD