WouterJD/FortiusANT #337: No tacx trainer found

wimkoutstaal, 2021-11-24:
Hi Wouter,

I have tried to setup Fortius Antifier. I have an old Fortius and I everytime I get the "No Tacx trainer found". 
The Tacx is connected with an USB-cable. On the same laptop I have the TTS4 software. In that software the fortius is being recognized without any problems. What can I do to the solve this issue?

kind regards, Wim

WouterJD, 2021-11-24:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -
TTS and FortiusAnt need different drivers, running both packages on the same system is a challenge. Please check installation procedure in the manual

wimkoutstaal, 2021-11-24:
We can close the call. I installed it on a seperate laptop, and that worked fine. An issue with conflicting usb drivers, as described in the documentation ;-)

closed 2021-11-24 by wimkoutstaal

WouterJD, 2021-11-24:
👍Thanks for sponsoring