WouterJD/FortiusANT #338: Set permanent PowerMode on Raspberry Pi Zero W.

planetSammy, 2021-11-24:
I have a problem switching from GradeMode to PowerMode after calibrating my Genius Trainer. If I start RGT Cycling on my iPhone, my Raspberry Pi immediately switches to PowerMode, but not on my Windows 10 computer. Is it possible to use the PowerMode permanently after calibration?

I use the following settings: ~/FortiusANT/raspberry/FortiusAntMain.sh -a -b -H-1 -P -t Genius

Actually my workaround is to start first with my iPhone and after calibration I establish a connection to my win10 pc...but that is really smart...

Best regards Sammy

planetSammy, 2021-11-25:
Now I tested a connection to an android device, works fine, only win10 does not switch to powerMode 🤔

WouterJD, 2021-11-25:
The CTP (RGT on your case) can send two types of command:
- power mode, e.g. asking for 123Watt
- grade mode, e.g. asking for 15%

No settings are required, if FortiusAnt does not switch to power mode, no such command is received. 

The -P flag is only useful when pairing with (for example) Zwift and TrainerRoad, to instruct FortiusAnt to listen to the power mode commands from TrainerRoad and ignoring the grade mode commands from Zwift. 

That is not applicable in your situation

WouterJD, 2021-11-25:
If RGT on windows continues not to work, please attach logfile as created with - d127

closed 2021-12-03 by WouterJD