WouterJD/FortiusANT #339: Hello

BastiaanTol, 2021-11-25:
Hello WouterJD,

I,m Bastiaan from Twello in the Netherlands. I bought the Tacx Genius T2021 from your Daughter? this week. Nice to see that you have put in so much effort in reviving old trainers. I cant connect the trainer with the Fotius Ant.. Yet. But i will post an issue when i have puzzeled out the manual. 

Hope it works out!

WouterJD, 2021-11-25:
Hi bastiaan. You can connect. - t Genius; see manual👍

BastiaanTol, 2021-11-27:
Hi Wouter,

Thank for your reaction. I have installed the Fortius Antifier and studied the manual.. I have connection with the ANT+ but the trainer stil is not found. I'm working with one laptop (windows 10).
I was told that the Fortius Ant schould work with one ANT+ dongle. But when i look at the picture in the manual under par 3.2.4 it seems that i need two dongles is that correct? Or is there an other way?

[image]

Looking forwar to your reaction.

Groeten,
Bastiaan

WouterJD, 2021-11-28:
FortiusAnt uses one ANTdongle to communicate to a) a CTP, like Zwift, and b) an Ant-based trainer.
If running on same PC a second dongle is required for Zwift itself. 

> I was told that the Fortius Ant should work with one ANT+ dongle.

The statement may be correct but is incomplete; Zwift needs the other one.

OR you chose to use BLE

BastiaanTol, 2021-11-28:
Ok Thanks!

BastiaanTol, 2021-12-01:
Hi Wouter,
I have bought a second ANT+ and got it all running! 
Thanks again!

WouterJD, 2021-12-01:
Succes, see you on strava!, I assume can be closed.
If not happy to reopen

closed 2021-12-01 by WouterJD