WouterJD/FortiusANT #340: Pairing HRM with FortiusANT via ANT+ dongle - how to do it?

KDrB, 2021-12-01:
Hi
I've got a problem with HRM pairing.
I use WAHOO Tickr 2 /ANT+ and BLE/.
a) HRM perfect pairs via ANT+ dongle with Zwift - it's OK
and (of course not in the same time)
b) HRM perfect pairs via Bluetooth dongle with Zwift too - it's OK,
I wanted to run VeloReality with FortiusANT and HRM.
VeloReality can see FortiusANT (FC-E
but HRM does not pair with VeloReality via its ANT+ dongle :(
(Bluetooth option is not available).
So I've tried to pair HRM via FortiusANT.
The FortiusANT ANT+ dongle is plugged in 
and Blootooth dongle is plugged out at this time of course.

But FortiusANT does not see HRM any way :(
HRM does not pair with FortiusANT via ANT+ dongle 
(I've tried all input line command/parameters, settings and without any input parameters too).

Could you help? 
How to run HRM via FortiusANT app to transmit HR to training application 
which does not recognize HRM via its ANT+ dongle?

Chris

WouterJD, 2021-12-02:
FortiusAnt has three modes:
- do not display HR at all
- display HR as received from TACX trainer (USB cable) in this case tacx head unit pairs with old style HRM
- display HR as received from ANT+ HRM

In the second option, FortiusAnt transmits the HR as a ANT+ HRM which can then be received by a CTP

It makes no sense in the third option to transmit the HR as HRM device, because it would duplicate the device *in the air"

If FortiusAnt AND the CTP do not receive the HRM signal, check that the HRM is working, pairing to a watch or so. 

Note: HR=heart rate, HRM=heart rate monitor, the belt on your chest

KDrB, 2021-12-02:
Hi @WouterJD 
thanks for explanation, I understand and agree.
Shortly: Zwift can see both ANT+ and Bluetooth and work in both modes perfect,
so problem is not with HRM strap and it’s data transmission to both ANT+ or Bluetooth dongles.
When I use Zwift I can see my HR from HRM strap without any problem :)
It works perfect!
Problem appeared when I wanted to use HRM strap with VeloReality app.
Because VeloReality does not see HRM via ANT+ (like FortiusANT do),
I wanted to do HR data transmission via FortiusANT as third option you’ve described (which normally is without sense ;) 
But I can’t see my HR from HRM strap on FortiusANT display and then in VeloReality app.
This is my problem:
How to force FortiusANT receiving and displaying HR bpm from my HRM WAHOO ANT+ strap? ;)

WouterJD, 2021-12-02:
This cannot be done; FortiusANT either receives OR transmits HRM data; it cannot read and then transmit.

By the way: if FortiusAnt would transmit the HRM data, why do you expect VeloReality app to see the FortiusAnt ?

KDrB, 2021-12-02:
Ad a) hmmmm :( it’s a pity, now I understand (which was not clearly described) that HR bpm on the FortiusANT display window is only display (for seeing bpm) and this HR bpm value is not transmitted via ANT+ dongle to second ANT+ dongle for receiving it by training app like Zwift. I thought that it was an idea (signal chain) for receiving HR signal from HRM strap :( 
Now I know, that it is impossible. Am I right?

Ad b) simply because VeloReality does not receiving (till now on my setup) HR signal via ANT+ dongle :( so I get an idea in my mind to use another way for HR data: HRM strap -> ANT+ dongle FortiusANT (as input) -> FortiusANT -> ANT+ dongle FortiusANT (as output) -> ANT+ dongle VeloReality (as input) -> VeloReality as display. 
Now I know it is impossible :(

I’ve got a contact with Kostya - engineer and creator of VeloRealoty - how to make VeloReality ANT+ dongle work (receiving) HR bpm data from HRM strap? :)
It is in progress. I’ll inform you about results :)

KDrB, 2021-12-02:
It’s interesting, as I mentioned earlier, on THE SAME setup Zwift receiving HR bpm signal from HRM strap perfectly via Zwift’s ANT+ dongle ;)

WouterJD, 2021-12-02:
> I’ll inform you about results :)

Thanks, I'm curious what they will say

WouterJD, 2021-12-02:
By the way, while investigating #341 I discovered that -H0 does not work :-(
I always started FortiusAnt with default parameters and did not use -H0 untill now

HeathB7, 2021-12-02:
Hi Wouter, I’m receiving emails for every issue. How do I unsubscribe to these? Many thanks and keep up the great work!

WouterJD, 2021-12-02:
> How do I unsubscribe to these?

I'm not a github expert and did not yet unsubscribe; the mail has an unsubscribe link at the bottom. Please experiment with that

KDrB, 2021-12-06:
Hi @WouterJD
I've talked with Kostya from VeloReality about possible sources of non recognition HR data via ANT+ by VeloReality app.

He suggested to try use not - WAHOO Tickr 2 (which I use), but any GARMIN HRM strap.

He explained that in VeloReality was applicated standard ANT+ HR data protocol and it works and was tested with GARMIN for sure. He explained that maybe WAHOO used in Tickr 2 something not standard in HR data transmission and he have to check it with this model of the strap.

He suggested to check all my setup with classic GARMIN HRM strap to be sure, that all setup works well. 
He suspects that most probably source of problem is in VeloReality HR data protocol which was designed some years ago and probably can't recognise nowdays HRM devices. 

So conclusion is - VeloReality was not updated to use nowdays HRM straps usage till now. 
As I mentioned above - Kostya asked me to check setup with GARMIN HRM to be sure that source of the problem is in the place explained above.
-
I'll be in touch with Kostya and with you here :)
---
So, because my knowledge about ANT+, HRM etc. data exchange is very low - I'm looking for solution as "blind man" in every possible places ;) 
Step by step I'm more close to find solution :)
By the way... with Zwift your FortiusANT works perfect all teh time and with VeloReality too - except this HRM :(

KDrB, 2021-12-06:
I'm sorry... I've forgot about the lines which generate bolds and big fonts ;)

WouterJD, 2021-12-06:
If you want to learn more, thisisant.com is the place to be.
Also, install the python version of FortiusAnt, you can see exactly how it all works.

When done so, start FortiusAnt in simulationmode; devices are created generating data without need to cycle.

Also, you can use ExplorAnt.py, creating master Or slave devices.

Both programs are available as.exe but then they remain black boxes

closed 2021-12-28 by WouterJD