WouterJD/FortiusANT #343: FortiusANT  window 10 error

Roguish000, 2021-12-05:
trying to start FortiusANT and get   'this app can't run on your PC'
runnning Windows 10 Pro, 20H2

sure would like to get it running.
trying on a Bushido.

thanks

WouterJD, 2021-12-05:
Please provide some more info, not seen before

Roguish000, 2021-12-05:
thanks for the quick reply.
What info?  I will do my best to get it all.....

WouterJD, 2021-12-05:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

Could you please include a screen print of the console (black window) showing the exact messages.
Copy/paste the text would even be handier)

WouterJD, 2021-12-05:
Google will send you [to](https://helpdeskgeek.com/windows-10/how-to-fix-this-app-cant-run-on-your-pc-in-windows-10/) and most likely the first option is what is hapening here.

I assume you run the windows executable...
It seems that you do not have sufficient rights to run the executable. The executable contains an embedded python runtime to execute the code of FortiusAnt and it seems this cannot be installed.
Either start the executable as administrator, give your account administrator rights or start as administrator

Roguish000, 2021-12-05:
Thanks for the response...
I am in the San Francisco Bay Area.
I purchase the Bushido approximately 6 years ago,  Did not spare a dime, so got TTS 4 premium, and even a BlackTrack (the steering platform.)  And a new (at that time) Asus Q550L laptop, with dedicated graphics card.  (I also used the laptop for engineering design with Solidworks.)  TTS4 is up to the absolute latest rev.  Even paid for the Google Earth support.....

I have complete admin on the laptop, and do a fair bit of computer using, but not programming.
First option I tried was to 'run as administrator' as I've been down that hole many times.

I just went through the link you gave above for HelpDeskGeek.  The only thing I have not done is go into developer mode, otherwise I should have privileges. Checked all that for the .exe file.....

Not sure what other info I can provide.

Does it need to run in a compatibility mode?  Windows 7 perhaps?  Specific screen size and resolution?

posted the error message here.  hope it comes through.

Thanks again.  Hope we can get it all going........

[image]

WouterJD, 2021-12-06:
Good to read it starts up and read where you're from! 

FortiusAnt should work well with bushido through ANT, where USB-connected trainers face some USB-driver issues.

WouterJD, 2021-12-06:
> I apologize for my mistake. No excuses.

Just to be sure... Does it run or not, if it does no further compatibility checks need to be done

closed 2021-12-06 by WouterJD