WouterJD/FortiusANT #344: Program FortiusANT is removed automatically

BastiaanTol, 2021-12-14:
Hello Wouter,

I have used the program for almost a month now and i am pleased that is works so well! But last week when in tried to activate the program it just seemed to be disappeard...There was no longer a file in the folder..  I have now tried to download the windows executable from your list again. That seems to work but  when the download is ready the file just dissapears again automatically before in can even open it. 

Does this sound familiair to you? Or could it be that an update from a antivirus program or microsoft defender causes this problem? 
Maybe you have heard of a simmulair issue already, and have a sollution?

Looking forward to your answer.. and to ride again :).

Kind regards,
Bastiaan

WouterJD, 2021-12-14:
Yes I heard this before. It's a false positive on the exe.
The exe unpacks and installs itself, which could be considered virus-behaviour. I would suggest you exclude the file or folder from scanning according your antivirus options.

WouterJD, 2021-12-14:
Alternative is to use the python-version.
OR (i did not try that) to save the resulting folder for future usage. I could instruct how if that appeals to you.

switchabl, 2021-12-15:
Easiest way to try that is downloading one of my dev builds (https://github.com/switchabl/FortiusANT/suites/4512675157/artifacts/120589869). They don't use the self-extract option, they just contain the folder itself (and a Node distribution for BLE but you can just ignore that).

It may or may not help. Unfortunately, it is often the PyInstaller bootloader itself that causes the false-positives. I have come to the conclusion that you can not win. If you want exe files from Python, you will always have problems with antivirus software. You can only report the false-positives and hope that they fix it.

BastiaanTol, 2021-12-16:
Hello Wouter,
Thank for your answers!

My laptop's virus scanner (Sophos) is indeed the problem. The program
removes the .exe without a warning. Because of safety procedures off my
employer i do not have permission to exclude folders or programs from the
antivrus software :(.. My officemanager is now trying to exclude de
program. If that doesnt work i'll try downloading one of your dev builds!

I will keep you informed!

WouterJD, 2021-12-28:
Hi @BastiaanTol how are things going?

BastiaanTol, 2021-12-28:
Hello Wouter,

The .exe file is working again, because it is excluded from scanning by
Sophos.. but that has to be done several times - after every update from
Sophos the .exe file wil be removed again.
That is why i have installed one of your dev build as you suggested. Thats
works fine!

Thanks again!

WouterJD, 2021-12-28:
@BastiaanTol good to hear it's working

> Easiest way to try that is downloading one of my dev builds (https://github.com/switchabl/FortiusANT/suites/4512675157/artifacts/120589869). They don't use the self-extract option, they just contain the folder itself (and a Node distribution for BLE but you can just ignore that).
> 

@switchabl do you have suggestions / instructions how to integrate this in the FortiusAnt distribution? Did you reverse engineer the pyinstaller-created .exe or did you use a standard approach. I did not find instructions in the python docs, suggestions welcome.

closed 2023-12-15 by WouterJD