WouterJD/FortiusANT #345: Pedal Stroke Analysis -display issue with V6.2

jujroy, 2021-12-20:
Hello Wouter,
as you have noticed, I am back inside on my old Fortius.  I downloaded version 6.2.  I noticed that the Pedal Stroke Analysis has a tendacy to bunch up in the first quadrant and become circular again for a few crank rotations. I find this a bit annoying.  The pedal resistance and feeling is normal.  I tried version 5.2 and the  pattern is very circular.  I have checked all my connections but no luck.  I attached a picture for you to see.  Any ideas what might be the problem,.  Many thanks.

Jean

[image]

WouterJD, 2021-12-20:
Hi @jujroy I understand that you can test with two versions. If that's correct, please provide logging of both versions (-d127 flag).
I am curious myself what could be the reason. PSA works well on my fortius, please also provide type of head unit you have

jujroy, 2021-12-22:
[V52FortiusAntGUI.2021-12-21 10-32-39.log](https://github.com/WouterJD/FortiusANT/files/7762994/V52FortiusAntGUI.2021-12-21.10-32-39.log)
[V62FortiusAntGUI.2021-12-21 10-24-30.log](https://github.com/WouterJD/FortiusANT/files/7762995/V62FortiusAntGUI.2021-12-21.10-24-30.log)
Hello Wouter, attached are the files.  I hope that you can make some sense out of them.  I am running Windows 10, Tacx Fortius and HU T-1932.  Do not loose too much time on this as my setup works fine other than the eratic PSA display as per the picture above.  Cheers.  Happy holidays to you

WouterJD, 2021-12-22:
There should also be .JSON files with same name, please attach those as well. Thanks

jujroy, 2021-12-22:
It looks like *.json are not taken by the system.  Hope you can view them here attached Sent from Mail for Windows From: Wouter DubbeldamSent: 22 décembre 2021 14:48To: WouterJD/FortiusANTCc: Jean Roy; State changeSubject: Re: [WouterJD/FortiusANT] Pedal Stroke Analysis -display issue with V6.2 (Issue #345) There should also be .JSON files with same name, please attach those as well. Thanks

jujroy, 2021-12-22:
[V52FortiusAnt.2021-12-22 15-30-42.log](https://github.com/WouterJD/FortiusANT/files/7765163/V52FortiusAnt.2021-12-22.15-30-42.log)
[V52FortiusAntGUI.2021-12-22 15-30-48.log](https://github.com/WouterJD/FortiusANT/files/7765165/V52FortiusAntGUI.2021-12-22.15-30-48.log)
[V62FortiusAnt.2021-12-22 15-24-54.log](https://github.com/WouterJD/FortiusANT/files/7765166/V62FortiusAnt.2021-12-22.15-24-54.log)
[V62FortiusAntGUI.2021-12-22 15-25-00.log](https://github.com/WouterJD/FortiusANT/files/7765168/V62FortiusAntGUI.2021-12-22.15-25-00.log)
.json files wont go thru.  Sent seperately.  Cheers

WouterJD, 2021-12-22:
Did you intend to close?

jujroy, 2021-12-22:
Indeed, I am not very good using this system. My intent was not to close this issue.  Hoping that you now have the 6 files

Télécharger Outlook pour Android<https://aka.ms/ghei36>

Reopened #345<https://github.com/WouterJD/FortiusANT/issues/345>.

WouterJD, 2021-12-22:
Please send the JSON files to wouter.dubbeldam@xs4all.nl

jujroy, 2021-12-26:
Over the Holidays, I had to move my setup.  When I reconnected everything this morning, running V6.2, the PSA worked normally.  I suspect that the RJ45 connector for the motorbrake might have been loose.  Bizzare it was working properly with V5.2 but not with V6.2.  Nonetheless,  I am a happy camper.  Thanks @WouterJD.

WouterJD, 2021-12-26:
I would have started with looking at the modified recovery-procedure. If the USB-connection gives errors, reading is retried which causes delays and might affect the PDA.
When this happens, I would expect errors on the console.
Reading-errors are most likely caused by connection issues.

Older versions did not (always) detect reading errors, keeping the measured values unchanged, causing a perfectly round PDA.

So that explain a some.
I will look into the provided files anyway

WouterJD, 2021-12-28:
Hi, I have checked the files, first conclusion is the following:

### There are no retries on the USB trainer
A few retries like below is normal and no reason to worry; therefore it's only in the logfile and not on the console.
```
---------- V52FORTIUSANT.2021-12-22.15-30-42.LOG
15:30:51,868: Retry because short buffer (len=64) or incorrect header received (expected: 0x21303 received: 0xc03)
15:30:54,493: Retry because short buffer (len=24) or incorrect header received (expected: 0x21303 received: -0x1)
15:31:06,127: Retry because short buffer (len=24) or incorrect header received (expected: 0x21303 received: -0x1)
15:31:10,492: Retry because short buffer (len=24) or incorrect header received (expected: 0x21303 received: -0x1)

---------- V62FORTIUSANT.2021-12-22.15-24-54.LOG
15:25:03,784: Retry because short buffer (len=64) or incorrect header received (expected: 0x21303 received: 0xc03)
15:25:06,628: Retry because short buffer (len=24) or incorrect header received (expected: 0x21303 received: -0x1)
```

WouterJD, 2021-12-28:
Now I look into the V62FortiusAntGUI.2021-12-22.15-25-00.log:

### This looks like normal, because there are sufficient samples in one cycle to mark power on a circle:
```
15:26:50,053: PedalStrokeAnalysis - Cadence= 83 info= 23 StartTime=1640204809.335085
15:26:50,053: Angle=  0 power=  101 Time=1640204809.335085
15:26:50,053: Angle= 15 power=  106 Time=1640204809.3663375
15:26:50,053: Angle= 31 power=  109 Time=1640204809.3975692
15:26:50,053: Angle= 46 power=  111 Time=1640204809.4288192
15:26:50,053: Angle= 62 power=  111 Time=1640204809.4600737
15:26:50,053: Angle= 77 power=  107 Time=1640204809.4913192
15:26:50,053: Angle= 93 power=  104 Time=1640204809.5225692
15:26:50,053: Angle=108 power=  100 Time=1640204809.5538206
15:26:50,053: Angle=124 power=   95 Time=1640204809.5850718
15:26:50,053: Angle=140 power=   90 Time=1640204809.6163206
15:26:50,053: Angle=155 power=   88 Time=1640204809.6475704
15:26:50,053: Angle=171 power=   88 Time=1640204809.6788275
15:26:50,053: Angle=186 power=   91 Time=1640204809.7100718
15:26:50,053: Angle=202 power=   92 Time=1640204809.7413225
15:26:50,053: Angle=217 power=   94 Time=1640204809.7725997
15:26:50,053: Angle=233 power=   95 Time=1640204809.803969
15:26:50,053: Angle=249 power=   95 Time=1640204809.835097
15:26:50,053: Angle=264 power=   95 Time=1640204809.8663476
15:26:50,053: Angle=280 power=   94 Time=1640204809.8977194
15:26:50,053: Angle=295 power=   94 Time=1640204809.928841
15:26:50,053: Angle=311 power=   93 Time=1640204809.9602232
15:26:50,053: Angle=326 power=   92 Time=1640204809.9913418
15:26:50,053: Angle=342 power=   91 Time=1640204810.0227199
```

### But the following is not good:
```
15:26:50,241: PedalStrokeAnalysis - Cadence= 83 info=  6 StartTime=1640204810.0538466
15:26:50,241: Angle=  0 power=   94 Time=1640204810.0538466
15:26:50,241: Angle= 15 power=   99 Time=1640204810.0850713
15:26:50,241: Angle= 31 power=  104 Time=1640204810.1163273
15:26:50,241: Angle= 46 power=  107 Time=1640204810.1477227
15:26:50,241: Angle= 62 power=  108 Time=1640204810.1788402
15:26:50,241: Angle= 77 power=  107 Time=1640204810.2100728
```

### Analysis

Since the cadence is 83, the two circles (drawn at each pedal cycle) should be after 60/83=0,72 seconds.
The elapsed time is 0.241-0.053 = 0.188.

If you look at the next two samples, the elapsed time is 1,553-0,803 = 0,75 which is as expected (0.72 see above).

And then at 15:26:53,150 the sample is too early, drawing an incomplete circle.

### Back to the logfile

pe=1 is the measurement that the pedal passes the magnet on the frame and indicates that one full stroke is done.
While cycling this is expected Cadence / 60 seconds and taking the time of the pe=1 records, the cadence can be calculated.
As you see from below it seems that the cadence sensor signals too often (sometimes) causing the described problem.

I do not have an explanation why this happens, nor why the trainer itself does not provide an incorrect cadence as well. Perhaps the internal software refuses to jump to unrealistic values. 

Let's be happy the problem is solved.

See also: [#345 Pedal Stroke Analysis.xlsx](https://github.com/WouterJD/FortiusANT/files/7785538/345.Pedal.Stroke.Analysis.xlsx)

Time of PE=1 | Cadence | Logfile record
-- | -- | --
49,335 | 82 | 15:26:49,335: ReceiveFromTrainer() = hr=0   Buttons=0 Cadence=84 Speed=30.5 TargetRes=1447 CurrentRes=1479   CurrentPower=101, pe=1 hdr=0x21303 Connected to Tacx Trainer T1932
49,335 |   | 15:26:49,366: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=84 Speed=30.6   TargetRes=1457 CurrentRes=1544 CurrentPower=106, pe=1 hdr=0x21303 Connected   to Tacx Trainer T1932
49,335 |   | 15:26:49,397: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=84 Speed=30.7   TargetRes=1453 CurrentRes=1586 CurrentPower=109, pe=1 hdr=0x21303 Connected   to Tacx Trainer T1932
49,335 |   | 15:26:49,428: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=84 Speed=30.7   TargetRes=1448 CurrentRes=1616 CurrentPower=111, pe=0 hdr=0x21303 Connected   to Tacx Trainer T1932
49,335 |   | 15:26:49,460: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=84 Speed=30.7   TargetRes=1449 CurrentRes=1612 CurrentPower=111, pe=0 hdr=0x21303 Connected   to Tacx Trainer T1932
49,335 |   | 15:26:49,491: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=84 Speed=30.7   TargetRes=1448 CurrentRes=1566 CurrentPower=107, pe=0 hdr=0x21303 Connected   to Tacx Trainer T1932
[... 40 lines cut ...]
50,241 |   | 15:26:50,772: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=83 Speed=30.6   TargetRes=1465 CurrentRes=1430 CurrentPower=98, pe=0 hdr=0x21303 Connected to   Tacx Trainer T1932
50,803 | 107 | 15:26:50,803: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=83 Speed=30.6   TargetRes=1452 CurrentRes=1505 CurrentPower=103, pe=1 hdr=0x21303 Connected   to Tacx Trainer T1932
50,803 |   | 15:26:50,835: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=83 Speed=30.7   TargetRes=1453 CurrentRes=1599 CurrentPower=110, pe=1 hdr=0x21303 Connected   to Tacx Trainer T1932
50,803 |   | 15:26:50,866: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=83 Speed=30.6   TargetRes=1450 CurrentRes=1687 CurrentPower=116, pe=1 hdr=0x21303 Connected   to Tacx Trainer T1932
50,803 |   | 15:26:50,897: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=83 Speed=30.8   TargetRes=1452 CurrentRes=1726 CurrentPower=119, pe=0 hdr=0x21303 Connected   to Tacx Trainer T1932
[... cut ...]
50,803 |   | 15:26:51,475: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=83 Speed=30.7   TargetRes=1453 CurrentRes=1480 CurrentPower=102, pe=0 hdr=0x21303 Connected   to Tacx Trainer T1932
50,803 |   | 15:26:51,506: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=83 Speed=31.0   TargetRes=1450 CurrentRes=1475 CurrentPower=102, pe=0 hdr=0x21303 Connected   to Tacx Trainer T1932
51,538 | 82 | 15:26:51,538: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=84 Speed=30.9   TargetRes=1434 CurrentRes=1549 CurrentPower=107, pe=1 hdr=0x21303 Connected   to Tacx Trainer T1932

WouterJD, 2021-12-28:
Cleaned up, and document for future reference.
@jujroy if you've seen it, can be closed.

jujroy, 2021-12-29:
Thanks @WouterJD for your research and detailed analysis.  Cheers

closed 2021-12-29 by jujroy