WouterJD/FortiusANT #350: Error in installing python

denbeiren, 2021-12-29:
I just did a fresh install on a PI 4 / 4G

This is the output of step 4 : 

Collecting tqdm
  Downloading https://www.piwheels.org/simple/tqdm/tqdm-4.62.3-py2.py3-none-any.whl (76 kB)
     |████████████████████████████████| 76 kB 763 kB/s 
Requirement already satisfied: soupsieve>1.2 in /usr/lib/python3/dist-packages (from beautifulsoup4->gdown) (2.2.1)
Collecting PySocks!=1.5.7,>=1.5.6
  Downloading https://www.piwheels.org/simple/pysocks/PySocks-1.7.1-py3-none-any.whl (16 kB)
Installing collected packages: PySocks, tqdm, filelock, gdown
Successfully installed PySocks-1.7.1 filelock-3.4.2 gdown-4.2.0 tqdm-4.62.3
Downloading...
From: https://drive.google.com/uc?id=1Uk1TSc6iLArx14QH8q85c9ytrw0IKLIn
To: /home/pi/Downloads/wxPython-4.1.1-cp37-cp37m-linux_armv7l.whl
100%|█████████████████████████████████████| 120M/120M [00:47<00:00, 2.51MB/s]
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: wxPython-4.1.1-cp37-cp37m-linux_armv7l.whl is not a supported wheel on this platform.
 Pre-built Python is installed, press Enter to continue:

WouterJD, 2021-12-29:
On my PI4 I have used Raspberry Pi OS, port of debian.
I am not a Pi-specialist regarding differences between OS's; perhaps somebody else has a suggestion?

denbeiren, 2021-12-29:
It is a standard Pi install,.. SD card written with the Pi installer from
their website.

WouterJD, 2021-12-29:
"Standard" is not clear, since you may choose from multiple operating systems.
And since there are mutiple suppliers "their website" does not give direction either.

Aprt from that, assembling a Pi with software from github and FortiusAnt is some work; the installation procedures are as clear as possible, but surprises remain there.
Please let me know when you found the issue and/or when you have additional information to assist with.

denbeiren, 2021-12-29:
BIG step forward,..

I am using the Raspberry pi imager 1.6.2, and a Sandisk Extreme Pro
microsdcard (64GB)
Before, i would flash the latest version of PI OS 32-bit "Bullseye" => gave
me issues.

I tried the Legacy "Buster" version,.. and the manual worked like a charm!
No issues what so ever!

WouterJD, 2021-12-29:
Nice step forward. And a new view on "standard" 😝
I assume issue can be closed, if not let me know

closed 2021-12-29 by WouterJD

denbeiren, 2021-12-29:
afaik this issue can be closed,.. Just make sure you use the "legacy"
version