WouterJD/FortiusANT #351: FortiusANT not starting on Raspberry Pi 0 W (error on lib_detect_testenv

edwinvanpoeijer, 2021-12-29:
I have installed FortiusANT on my Pi 0W using the manual. 
First i created an SD with the latest Pi OS and then followed the steps in the manual.

When starting with FortiusAntMain.sh -a -b -D-1 i get the following error:

Trashcan emptied - keep 1 month
find: ‘/home/pi/.local/share/Trash/’: No such file or directory
find: ‘/home/pi/.local/share/Trash/’: No such file or directory
find: ‘/home/pi/.local/share/Trash/’: No such file or directory
Log-files from previous session are deleted - keep two days
Traceback (most recent call last):
  File "/home/pi/FortiusANT/raspberry/../pythoncode/FortiusAnt.py", line 45, in <module>
    import antDongle            as ant
  File "/home/pi/FortiusANT/pythoncode/antDongle.py", line 112, in <module>
    import FortiusAntCommand    as cmd
  File "/home/pi/FortiusANT/pythoncode/FortiusAntCommand.py", line 73, in <module>
    import settings
  File "/home/pi/FortiusANT/pythoncode/settings.py", line 18, in <module>
    import lib_programname
  File "/home/pi/.local/lib/python3.9/site-packages/lib_programname/__init__.py", line 3, in <module>
    from lib_detect_testenv import *
ModuleNotFoundError: No module named 'lib_detect_testenv'
FortiusAnt finished

Any tips on resolving this error?
I would think nothing ANT related is loaded because of the -D-1 parameter?

Python version is 3.9.2

WouterJD, 2021-12-29:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

edwinvanpoeijer, 2021-12-29:
Thanks, i made my introduced myself under issue #14 :)

Short introduction:
My name is Edwin van Poeijer, living and biking in the Netherlands. I am using a Tacx Flow T1901 with T1932 head unit which i picked up for free a few weeks ago :)

edwinvanpoeijer, 2021-12-29:
It seems that lib_detect_testenv wasn't installed.
I did a manual install and FortiusAnt is now starting:
python -m pip install --upgrade lib_detect_testenv

WouterJD, 2021-12-29:
Hi @edwinvanpoeijer great to see you found the issue.
The FortiusAnt dependencies are the direct dependencies and assume that -when those are installed- their dependencies are also resolved.

Succes with further work on your pi0W, which I know can work well, I have assembled multiple Pi's like that.

I assume can be closed; if not happy to reopen

closed 2021-12-29 by WouterJD