WouterJD/FortiusANT #353: Removing the flickering of the FortiusAntGui elements

rocky3355, 2022-01-02:
Hi WouterJD,

I am currently trying out your software and I was really impressed that it found and forwarded my old TACX setup immediately! :)

However, I noticed an issue that I could also reproduce on another Win10 machine: The GUI elements of FortiusAnt are flickering. I had a look at the code and noticed that the GUI had no buffering enabled. I created a small PR that activates the double buffering and fixes as well a small problem that came along with it (details in the PRs description).

Would you have a look at the PR and give me some feedback? Thank you!

https://github.com/WouterJD/FortiusANT/pull/352

Best regards,
rocky3355

P.S.:  This is my first contribution to a foreign repository ever, so I was not sure if my fork's PR would already draw your attention or I also have to create an issue.

WouterJD, 2022-01-02:
@rocky3355 thank you very much for issue & pr, resolving an issue I knew and could not find a solution for. I will che k asap and let you know

WouterJD, 2022-01-02:
By the way; Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

WouterJD, 2022-01-04:
I've looked at your PR and if course, if one control is on top of another that must be the parent😝
I'll do further testing and integrate asap

WouterJD, 2022-01-04:
Works like a charm

closed 2022-01-04 by WouterJD

rocky3355, 2022-01-04:
> I've looked at your PR and if course, if one control is on top of another that must be the parent😝 I'll do further testing and integrate asap

Hmm I thought that drawing on top of each other should still work, like it did before enabling the buffering. But yes, declaring a parent/child relation is the better solution :)