WouterJD/FortiusANT #355: Dynastream Innovations Inc. ANT USB-m Stick

kbruneel, 2022-01-05:
Hi Wouter,

This is Karel from Belgium.

I just got a wahoo ANT+ dongle. I'm trying to connect my old Bushido trainer, but didn't get that far. When I start FortiusANT as follows:

```python3 FortiusAnt.py -a -g -H0 -t Bushido```

It says "No (free) ANT-dongle found".

When I do lsusb it does detect the Wahoo dongle as "Dynastream Innovations Inc. ANT USB-m Stick".

I'm on a MacBook Pro running Monterey.

Many thank and happy new year,
Karel

WouterJD, 2022-01-05:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

WouterJD, 2022-01-05:
Thanks for the good wishes; same to you and let's go to resolve your issue.

The command-line for FortiusAnt seems OK.
The python environment requires installation of libusb (see manual 3.6.3) is that done?
Further I do not have MacOS myself, so am limitted in providing assistence.

I see you are "Founder or roboton.io and circuits.io (acquired by Autodesk)." [I assume or=of?] so I hope you can join in our community for further enhancements when required.

kbruneel, 2022-01-05:
Hi Wouter

It was installed, but maybe not that good since I upgraded OS X. I reinstalled it and now I get the following message:

GetDongle - ANT dongle in use

I guess now it sees the dongle, but thinks it is in use. Any idea what this could be?

Yes maybe I can help out a bit, but first I need to get it running :)

Thanks,
Karel

kbruneel, 2022-01-05:
I got it working!

Apparently "Garmin Express" grabs the dongle immediately. I found that info here: https://support.trainerroad.com/hc/en-us/articles/201408970-ANT-USB-Stick-Troubleshooting

kbruneel, 2022-01-05:
I do still get:

Tacx Bushido paired: 5817 - Configuring head unit COMMUNICATION ERROR

WouterJD, 2022-01-05:
Sorry, should have said that immediatly.
If libusb would not be there, another error would occur.

On windows this error also occurs -unpredictably- after closing FortiusAnt; the only remedy is to unplug the ANTdongle.

WouterJD, 2022-01-05:
> Tacx Bushido paired: 5817 - Configuring head unit COMMUNICATION ERROR

I do not know, since I do not have a Bushido (and never had one).
5817 is the device number, the interface returns the error [Refer to code](https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L2073)

> But this looks like its not important?

Does it work?

kbruneel, 2022-01-05:
It works!

The error goes away when I start pedaling. I guess the communication error is related to the connection between the brake and the headunit.

WouterJD, 2022-01-06:
If I recall well, FortiusAnt connects to the headunit, which connects to the brake: `FortiusAnt <--> HeadUnit <--> Brake`.
Both `<-->` links are through ANT and it mayt well be that the brake wakes up as soon as you make a move.

I consider solved and can be closed. If not happy to reopen

closed 2022-01-06 by WouterJD