WouterJD/FortiusANT #356: Edge 810 does not show speed

kbruneel, 2022-01-05:
I'm trying to connect FortiusANT to my Garmin Edge 810. This works for cadence and power, but not for speed.

It looks like it wants to display the speed based on GPS, but I'm inside and stationary :) Is there any way to force the Edge to display the speed coming from FortiusANT? Or should I change some command line parameters to connect to an Edge?

WouterJD, 2022-01-06:
I have nu 810 so do not know; google provides links to [Garmin, click here](https://support.garmin.com/en-US/?faq=EnYsdQoo623Z535geTx2e9) some more searching may be usefull.

@bikebeppe64 do you have an 810?

BikeBeppe64, 2022-01-06:
Hi All, I have an Edge 820 paired with an external ant+/Bluetooth sensor positioned on the rear wheel, and no problem to have speed measurement.

WouterJD, 2022-01-06:
FortiusAnt publishes SCS, I will check whether it's detected correctly by a client, like a Garmin watchor Edge...

WouterJD, 2022-01-06:
To verify, start FortiusAnt in simulate mode (`FortiusAnt.py -g -a -A -s`) [see github](https://github.com/WouterJD/FortiusANT/blob/master/StartUp/FortiusAnt%20(gui%2C%20autostart%2C%20simulate).bat)

Then pair Garmin Edge (explore in this case) to the Speed-and-Cadence sensor and voila:

[image]

WouterJD, 2022-01-06:
> It looks like it wants to display the speed based on GPS

@kbruneel Please explain why you think that is the case

kbruneel, 2022-01-06:
@WouterJD this is what I get when I use the simulation:
[image]

WouterJD, 2022-01-06:
Fascinating; the Cadence info comes from the SCS-device, so why would the speed not be displayed?
FortiusAnt transmits speed and cadence, as demonstrated above.

Additional tests:
- Try using another ANT+ device and check what is displayed there
- Try a SCS on the bike and check what the Edge810 does

Not much I can do for you at this moment, I think...

WouterJD, 2022-01-06:
PS. Did you define the wheelsize at the SCS sensor. If the wheelsize is zero, the speed will be zero as well.

https://www8.garmin.com/manuals/webhelp/edge810/NL-NL/GUID-EE37BDD9-FDFF-4EA4-B8C8-8B55EB14491C.html

kbruneel, 2022-01-06:
@WouterJD Yes setting the wheel size did the trick. Thanks again! I would not have thought of that :)

WouterJD, 2022-01-06:
Great, happy I could assist.

Perhaps leave a comment under issue #14 or consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
I assume can be closed, if not I'm happy to reopen

closed 2022-01-06 by WouterJD