WouterJD/FortiusANT #357: Fortiusant with other rpm sensor

TitooNL1981, 2022-01-07:
Last week I broke my t1947 rpm sensor. 
i paired in trainerroad with my Garmin. 
is it possible that fortiusant can read this valua also?

I can’t find a new t1947 rpm sensor. Or a other solution. Maybe i don’t need it?

WouterJD, 2022-01-07:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

WouterJD, 2022-01-07:
FortiusAnt could work with a separate sensor, but Tacx Fortius itself can't. No sensor = no operation.

I never disassembled one, it might just be read relais, closing on magnet approach

TitooNL1981, 2022-01-07:
Hi. I’m Eric from the Netherlands. 

Maybe I’m lucky. I using a T1932 on a Flow T1901.(motor brake unit 0x1902) 

How can Fortiusant read mine Garmin? I got a ant id.

TacxBiker, 2022-01-07:
Hi Eric,

The sensor is just a normal read conract. You can use something like this:
https://www.hobbyelectronica.nl/product/mc-38-magneet-contact/?gclid=CjwKCAiA5t-OBhByEiwAhR-hm84kPUn-LHkQEHcNo6gb3WXTF_jJW2qnsXkarUNE7BP2sg2tne4LexoCOzAQAvD_BwE

WouterJD, 2022-01-08:
> FortiusAnt could work with a separate sensor, but Tacx Fortius itself can't. No sensor = no operation.

I was quite convinced, but never actually experimented here.
The Tacx Fortius resistance is calculated as resistance = f(TargetPower, Cadence) and if Cadence is zero, the result will go up to infinit.

I tried to set a fixed resistance rtn=1000, [here](https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L2929) to see what happens without a cadence sensor. And -as expected- Tacx Fortius gives a certain resistance for some seconds and then goes to a maximum, causing the brake to block.

**Conclusion**
The Tacx Fortius headunit uses the cadence as well and cannot work without it's own sensor.
Adding a Cadence sensor to provide the cadence to FortiusAnt will not help.

closed 2022-01-18 by WouterJD