WouterJD/FortiusANT #358: Connecting T2790 (Bushido for Tablet BLE)

StijnVerm, 2022-01-08:
Hello,

The T2790 is a strange device in the Tacx line-up. It is located somewhere between the T1980 and T2780 regarding connection and functionality. It is a self powered device that has only BLE as communication. It didn't receive the same firmware-updates as the T2780, so it doesn't work out the box with the new trainer software.
Currently I can't find a way to install the T2780 firmware on the device, nor I can FortiusANT get it recognized.
The device is recognized on my computer as a BLE device: "Tacx ANT Brake", while it has no ANT. 

If I can assist helping to support T2790

WouterJD, 2022-01-08:
Hi @StijnVerm 

Interesting subject
- Your T2790 ONLY has BLE-communication
- It is NOT recognized by Zwift
- Your computer says it's a Tacx ANT Brake

If that is actually the case, `-t Bushido` will not work, because that means FortiusAnt is searching for an ANT-device.

If your computer (what OS do you use) detects the Tacx and has no ANT-dongle, the name might be wrong, which would be quite special. 

Please provide some more information
- Do you have a smartphone with BLE
- Try there what you see
- Show some screen prints from Zwift, Rouvy, Trainer Road what they detect

Interesting for further investigation

StijnVerm, 2022-01-08:
I tested the unit using Windows 10, windows 11, both gave me the Tacx ANT Brake. The device is using the Microsoft Bluetooth Legacy Enumerator driver dated 2006.

I digged some more into it, using the Microsoft BlueTooth LE Explorer. Following characteristics:

Service name: GenericAccess
Service UUID: 00001800-0000-8000-00805f9b34fb
Characteristic name: DeviceName  UUId:0x2A00  , Handle 2 ,  Value: Tacx ANT Brake
Characteristic name: Appareance  UUId: 0x2A01  , Handle 4,  Value: 80-04
Characterisic name: PeripheralPreferrredConnectionParameters UUId: 0x2A4 , Handle 6,  Value: 0A-00-50-00-00-00-90-01

Service Name: GenericAttribute
Service UUID: 00001801-0000-1000-8000-00805f9b34fb
Characteristic name: ServiceChanged  UUID: 0x2A05 , Handle 9 ,  Value: Unreachable

Service name: 669aa405-0c08-969e-e211-86ad5062675f
Service UUID: 669aa405-0c08-969e-e211-86ad5062675f
Characteristic name :669aab01-0c08-969e-e211-86ad5062675f , User Description: ANT C0 In , Handle: 13 , Value: Unreachable
Characteristic name :669aab02-0c08-969e-e211-86ad5062675f , User Description: ANT C0 Out , Handle: 17 , Value: Unreachable
Characteristic name :669aab21-0c08-969e-e211-86ad5062675f , User Description: ANT C0 Burst in , In Handle: 22 , Value: Unreachable 
Characteristic name :669aab22-0c08-969e-e211-86ad5062675f , User Description: ANT C0 Burst Out , In Handle: 36 , Value: Unreachable 
Characteristic name :669aab03-0c08-969e-e211-86ad5062675f , User Description: ANT C1 In , Handle: 31 , Value: Unreachable
Characteristic name :669aab04-0c08-969e-e211-86ad5062675f , User Description: ANT C1 out , Handle: 35 , Value: Unreachable
Characteristic name :669aab23-0c08-969e-e211-86ad5062675f , User Description: ANT C1 Burst In , Handle: 40 , Value: Unreachable
Characteristic name :669aab24-0c08-969e-e211-86ad5062675f , User Description: ANT C1 Burst Out , Handle: 44 , Value: Unreachable

Screenshots from my phone (that works with the Tacx Training app:
[image]
[image]

StijnVerm, 2022-01-08:
I tried Zwift and Kinomap on PC and iPhone, but both didn't recognize the unit. Rouvy says clearly that Bushido T2780 is supported. Not a word about T2790.

WouterJD, 2022-01-08:
Tacx ever wanted to build hardware AND software and protect it's market by keeping the protocol between the two a secret.
TTS is Tacx Training Software and by keeping the interface closed, this was a lock-in for all tacx users.

As far as known now, tacx has build
- USB connected trainers, two implementations: legacy and new
- ANT connected trainers, two implementations: Bushido and Genius
- and now there is the next flavor: BLE-connected trainers

[TotalReverse](https://github.com/totalreverse/ttyT1941/wiki) described the interface for USB and based upon that information and [antifier's](https://github.com/john-38787364/antifier) work, FortiusAnt was build, later extended with the ANT-protocol.

For the USB-trainers, FortiusAnt translates the USB-signal to the open ANT-protocol.
For the ANT-trainers, FortiusAnt translates the proprietary ANT-signal to the open ANT-protocol.

Also, FortiusAnt is able to translate not only into the ANT-protocol, but also to the BLE-protocol.

---

Technically it is possible to add the option:
For the BLE-trainers, FortiusAnt translates the proprietary ANT-signal to the open ANT-[or BLE] protocol .

BUT: who is going to reverse engineer the proprietary Tacx-BLE protocol and described it?

From the fact that the T2790 advertises itself as an ANT-trainer (I assume erroneously copied from the proprietary Tacx-ANT-trainers) I would guess that this is an experimental set of trainers (just a few sold), especially because you are the first mentioning this type of trainer as you already guess in your initial question:
> The T2790 is a strange device in the Tacx line-up.

---
Unless anybody reading this post has additional information, I think I must disappoint you; of course your are absolutely free to take the code, resolve the issue. I will lcertainly support you and eventually integrate into FortiusAnt.

Just le me know what your thoughts are

WouterJD, 2022-01-08:
[**Bushido voor Tablet T2790**](https://www.futurumshop.nl/blog/article/keuzehulp/verschillen-tussen-de-tacx-bushido-trainers)
Dit is de eerste Smart trainer van Tacx en is verkrijgbaar vanaf seizoen 2013/2014. ... 
De Bushido T2790 kan gekoppeld worden aan een iPad 3/Mini/Air of nieuwer of een Android tablet met sotware versie 4.3 of nieuwe en Bluetooth Smart. De verbinding tussen de trainer en de tablet verloopt draadloos middels Bluetooth Smart en de Tacx Cycling App. 
Het is ook mogelijk om de T2790 te koppelen met een iPhone of Android smartphone. Je kan dan de Tacx Training App downloaden Uitgebreide informatie over de Tacx Cycling App (tablet) en Tacx Training App (smartphone) vind je in ons andere blog.  ...

**Toekomstbestendig**
De T1980 e n T2790 zijn beide oudere modellen met een gesloten Bluetooth Smart en ANT+ signaal. De Bushido Smart T2780 heeft een open signaal. De T1980 en T2790 zijn hierdoor niet toekomst bestendig. Dit betekend dat deze niet geschikt zijn voor software en upgrades die Tacx in de toekomst gaat uitbrengen. De T2780 is door zijn open signaal wel geschikt voor technologische ontwikkelingen en nieuwe producten en diensten van Tacx.

WouterJD, 2022-01-08:
Community: who would like to integrate this trainer into FortiusAnt ?

switchabl, 2022-01-09:
Fascinating. So there still is a Tacx legacy model I don't have. I somehow assumed the T2790 was just a T2780 bundled with a tablet holder. But it seems it is rather a T1980 with a different firmware. It appears to be somehwat rare, so not sure I will be able to buy one.

I have some idea about how it works and could probably write up some best-guesses to be used as a starting point if someone wants to try. I probably won't be able to implement it myself anytime soon.

Also, while I think eventually adding support for this to FortiusANT would be nice, I would very much recommend getting this to work separately first.
- Integrating means adding another "category" of device. There is no existing infrastructure in FortiusANT for BLE peripherals yet so you will need to build that anyway. Easier to get this working if you do not have to worry how it fits into the existing code. Also, when we finally refactor the FortiusANT main loop (one day), this will get _much_ easier.
- FortiusANT currently uses a separate BLE stack (through Node.js and bleno) and not the native one. So you would have to either 1) write the BLE code in JavaScript (with the noble library) to use the same USB dongle or 2) require a _third_ USB dongle.

So, for a first proof-of-concept, I would either:
- Take the BLE FTMS (JavaScript/bleno) code from FortiusANT and add an implementation for the T2790 BLE interface, also in JavaScript (using noble)
- Use Python with Bleak to control the T2790. Reimplement BLE FTMS in Python using the new Bless library. This unfortunately wasn't really available when @marcoveeneman added BLE to FortiusANT, but hopefully we can eventually move FortiusANT to Bless as well. Removing the need for Node.js and Zadig would make BLE much easier to use.

switchabl, 2022-01-09:
> I have some idea about how it works and could probably write up some best-guesses to be used as a starting point if someone wants to try.

Actually, I won't have to, I just remembered this exists:
https://github.com/jedla22/BleTrainerControl/blob/master/How-to%20FE-C%20over%20BLE%20v1_0_0.pdf

WouterJD, 2022-01-09:
> * Removing the need for Node.js and Zadig would make BLE much easier to use.

If you give me some hints where to start (even in 1:1 email) I am happy to implement; I consider the node.js dependency  as a "too big overhead".

Let's close the open developments first so we have a solid baseline for a next BLE implementation.

WouterJD, 2022-01-09:
> so not sure I will be able to buy one.

https://link.marktplaats.nl/m1795113379

StijnVerm, 2022-01-09:
I can send you the electronics board of my device for some time if that would help. The generator is identical of the t2780.

CAUTION: External Email

https://link.marktplaats.nl/m1795113379<https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Flink.marktplaats.nl%2Fm1795113379&data=04%7C01%7CStijn.vermeersch%40Eastman.com%7C2a07f76065004c39ecff08d9d39b317b%7C9b50c80f103b4433a91cb4b06c507387%7C0%7C0%7C637773485789110173%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=G4XKbDSZCZ%2FqNpXKBPvCpiKHtW4ovQpPtYPUQ6frcyk%3D&reserved=0>

vaneeDutch, 2022-01-20:
I'd personally be extremely interested if you ever managed to get this working.  I bought this bushido for tablet mistakenly thinking that the ble would work with zwift, how wrong I was.  I intended for it to be a way for me and my kids to be able to cycle together but am rather disappointed so far especially as the turbo works just fine with the tacx app but not with anything else.  Appreciate all the effort you've put in and hopefully you find a work around for this turbo at some point in the future.

closed 2022-08-17 by WouterJD