WouterJD/FortiusANT #359: I am totally Lost ... and i am not a GEEK. Please help!

Ghostina84, 2022-01-08:
Hey guys,

i am really lost. I bought a Tacx Genius T2021 and the BlackTrack. Also i have the dongle from tacx Ant+ T2018.

The Dongle is also Found in the lastest Software from Tacx as ANT Device (green) same as the bluetooth!

I know that my Trainer is too old  but there is a workaround for with you guys. But i am not able to get through the "help PDF" ... maybe i am just not able to understand it or i am too stupid! ;-) 

I have understood if i need to run the Fortius Antifier in order to connect my Trainer to Swift or Tacx.

So i downloaded the Fortius ANT-Master and run the FortiusANT.exe in the folder Windowsexecutable.

But it says that there is no TACX trainer found? Please help!

WouterJD, 2022-01-08:
Hi @Ghostina84 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

WouterJD, 2022-01-08:
> So i downloaded the Fortius ANT-Master and run the FortiusANT.exe in the folder Windowsexecutable.

In addition to my previous remark; well you are not stupid and on the right way.

If you start FortiusAnt.exe by double-clicking, it starts with the standard settings, searching for a USB-connected Tacx trainer. This will not work, since your trainer communicates through an ANT-dongle.

You have started from the WindowsExecutable folder, now take a look in the StartUp folder, where you find many ways to start FortiusAnt.

As a first step, double-click **FortiusAnt (exe).bat** which will give you the same result. Now you know how starting an executable through a BAT-file (also called script) works. The BAT-file is a text-file that can be edited with notepad, containing a command-line.

Now double-click **FortiusAnt (gui, Genius, autostart).bat** which contains another command-line. Instead of looking for a USB-connected tacx-trainer it will search for an ANT+-connected Genius.

Perhaps you think, why not just search, after that no USB-trainer is found for a Genius or a Bushido. There are two reaons: 1. we did not build that, 2. detecting a Genius/Busido may take minutes. So we would be waiting minutes for being able to state that no Bushido is found before even trying to search for the Genius. Therefore the three scripts exist, telling what Tacx you have, avdoing unnecessary searching for something that is not there anyway.

jujroy, 2022-01-12:
Welcome aboard.  I am not a geek either but with the good advice from @WouterJD and the manual, I managed to get evreything running.  There is hope :)  I love it especially the virtual gearbox.  Thanks to Wouter.

WouterJD, 2022-01-12:
@Ghostina84 how are things going?

closed 2022-01-18 by WouterJD