WouterJD/FortiusANT #361: Crash on float value in Cadence and Power when using test mode (-s) on Raspberry Pi Zero W

edwinvanpoeijer, 2022-01-12:
When starting FortiusANT in testmode and bluetooth + display enables it crashed with the following error:

17:35:24,136: Tacx   - Simulated Tacx Trainer to test ANT-interface
17:35:24,787: Target=100W 34.0km/h hr= 95 Current= 12W Cad= 97 r=2345 100% -----
Traceback (most recent call last):
  File "/home/pi/FortiusANT/raspberry/../pythoncode/FortiusAnt.py", line 716, in <module>
    mainProgram()
  File "/home/pi/FortiusANT/raspberry/../pythoncode/FortiusAnt.py", line 661, in mainProgram
    Console.Autostart()
  File "/home/pi/FortiusANT/raspberry/../pythoncode/FortiusAnt.py", line 218, in Autostart
    Tacx2Dongle(self)
  File "/home/pi/FortiusANT/raspberry/../pythoncode/FortiusAnt.py", line 161, in Tacx2Dongle
    rtn = FortiusAntBody.Tacx2Dongle(self)
  File "/home/pi/FortiusANT/pythoncode/FortiusAntBody.py", line 653, in Tacx2Dongle
    rtn = Tacx2DongleSub(FortiusAntGui, Restart)
  File "/home/pi/FortiusANT/pythoncode/FortiusAntBody.py", line 1107, in Tacx2DongleSub
    o.SetValues(TacxTrainer.VirtualSpeedKmh,
  File "/home/pi/FortiusANT/pythoncode/raspberry.py", line 843, in _SetValuesSt7789
    [ 'Power'  , "{:d}Watt".format(iPower) ],\

I have changed to code on lines 843 to the following, casting the float to int:

                                    [ 'Cadence', "{:d}/min".format(int(iRevs))  ],\
                                    [ 'Power'  , "{:d}Watt".format(int(iPower)) ],\
Now the testmode is working and sending data to Zwift using bluetooth.

WouterJD, 2022-01-13:
@edwinvanpoeijer thanks for the investigation.

The problem is in usbTrainer.py (CurrentPower is not integer) and of course it can be corrected as you did; I submitted a change (in master, usbTrainer.py) to solve at the place where the problem was created.

Could you please verify, doing the simulation (of course without your solution).
Thanks for cooperation.

edwinvanpoeijer, 2022-01-14:
Hi,

You will also need to do the same on the Cadence value, after that it's working.

WouterJD, 2022-01-14:
Done, please verify

edwinvanpoeijer, 2022-01-15:
Thanks, working perfectly now!

WouterJD, 2022-01-15:
Great thanks for feedback

closed 2022-01-15 by WouterJD