WouterJD/FortiusANT #362: Grade scaling bug with bluetooth

decodeais, 2022-01-13:
Hello, this little bug  droves us crazy. For my wife the Grade parameter worked but not for me. She rides with the old Rouvy App and I with the new one. We found out that the reason was that the old Rouvy app uses only ANT+ and the new one only Bluetooth.
The source code showed the reason. The calculation was OK but not used.

Here in  line 1357is the original grade sent to the trainer
`TacxTrainer.SetGrade(bleCTP.TargetGrade)`

with this change it worked 
`TacxTrainer.SetGrade(Grade)

please put this little change to the source.

WouterJD, 2022-01-13:
You are right, I have updated FortiusAntBody.py; please verify

decodeais, 2022-01-13:
I tested it one hour with Rouvy and worked perfectly. I don' t understand how  it could be  undetected so long. It is absolute important for this App. 
Thank you for your quick respond.

WouterJD, 2022-01-13:
Hi @decodeais evidently you were the first using this combi :-)
Thanks for the work and I'll close the issue

WouterJD, 2022-04-11:
Implemented in https://github.com/WouterJD/FortiusANT/tree/Implement-BLE-using-bless

closed 2022-04-11 by WouterJD