WouterJD/FortiusANT #363: Request/Idea: Add Waveshare 1.3 LCD (button IO is different)

edwinvanpoeijer, 2022-01-14:
I am using a Waveshare 1.3 LCD, the one with 3 buttons and joystick.

This will need some small tweaking but this display can be bought cheaply through Aliexpress or Amazon,
https://www.waveshare.com/wiki/1.3inch_LCD_HAT

The display will work right out the box but the buttons need some tweaking in de _SetupDisplatyT7789, would be nice to do this while starting up as argument in the -O display. 
The buttons are on D16, D20, D21.

            self.buttonA = digitalio.DigitalInOut(board.D20)
            self.buttonB = digitalio.DigitalInOut(board.D16)

When changing the portnumbers FortiusANT went into shutdown without any button press. After some investigation i found out that the ports are switched to input but without Pull.Up. They may not be needed for the Adafruit display?

            self.buttonA.switch_to_input(digitalio.Pull.UP)
            self.buttonB.switch_to_input(digitalio.Pull.UP)

edwinvanpoeijer, 2022-01-14:
Busy getting it operational :)
Running into a small problem, but that has nothing to do with de display.
I am using a distribution without xwindows, everything works fine if i login as pi and start FortiusANT.
If i do the same but then automatically (tried all variations) FortiusANT will start but bluetooth is not working. Well, FortiusANT thinks its working but Zwift doesn't see any bluetooth device.

As soon as everything is working i will post the setup :)

WouterJD, 2022-01-14:
FortiusAnt uses node.js for BLE, debugging may be complex to understand what is not working well. If youhave a clear problem-description, please create new issue.

WouterJD, 2022-01-14:
I've gone for the easy way: `-O st7789b` which should do the job. If there will pop up more displays, a more sophisticated implementation can be done.

Please validate the branch:
https://github.com/WouterJD/FortiusANT/tree/Issue-%23363-Add-Waveshare-1.3-LCD
as referred in the pull request:
https://github.com/WouterJD/FortiusANT/pull/365

WouterJD, 2022-08-09:
Implemented and released

closed 2022-08-09 by WouterJD