WouterJD/FortiusANT #366: Implement BLE, using bleak and bless

WouterJD, 2022-01-18:
By using bless, native python -> BLE libraries are used and the node.js construction can be kept as an alternative.

Advantage:
- Easier installation
Risks:
- Node.js is a proven solution that works well (after the complex installation); bless and bleak still must be proven to be stable
- The new implementation must/should be tested on windows, raspberry and MacOS
Remark for windows users
- Native bluetooth interfaces cannot be used to advertise BLE (only client devices can be connected)
- When node.js is used, Bluetooth-4 dongles must be used and the driver replaced by Zadig
- When bless is used, Bluetooth-5 dongles must be used and no special drivers are needed

---
It's under investigation; see bless and bleak for currently open issues.

WouterJD, 2022-02-23:
I'm working on this issue for a native python implementation for BLE; to remove the nodejs complexity AND to resolve issues (the structure is too complex).

First version is in https://github.com/WouterJD/FortiusANT/tree/Implement-BLE-using-bless

cleCollector.py is a client to any FitnessMachine (trainer, FortiusAnt) like TrainerRoad and/or Zwift with purpose to show data and make some simple simulations.

Although at 98% I do not get it work and have raised an issue with the bleak developer
https://github.com/hbldh/bleak/discussions/772#discussioncomment-2239103

Who can tell me what I'm doing wrong 🙂

decodeais, 2022-02-23:
I tried to understand what you are doing. I have not the background and the knowledge to help you. But why can't you you use a ready BLE Analyser like the one from Keuwlsoft. It reads all messages and had the possibility send something? If the question to stupid please erase this text.

Toetta, 2022-02-23:
Im happy to try on my installation if you can harvest any info from it... Otherwise I think everything is way beyond my level

WouterJD, 2022-02-23:
@decodeais not a stupid remark at all. When I learn an interface, I tend to build something useful. In doing it, I understand the issues, also for other programmers.
So it's not so much BLE-analysis but more end2end interface building. This time for bleak.

MeanHat, 2022-02-24:
@WouterJD - I remember you saying that you wanted to simplify the BLE installation - that's great work to remove node.js. I've no expertise in this area but would be very happy to lend a hand with RPi testing (0W, 02W, 3B+) and the manual if you want any assistance. Keith

WouterJD, 2022-02-24:
@MeanHat thanks. First client version will be available this week. This client can be used to "see" (and Analyse) what is transmitted by FortiusAnt.

MeanHat, 2022-02-24:
Great work @WouterJD !

Toetta, 2022-02-24:
👍🏻🥂🤘🏻🔥

WouterJD, 2022-02-24:
https://github.com/WouterJD/FortiusANT/blob/Implement-BLE-using-bless/pythoncode/bleClient.py
is now available.
You can use this program to validate the transmission of FortiusAnt (BLE).
I see it works on raspberry rpi0W raspbian (10) buster
It does not work weel on Windows 10 (indications not received), reason unknown

- Start `FortiusAnt -b -s`  (bluetooth, simulation mode)
- Start `bleClient.py` (no parameters required)

WouterJD, 2022-02-25:
Good news, prototype for bleServer.py is working and pairing with Trainer Road and Rouvy.
To be cleaned for testable version...
Work for next week :-)

MeanHat, 2022-02-25:
> Good news, prototype for bleServer.py is working and pairing with Trainer Road and Rouvy. To be cleaned for testable version... Work for next week :-)

That's really great news @WouterJD - great work! That is a gamechanger! I'd be delighted to test with Zwift and Fortius when ready. On standby...

decodeais, 2022-02-25:
> Good news, prototype for bleServer.py is working and pairing with Trainer Road and Rouvy. To be cleaned for testable version... Work for next week :-)

I would like to test  pre alpha state to. It would be interesting to get a chance for bug hunting.

WouterJD, 2022-02-25:
It's like a cliffhanger... To be continued in next episode😀

MeanHat, 2022-02-25:
Can't wait!

BikeBeppe64, 2022-02-25:
> It's like a cliffhanger... To be continued in next episode😀

All we need is.... your next episode!!

MeanHat, 2022-03-04:
Hi @WouterJD  - I've been testing today as you requested (using Pi 0 2 W as it is a lot quicker and I can then revert to Pi 0 W). I have done / found the following:

1. There was an error cloning the branch. I increased the swapfile size from 100 (to 1024) to allow me to clone the branch. `sudo dphys swapfileoff`;  `sudo nano /etc/dphys-swapfile` changing to CONF_SWAPSIZE = 1024; `sudo dphys-swapfile setup`; `sudo dphys-swapfile swapon`
2. I modified FortiusAnt.sh to -a -bb -D-1 -s -d CEWIDbu
3. There was an error finding lib_detect_testenv so I installed manually `git clone https://github.com/bitranox/lib_detect_testenv.git` followed by `cd lib_detect_testenv` and `sudo python setup.py install`
4. There was a service_identity error so I installed manually `python3 -m pip install service_identity`
5. I am now getting an error: "org.freedesktop.DBus.Error.AccessDenied: Connection 1.10 not allowed". I ran `sudo usermod -G bluetooth -a pi` and this resulted in the same error but Connection 1.49

The log file is attached for info: [FortiusAnt.2022-03-04 16-25-06.log](https://github.com/WouterJD/FortiusANT/files/8187482/FortiusAnt.2022-03-04.16-25-06.log)

I've had to stop at this stage but happy to pick up again at a future date. Hope this is a small help.

WouterJD, 2022-03-14:
Hi guys, I'm still engineering the code. It's not trivial.
As you see in the headers, some questions and support required from bless & bleak :-)

WouterJD, 2022-03-14:
> 5\. I am now getting an error: "org.freedesktop.DBus.Error.AccessDenied: Connection 1.10 not allowed". I ran `sudo usermod -G bluetooth -a pi` and this resulted in the same error but Connection 1.49

See bless/issue/#60

In
`/etc/dbus-1/system.d/bluetooth.conf`
add
```
  <policy user="pi">
    <allow own="org.bluez.FortiusAntTrainer"/>
    <allow send_destination="org.bluez.FortiusAntTrainer"/>
  </policy> 
```

MeanHat, 2022-03-15:
Great work @WouterJD! I should have some time for more testing to support you next week and will run the addition you suggest.

WouterJD, 2022-03-22:
Current version is 99% final.
There is one [open issue](https://github.com/kevincar/bless/discussions/76); the DeviceName and Appearance cannot be set. Let's hope @kevincar can answer how to do that. It's not blocking for TrainerRoad or Rouvy.

Toetta, 2022-03-23:
Fantastic news, great work

WouterJD, 2022-03-23:
> Fantastic news, great work

Hi Malkus, feel free to test!

WouterJD, 2022-03-24:
When FortiusAnt is advertising over BLE, depending on the interface and backends, the CTP (Trainer Road, Rouvy) will see the following devices:

| a  | Windows | Raspberry | |
|---|---|---|---|
| -b using NodeJs | hostname | FortiusANT trainer | |
| -bb using Bless | hostname | BLE-ADDRESS | |
| | 
| Useful sources: | | [check here](https://stackoverflow.com/questions/26299053/changing-raspberry-pi-bluetooth-device-name)  and [here](https://punchthrough.com/creating-a-ble-peripheral-with-bluez/) | It did not help me |

With my knowledge and what I found, there is not much I can do about it.

There is no practical difference, since your CTP sees one usable device and you select it.
Having "FortiusAnt Trainer" as displayed name would have been more elegant.

For I now, I surrender.

Toetta, 2022-03-27:
64bit Bullseye up and running with FortiusANT using bless =)

[image]

WouterJD, 2022-03-29:
Congrats. Does your app recognize it?

Toetta, 2022-03-29:
Yes! Have taken a ride with Zwift and worked well!

Possibly dropping cconnection adter some time. I will have to investigate that further whats going on

WouterJD, 2022-03-29:
Cheerio! Let me kno😀

MeanHat, 2022-03-30:
Hi @WouterJD - I've been trying bless branch on RPi0 2W. Clean reinstall with Debian Bullseye 32bit.
- I had to change the permissions on some of the new raspberry *.sh files
- I had to install `lib_detect_testenv` using the install I described above [https://github.com/WouterJD/FortiusANT/issues/366#issuecomment-1059339128](url) and also increase the swapfile size to 1024 to prevent error when using git clone to get the new branch "Implement-BLE-using-bless".
- I used options -a -bb -s -D-1 -d CEWIDbpjmuafA in FortiusAntMain.sh
- FortiusANT ran yet while using bluetooth on CTP=Zwift running on either Win11 or Android devices I could not detect any bluetooth devices
- I have attached the logfile:
[FortiusAnt.2022-03-30 16-24-32.log](https://github.com/WouterJD/FortiusANT/files/8381972/FortiusAnt.2022-03-30.16-24-32.log)
-  I have no doubt made a mistake somewhere. Do you have any suggestions?

Toetta, 2022-04-04:
I had problems getting Zwift to have a stable connection with the 64 bit. The raspberry disappeared from the available devices and got lost signal responce from zwift. Possibly resolved just now by installing the x11-utils package..? Now managed to pedal a couple of km with stable connection :) will continue later

WouterJD, 2022-04-07:
I have uploaded some error-handling improvements and access rights settings

WouterJD, 2022-04-07:
> * I used options -a -bb -s -D-1 -d CEWIDbpjmuafA in FortiusAntMain.sh

To avoid this, you can use `-d all` or `-d log`; the difference is that log will not create a json-file.

Note that in the standard python logging system, the messges are hierarchical:

- D(ebug) produces D, I, W, E, C
- I(NFO) produces I, W, E, C
- W(ARNING) produces W, E, C
- E(RROR) produces E, C
- C(RITICAL) produces Critial messages only

`-d CEWID` is therefore identical to `-d D`

As opposed to the FortiusAnt flag-approach, where `-d DW` would produce Debug and Warning but not the others.

MeanHat, 2022-04-14:
In answer to your points above  https://github.com/WouterJD/FortiusANT/issues/366#issuecomment-1090532681:

- OK
- I suspect it might be Bullseye specific
- My mistake - you are correct. I meant FortiusAnt.sh
- Yes, I tried running bleBleak.py but without success
- OK
- OK

Testing again...

WouterJD, 2022-04-14:
>  I tried running bleBleak.py but without success

Please explain, copy output...

MeanHat, 2022-04-15:
Hi @WouterJD - I'm testing again, this time using Debian Buster rather than Bullseye on an RPi0 2 W and here is an update of the results:
- in `3_InstallBless_Temporarily.sh`: (a) line 3 there is a semicolon ";" which causes an error - I replaced with hyphen "-" which has removed error; (b) error reported stating need python >=3.6 and only v2.x available. In line 4 I replaced "pip" with "pip3" which removed the error; (c) Path error so added `home/pi/.local/bin` to PATH
- ran `FortiusANT.sh` with `-a -bb -s -D-1` but Zwift did not find any device
- ran `Test bleBleak.sh` but MAC address of Android device running Zwift (Google Pixel 4a 5G) not found

I will do a fresh install of Buster and try Node.js to check "original" setup is still functioning and problem is not related to RPi0 2 W

MeanHat, 2022-04-15:
ps - I've been using RPi0 2 W rather than RPi0 as it is a lot faster to set up etc.

MeanHat, 2022-04-15:
RPi 0 2 W worked with Node.js (though needs fix in issue #57 

Will now test RPi 0 with Bullseye and bless

WouterJD, 2022-04-15:
Special.
I've updated `3_InstallBless_Temporarily.sh` (to be published)
Will communicated what device I have, I think it's a new type because booting much faster than I was used to.

MeanHat, 2022-04-18:
Here's an update on the RPi testing:
| OS       | Buster 10.12                               | Buster 10.12                   | Bullseye 11.3                              | Bullseye 11.3          |
|----------|--------------------------------------------|--------------------------------|--------------------------------------------|------------------------|
| **BLE Mode** | **Node.js**                                    | **bless**                          | **Node.js**                                    | **bless**                  |
| RPi0     | OK (other than disconnect - see issue #57) | NO (see log)   | OK (other than disconnect - see issue #57) (see log)| NO (see log)           |
| RPi0 2   | OK (other than disconnect - see issue #57) | NO (see log posted previously) | NO                                         | NO (see previous post) |

I have focussed on RPi0 as RPi0 2 is fairly different (ARMv7-l vs ARMv6-l, Bluetooth 4.2 vs 4.1 etc). The closest I came was RPi0 running Buster and bless linking to Zwift as the CTP. Zwift recognised FortiusANT but could not achieve a stable connection in simulation mode. Hopefully the log file will help. During the install (running shell scripts 1, 2, [3_InstallBless_Temporarily.sh](https://github.com/WouterJD/FortiusANT/blob/Implement-BLE-using-bless/raspberry/3_InstallBless_Temporarily.sh), 5, [9_GrantAccessToBluetoothForBless.sh](https://github.com/WouterJD/FortiusANT/blob/Implement-BLE-using-bless/raspberry/9_GrantAccessToBluetoothForBless.sh) and [9_GrantAccessToUSB_withReboot.sh](https://github.com/WouterJD/FortiusANT/blob/Implement-BLE-using-bless/raspberry/9_GrantAccessToUSB_withReboot.sh) the main errors were (a) in shell 3 I had to change line 4 from python to python3 and (b) there was a prompt to add /home/pi/.local/bin to PATH.

Hope this helps a little. I'm keen to carry on the work to get RPi0 running Buster working with bless. (Next steps could then be to extend the solution further to use Bullseye and then RPi0 2?)

[RPi0_bullseye_bless.log](https://github.com/WouterJD/FortiusANT/files/8506939/RPi0_bullseye_bless.log)
[RPi0_bullseye_nodejs.log](https://github.com/WouterJD/FortiusANT/files/8506940/RPi0_bullseye_nodejs.log)
[RPi0_buster_bless.log](https://github.com/WouterJD/FortiusANT/files/8506942/RPi0_buster_bless.log)

MeanHat, 2022-04-18:
> Special. I've updated `3_InstallBless_Temporarily.sh` (to be published) Will communicated what device I have, I think it's a new type because booting much faster than I was used to.

Thank you - I used this in the testing reported above

WouterJD, 2022-08-09:
Implemented and released

closed 2022-08-09 by WouterJD

WouterJD, 2022-08-09:
> Here's an update on the RPi testing:

Hi @MeanHat I have released the BLE/bless version today, and I come across this test-work in my Inbox-to-be-handled.
Please share your thoughts.

I'm trying to get clean for the tacx-season :-)

MeanHat, 2022-08-15:
Hi @WouterJD - I am sorry for the slow reply. I've been working on some other projects and got distracted! I have not tested bleak & bless since my last trials (I submitted feedback and comments). If you'd like me to test anything again on RPi3, RPi0 or RPi0 2 please let me know.

WouterJD, 2022-08-15:
Hi @MeanHat
I did not do a full regression test, so such a test would be very welcome.
If there are any issues, we can quickly resolve.
---
I wanted to proceed with steering, using the BLE base, without keeping branches open - although technically possible

MeanHat, 2022-08-15:
Ok @WouterJD - leave it with me for a few days and I'll do a full test.

WouterJD, 2022-08-15:
👍

WouterJD, 2022-12-28:
Have been quite busy lately; just stepped in again and did some small stuff.
RPi testing will come later.

Now trying to get TrainerRoad back up again.