WouterJD/FortiusANT #367: Rem wordt niet gevonden

Anniehoho, 2022-01-23:
Na vanochtend een goede sessie te hebben gehad, lukt het vanmiddag niet om de rem te kalibreren. De fortius ANt maakt connectie met de smartfone en ook met de Genius, maar er is geen interactie tussen stijgingspercentage en weerstand op de rem. Wat te doen?

WouterJD, 2022-01-23:
Welkom in de FortiusANT community
- - -
Ik ben altijd nieuwsgierig met wie ik communiceer, waar FortiusANT wordt gebruikt en in welke configuratie.
Welke bundle heb je gekocht, en welke rem en head-unit gebruik je?
Vertel meer over jezelf, eventueel onder issue #14.
- - -

Hoi @Anniehoho jammer dat eea niet meer werrkt. Je geeft echter niet veel info. 

Begin Eens met een foto van FortiusAnt, bij voorkeur met een tracefile (-d255)

Anniehoho, 2022-01-29:
Ik gebruik de Raspberry PiO met Forrius Ant en de Break is de Tacx Genius Brake T2021. Plaatjes heb ik bijgevoegd.  Ook van het script in een video omdat ik niet zo handig ben dat ik een tracefile weet te pakken. 
[image]
[image]
[image]
[image]
[image]

Anniehoho, 2022-01-29:
Ik hoop dat je mij kan helpen

WouterJD, 2022-02-01:
Actually, the messages look good "Pair with Tacx Genius this can take a minute". What happens after that minute?

WouterJD, 2022-02-01:
You say you are not so very handy, but did a series of good steps already:
- Raspberry Pi0
- Start FortiusAnt with -t Genius option; requiring ANT-dongle
- ANT-dongle is present in your pi
- ...

Just some additional steps and it will work; assuming no device is broken.

WouterJD, 2022-02-15:
Since there is no communication here, I assume can be closed.
If not happy to reopen

closed 2022-02-15 by WouterJD

Anniehoho, 2022-02-17:
Unfortunately I am to busy to respond quickly. In two weeks I will get back on the issue (which is not solved) Verzonden vanuit Mail voor Windows Van: Wouter DubbeldamVerzonden: dinsdag 15 februari 2022 13:18Aan: WouterJD/FortiusANTCC: Anniehoho; State changeOnderwerp: Re: [WouterJD/FortiusANT] Genius brake not found (Issue #367) Since there is no communication here, I assume can be closed.If not happy to reopen

WouterJD, 2022-02-17:
As soon as you have time, we can reopen and pick it up.