WouterJD/FortiusANT #368: Communication with old TTS

snarf-snarf-snarf, 2022-01-30:
Is it possible to run Fortius ANT on a laptop (connected with USB cable to an iMagic or Fortius trainer) with an ANT+ dongle or BLE and then connect to another PC with its own ANT+ dongle or BLE *but* that PC runs the OLD Tacx trainer software?

I ask since I don't want to pay for rental software per month as it adds up and I bought all the Tacx real life videos at great expense years ago. I can't run the old Tacx Trainer Software on my laptop (I won't go into all the details why) but I can run it on a PC in another room.

So essentially I want Fortius ANT to present the received data stream as if I'd plugged in the real USB head unit.

Many thanks, this is a great project keep up the good work!

WouterJD, 2022-01-30:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

I would say Yes but admit I do not fully unusually understand your intended setup. Adding a picture would help.
What CTP do you want to use?

snarf-snarf-snarf, 2022-01-30:
I want to use the old Tacx trainer software I have but instead of running it on the machine that is connected to the head unit I want to run it on another PC and use your software to route the data across to it via ANT+. I don't want to use any of the new subscription based CTPs as these are expensive and I already gave Tacx thousands of pounds over the years. I have an iMagic, Fortius, several editions of TTS and just about every RLV they ever sold.

WouterJD, 2022-01-30:
Hi @snarf-snarf-snarf
Clear. If TTS pairs with a ANT trainer, yes you can

Frenske8, 2022-01-30:
The distance between the ANT-dongles of the  laptop and PC in another room  might be a problem ?

snarf-snarf-snarf, 2022-01-30:
True I don't know how far they work from, or if they can work through an internal wall? Pity it's not possible to use the WiFi!

WouterJD, 2022-01-30:
To be honest I do not understand your configuration; what and who is where.
As @Frenske8 says, I would not rely on comms through wall, that's not what ANT has been thought for.

Please provide image what you want to achieve

snarf-snarf-snarf, 2022-02-02:
So I installed FortiusAnt and got the GUI showing on my laptop and could ride the iMagic, change the power etc. from that which is great. I tried to get it to communicate with my PC via Bluetooth by starting it with -b switch. But I got error code 2 file not found showing. I can connect via Bluetooth in Windows to my PC so reception is OK.

I found the documentation confusing about Bluetooth, so I just ran the exes to install the driver and run FortiusAnt. Do I need to do other steps to get Bluetooth working? I see there are detailed steps further down saying to install nodejs etc. Is this required when running from the downloaded exe?

WouterJD, 2022-02-02:
Hi @snarf-snarf-snarf yes, node.js is required.
Sorry that the BLE installation is so complex but it's the way it is. Suggestions where manual is confusing are welcome

snarf-snarf-snarf, 2022-02-02:
Well I've made some more progress, I installed nodejs and then had trouble with npm install as it said my visual studio 22 was too new and wouldn't accept it. So installed visual studio 2017 as well which resolved that. Now when I start FortiusAnt it says it has found the Bluetooth interface which sounds promising.Calling it a night now, will try another evening to see if I can get it to connect to TTS4 in the other room. I'm running that on VMware player so I don't know whether that will work but we will see. Then if that works I will remote desktop from the laptop into the PC so I can see TTS4 on the bike. I'm doing this because I'm restricted on what I can install on the laptop as it's not mine. I can probably get away with github folders but not a full blown TTS4 installation!For the manual I'd recommend saying explicitly after the Windows install section that if you want BlueTooth you need to follow the steps further down. What confused me was they seemed to be part of the raspberry pi stuff so I didn't look at any of that until later.Thanks

WouterJD, 2022-02-03:
Thanks for hint on manual, you're right; bluetooth was added after that the .exe was described.
It will go to github next time the manual is published.

[image]

closed 2022-02-15 by WouterJD