WouterJD/FortiusANT #369: Connections fail very often

WouterJD, 2022-02-01:
I have problems connecting FortiusAnt installed on Win 11, Intel gen 10 i5 Notebook to Tacx Desktop Training App on samme notebook and ANt+ dongle (only connect when I use 2 Ant+ dongles. But connections failed very often. I have Duraace Powercrank but I cannot put D to release analyze?

Med venlig hilsen

John Pihl

Smedeengen 14
3390 Hundested

Tlf.: 4793 8685
Mobil: 2146 7936, John

> Den 30. jan. 2022 kl. 10.23 skrev snarf ***@***.***>:
> 
> ﻿
> Hi, I live in East Anglia in the UK and I have an iMagic and Fortius.
> 
> —
> Reply to this email directly, view it on GitHub, or unsubscribe.
> Triage notifications on the go with GitHub Mobile for iOS or Android. 
> You are receiving this because you were mentioned.

_Originally posted by @John-Imagic in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-1025200834_

WouterJD, 2022-02-01:
Hi @john-imagic Could you please provide some more data on the issue; this is quite short.

John-Imagic, 2022-02-01:
No problem now. I didn’t follow your start-up description. And I run driver install and then restart my laptop twice before I connect. That helped. 

Hilsen John Pihl, sendt fra min iphone

WouterJD, 2022-02-01:
Great to hear, so I super issue can be closed

John-Imagic, 2022-03-18:
> Originally posted by @John-Imagic in #14 (comment)

I sometimes hve this problem:

Hilsen John Pihl, sendt fra min iphone