WouterJD/FortiusANT #372: Problem Node setup on Raspberry

decodeais, 2022-02-06:
When I made a new setup from scratch with the new PI OS ( Bullseye ) every thing worked, accept receiving grade from the training program (Rouvy). Downgrading to the last version  ( Buster ) did not help. When I copied the old node directory from my "Buster" installation into the new setup every thing was ok. I found out that, if I copy the file `package-lock.json` into the node directory, the command `npm install` gives a good result. 
I hope you knows how to fix it. For me it looks like a version conflict, but there are no error messages.
 You find this problem only if you start riding and the trainer remains in Power mode.

decodeais, 2022-02-14:
Has anybody succeeded in installing a new Raspberry Pi 4 with Bluetooth in the last time ?
Where reading from the training program worked.

WouterJD, 2022-02-15:
> You find this problem only if you start riding and the trainer remains in Power mode.

Hi @decodeais does this mean that powermode is correctly working?
Do you use a Rouvy training? Since usually Rouvy will use Grademode.

Please provide some more details

decodeais, 2022-02-15:
When I started  riding a Rouvy route the program shows 100W for the beginning. This is defined in the software. When I rode, this should have changed to grade, but it did not. FortiusAnt was recognized as a smart trainer, but the messages seemed not to arrive at the program. When I added the following file `https://github.com/decodeais/FortiusANT/blob/master/node/package-lock.json`    from my old installation to your node direction, your installation procedure works. Then I receive the grade values.
May be the bug would appear on your RPI4  when you erase the package_lock file and the module directory  and  then start a new node installation.
 It was very confusing to me, but now it works for me.

WouterJD, 2022-02-15:
The file you mention is (I assume) created by the installation procedure of node.js.
Node.js is very loosely coupled and relatively complex for what it does, but solves the lack (at time of BLE implementation) of a native BLE-library for python.

I think there is not much I can do, other than using the recently available BLE libraries for python, which takes some time to implement

WouterJD, 2022-02-15:
In the meantime I'm happy it now works for you.
You want this issue to remain open?

decodeais, 2022-02-16:
I think  it's interesting whether I am  the only one. My file would help someone else to solve the problem. When I have a little more time I will test power mode with rouvy or Golden Cheetah. 
Feel free to change the header, I took this dramatic words to get faster response.
Is it possible to put the file to the installation?
 I tested it with last 2 RPI OS versions.

planetSammy, 2022-02-16:
Hello together, 
I have exactly the same Problem by using the program with my RP Zero W and RGT-Cycling. When I start the Skript and finished the calibration, i start RGT-Cycling on my PC (Win 10) incl. pairing the "FortiusAnt" Trainer over BLE,
but the Application did not switch to GradeMode, it still shows me 100 Watt.

If I start RGT-Cycling on my iPhone, the Application switch to GradeMode (0%) after a few seconds (this is actually my workaround...)

I try your old JSON-file, maybe this will solve my Problem, too!

Best regards Sammy

planetSammy, 2022-02-16:
Checked with the old Package-lock.json file from your repository, but same behaviour, IOS is okay, Win 10 not!

decodeais, 2022-02-16:
> Checked with the old Package-lock.json file from your repository, but same behaviour, IOS is okay, Win 10 not!

Did you delete the directory from the old installation?

planetSammy, 2022-02-16:
> > Checked with the old Package-lock.json file from your repository, but same behaviour, IOS is okay, Win 10 not!
> 
> 
> 
> Did you delete the directory from the old installation?

No, only replaced the file.

decodeais, 2022-02-16:
No you have to erase the directories from the old installation an make a complete new node installation. The installation will be modified by the package-lock.json file.

WouterJD, 2022-02-16:
> Feel free to change the header, I took this dramatic words to get faster response.

Good idea; although I try to respond as quick as possible and monitor open issues on a regular basis.

> Is it possible to put the file to the installation?

The file is explicitly **excluded** from distribution

After installation of the node.js package you have to execute the **`npm install`** command (iOS and Windows); this command takes the input definition "package.json" and generates **`Package-lock.json`** and the **`node_modules`** directory; both are explicitly excluded from distribution (see .gitignore) because they are considered platform-specific and npm-generated.

So even though your solution may work, I think distributing the mentioned file is not the correct solution to the problem; I would not like to distribute node_modules with the contain 1821 files and 349 folders either.

WouterJD, 2022-02-16:
> Has anybody succeeded in installing a new Raspberry Pi 4 with Bluetooth in the last time ?

I think this is the correct question.
First steps in solving it is to tell whether the file is not created at all or containing incorrect content

decodeais, 2022-02-16:
> > Feel free to change the header, I took this dramatic words to get faster response.
> 
> Good idea; although I try to respond as quick as possible and monitor open issues on a regular basis.
> 
> > Is it possible to put the file to the installation?
> 
> The file is explicitly **excluded** from distribution
> 
> After installation of the node.js package you have to execute the **`npm install`** command (Raspberry, iOS and Windows); this command takes the input definition "package.json" and generates **`Package-lock.json`** and the **`node_modules`** directory; both are explicitly excluded from distribution (see .gitignore) because they are considered platform-specific and npm-generated.
> 
> So even though your solution may work, I think distributing the mentioned file is not the correct solution to the problem; I would not like to distribute node_modules with the contain 1821 files and 349 folders either.

I did not copy all the modules. The JSON file modifies the result of your normal installation script.

WouterJD, 2022-02-16:
[This is what node.js says](https://docs.npmjs.com/cli/v8/configuring-npm/package-lock-json)
[Article 1 on StackOverflow](https://stackoverflow.com/questions/44206782/do-i-commit-the-package-lock-json-file-created-by-npm-5)
[Article 2 on StackOverflow](https://stackoverflow.com/questions/68260784/npm-warn-old-lockfile-the-package-lock-json-file-was-created-with-an-old-version)

For me it's not clear whether node.js wants us to distribute the file yes/no;

- **`package-lock.json`** is automatically generated...; suggests it's not needed
- It was not required before and seems inconsistent over the platforms

@decodeais what happens if you remove **`node_modules`** AND **`package-lock.json`** and then run rpm install?

decodeais, 2022-02-16:
> Some stuff I found: [This is what node.js says about package-lock.json](https://docs.npmjs.com/cli/v8/configuring-npm/package-lock-json) [StackOverflow - do-i-commit-the-package-lock-json-file-created-by-npm-5](https://stackoverflow.com/questions/44206782/do-i-commit-the-package-lock-json-file-created-by-npm-5) [StackOverflow - npm-warn-old-lockfile-the-package-lock-json-file-was-created-with-an-old-version](https://stackoverflow.com/questions/68260784/npm-warn-old-lockfile-the-package-lock-json-file-was-created-with-an-old-version) [StackOverflow - npm-install-not-creating-a-new-package-lock-json](https://stackoverflow.com/questions/45866533/npm-install-not-creating-a-new-package-lock-json)
> 
> For me it's not clear whether node.js wants us to distribute the file yes/no;
> 
> * **`package-lock.json`** is automatically generated...; suggests it's not needed
> * It was not required before and seems inconsistent over the platforms
> 
> @decodeais what happens if you remove **`node_modules`** AND **`package-lock.json`** and then run rpm install?

> Some stuff I found: [This is what node.js says about package-lock.json](https://docs.npmjs.com/cli/v8/configuring-npm/package-lock-json) [StackOverflow - do-i-commit-the-package-lock-json-file-created-by-npm-5](https://stackoverflow.com/questions/44206782/do-i-commit-the-package-lock-json-file-created-by-npm-5) [StackOverflow - npm-warn-old-lockfile-the-package-lock-json-file-was-created-with-an-old-version](https://stackoverflow.com/questions/68260784/npm-warn-old-lockfile-the-package-lock-json-file-was-created-with-an-old-version) [StackOverflow - npm-install-not-creating-a-new-package-lock-json <-- Read this one!!](https://stackoverflow.com/questions/45866533/npm-install-not-creating-a-new-package-lock-json)
> 
> For me it's not clear whether node.js wants us to distribute the file yes/no;
> 
> * **`package-lock.json`** is automatically generated...; suggests it's not needed
> * It was not required before and seems inconsistent over the platforms
> 
> @decodeais what happens if you remove **`node_modules`** AND **`package-lock.json`** and then run rpm install?

I get a installation with a newer release of node, but then with the bug. When I replace the JSON file and the module directory again, start your script, everything works.

decodeais, 2022-02-16:
> I get a installation with a newer release of node, but then with the bug. When I replace the JSON file and the module directory again, start your script, everything works.

The JSON file is not missing, it shows a newer version.
I did this test with different fresh RPI OS installs. Only using your scripts.

WouterJD, 2022-02-16:
> The JSON file is not missing, it shows a newer version.
> I did this test with different fresh RPI OS installs. Only using your scripts.

Really looks like a node.js problem...

planetSammy, 2022-02-17:
My installation is from November 2021 and the Version of NPM is 6.14.15, which version do you have? Does the version depends to the RPi Model?

decodeais, 2022-02-17:
Npm 6.14.9 
Node has the version 15.14.3
If you want to change the version node for testing you can install `npm install -g n`
If you want to install my node version
You have to enter `n 15.14.3`

WouterJD, 2022-02-17:
When writing the documentation, the versions were (see 3_InstallNodeJs.sh):
```
	node version: v14.15.3
	npm  version: 6.14.9
```
since I use a blackberry image, version is unchanged untill today.

Currently on Windows 10, I use:
```
	node version: v16.13.1
	npm  version: 8.1.2
```

planetSammy, 2022-02-17:
I tested an old node version "14.15.3" and the newest 14.19.0 from here https://unofficial-builds.nodejs.org/download/release/

I think the error occours by using a new/different bleno version (see also my post in November 21 https://github.com/WouterJD/FortiusANT/issues/57#issuecomment-962626730)

Can you tell me which bluetooth_hci_socket do you use when it works?

WouterJD, 2022-02-17:
> Can you tell me which bluetooth_hci_socket do you use when it works?

Oops... What do you refer to? I don't know.
If you explain, I will check

planetSammy, 2022-02-17:
> [#57 (comment)](https://github.com/WouterJD/FortiusANT/issues/57#issuecomment-962626730)

MeanHat, 2022-02-17:
I've just finished a re-install of FortiusANT on an RPi 0 W. In order to get is working I used the fix from @planetSammy in #57 (noting that it has to be run from the ~FortiusANT/node folder). I also had to run `npm audit fix` there. In order to get the display (ST7789 minitft 240x240) to work I ran `sudo apt-get install fonts-dejavu` as I mentioned in #379.
Now fully operational. Hope this is helpful.

planetSammy, 2022-02-18:
> I've just finished a re-install of FortiusANT on an RPi 0 W. In order to get is working I used the fix from @planetSammy in #57 (noting that it has to be run from the ~FortiusANT/node folder). I [...]

Hi Meanhat, which Software do you use (Zwift, Bkool, RGT, Rouvy) and switch "FortiusANT" automatic to GradeMode after connecting the VT sensors over BLE?
[image]

MeanHat, 2022-02-18:
Hi @planetSammy, I use Zwift. Though I've been doing the testing in simulation (`-s`) mode and it has been fine running for a mile or so without any problems. However, in simulation, once connected the target power increases markedly from 100w to 200w-300w. I'll be testing it on a Fortius trainer next.

decodeais, 2022-02-20:
I think I erased my SD-Card 100 times. But now I succeded with my installation script.
When I had installed the new OS for RPI ( Bullseye ) and started the last version of that script, FortiusAnt runs on my RPI4 with Bluetooth without any problems. 
At Ubuntu I found a wheel for wxpython . The new OS uses Python3.9 by default, so the old install for wx did not work for me.
UnfortunatlyI did not get rid of the package-lock file, but with an "one click" install it should be easy to test. 
 Install script for RPI4 and OS Bullseye :
[https://github.com/decodeais/FortiusANT/blob/master/raspberry/PiBulleye_Install.sh](url)
Now the connection is stable and the communication for the grade works too.

WouterJD, 2022-02-20:
Congratulations @decodeais
It's quite difficult to support all tacx models, operating systems, versions of os'es and also raspberry "dialects", including the BLE anomalities

decodeais, 2022-02-20:
I think this script is only a starting point. I tested Golden Cheetah, Rouvy Windows and Android, RPI4. I took a lot from your Skripts, but RP Zero has to be implemented in a further step.
 It was a lot of try and error.  You can easy switch to your repository by changing a comment sign. If you do this you will get a total unstable connection without grade because of the missing lockfile for the node modules.

planetSammy, 2022-02-21:
I am lost 😵 ... after some tests my "workaround" to start FortuisAnt over iOS and then with Win10 does not work anymore. Now the software is freezing when I connect my mobile device and run further after stopping my BLE device on my iPhone...

The connection to windows is stable, but the FortiusAnt Software does not switch to GradeMode when I start RGT-Cycling.

Is there any way to start FortiusAnt only in GradeMode by "default"?

WouterJD, 2022-02-21:
hi @planetSammy sorry to read that there are issues.
Starting FortiusAnt in grademode would not be a solution, you also need the grade itself.

Now, FortiusAnt remains in PowerMode/100W (I assume); then FortiusAnt would be in GradeMode/0% (or whatever grade you would like) giving the same static effect.

In the meantime I'm working on a python-native BLE version, which may take some time though.

decodeais, 2022-02-21:
What happens when you install a fresh PI OS Bullseye and try to start my install script.
[PiBulleye_Install.sh](https://github.com/decodeais/FortiusANT/blob/master/raspberry/PiBulleye_Install.sh)
On my RPI4 it works without problems. I know nearly nothing about RP Zero.
 It should be possible to adapt.

Toetta, 2022-02-21:
Struggling with BLE issues as well on . Forget my ANT dongle at home and now trying to get BLE to work. It shows and can connect on Mac and appletv. I get about 2-3s connection and then all inputs freezes on the Zwift. On Pi4 everything looks fine and is receiving heart rate data and all trainer info from the vortex. In Zwift heart rate only shows live values 2-3s then sticks at the noted rate. BLE is still paired but in rider mode it says connection error.

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster

decodeais, 2022-02-21:
For me it was a workaround to copy the node directory from the working one into the RPI with unstable connection. After this both worked.

Toetta, 2022-02-21:
> For me it was a workaround to copy the node directory from the working one into the RPI with unstable connection. After this both worked.

Ok, I might have misunderstood you. (Im completely new to all this) I must get FortiusANT working on some device over BLE and when that is done I can copy that node directory to the RPI? I have never run FortiusANT yet on anything else than on this Pi4...

Cheers

Malkus

BikeBeppe64, 2022-02-21:
> What happens when you install a fresh PI OS Bullseye and try to start my install script.
> [PiBulleye_Install.sh](https://github.com/decodeais/FortiusANT/blob/master/raspberry/PiBulleye_Install.sh)
> On [...]

Just to know if your "Bulleye" Is 32 or 64 and your script *.sh has to be run at the end of all FortiusAnt script ?

Toetta, 2022-02-21:
> For me it was a workaround to copy the node directory from the working one into the RPI with unstable connection. After this both worked.

I just ran your PiBulleye_Install.sh right over my current installation and now it works!! thanks a million!!

decodeais, 2022-02-21:
I have tested it on 32bit. It makes a complete FortiusAnt installationm, except specials like autostart. At the beginning it erases the complete FortiusAnt directory to get a defined starting point. So, if @Toetta  really started my script, he  made a complete install. I think ifm if he had known this, he would not have done this. I am shure The wxPython installaton has been crashed on Buster, but the old is still working.

Toetta, 2022-02-21:
All seemed fine,,,, but

Now it lost the erg mode and does not give variable resitance on the Vortex.

Tried on the apple tv and the mac running swift...

Dont really have a clue what Im doing or what to try next.

The settings confuses me with rhenpower curve adjustments and seem to want to return always to 100 and occasionally 20. I dont know what impact they should have..

Need to sleep on this. Happy that it connects and I was able to ride fir one hour though :)

decodeais, 2022-02-22:
Now it is  interesting, wether  this problem exists on Ant+ too. It seems to be a complete new issue. I am only using grade mode but it had a little bug on BT too. This bug was resolved last month, may be there is an additional bug on erg mode.

planetSammy, 2022-02-22:
> hi @planetSammy sorry to read that there are issues. Starting FortiusAnt in grademode would not be a solution, you also need the grade itself.
>
> Now, FortiusAnt remains in PowerMode/100W (I assume); [...]

I get my workaround back 😁 the device was already paired by bluetoothctl...after i remove my iPhone the workaround works again 👍

decodeais, 2022-02-22:
> All seemed fine,,,, but
>
> Now it lost the erg mode and does not give variable resitance on the Vortex.
>
> Tried on the apple tv and the mac running swift...
>
> Dont really have a clue what Im doing or [...]

I testet ergo mode the first time with Rouvy and it worked perfect after I ireduced the Power given from the trainer programm. I think FortiusAnt works perfect in ergo mode.
Have you tested another trainig software (Rouvy / freeware Golden Cheetahh)?

Toetta, 2022-02-22:
Just installed GoldenCheeteah to give it a try. Took version V3.6. It recognises TacX trainer over BLE and all the sensors. However I get no response at all when trying a training program. Nothing is communicated from FortiusANT to GoldenCheetah that I can see. FortiusANT is operational and registering the inputs... Hmm...

WouterJD, 2022-02-22:
I'm working on a debugging program to breidt what is happening. Give me some days, almost done.

decodeais, 2022-02-22:
I had the typical problems with  Cheetah, it's not very stable. After a reboot video with grade worked and erg mode controlled from Cheetah too. I used RPI4 with Bullseye with installation by my actual script , BT , Golden Cheetah on Windows with BT4.0 stick.

Toetta, 2022-02-23:
Reinstalled with new Bullseye. Still did not work ok. Upgraded to wxpython4.1.1. Now it works over BLE giving resistance feedback =)

decodeais, 2022-02-23:
> Is now every thing ok ?  Whatever happend, it is a little starange because wxpython is for the grafik  user interface and has nothing todo with  communication.

Toetta, 2022-02-23:
I went through all the installation logs.

Fixed all the failures one after one until the Bulleye installerscript
could run all the way without errors. Manually downloading stuff and
getting latest versions etc

One of the major hickups seemed to be the wxpyrhon and getting it sorted to
be installed

Now only hickup left is that i have to restart fortiusant once efter reboot
and rhe autostart to get it communicating. That I can live with :)

decodeais, 2022-02-23:
I think it would be the easiest way to make one  good self-expanding installation shrinked to 2GB. In my opinion this make sense because the RPI installation runs on RPI3 without any changes. It is a little complicate but has to be only one time. Ant then every thing is fixed for this release. There is only the question where to find a free server for it.

WouterJD, 2022-02-23:
> > Is now every thing ok ?  Whatever happend, it is a little starange because wxpython is for the grafik  user interface and has nothing todo with  communication.

Memory overwritten?

WouterJD, 2022-02-23:
Hi guys, perhaps you are interested in following work on issue #366.
If so, please check my question: https://github.com/WouterJD/FortiusANT/issues/366#issuecomment-1049169017

decodeais, 2022-02-23:
> > > Is now every thing ok ?  Whatever happend, it is a little starange because wxpython is for the grafik  user interface and has nothing todo with  communication.
> 
> Memory overwritten?

I don't think so, because the bullseye script uses a precompiled wheel for wxpyton.
When I  heard from the problems I took a fresh SD-card put the OS on it and started the installation update. The script did the rest. Can someone else do a test too, before It tooks my mind. No idea what's going wrong.

Toetta, 2022-02-23:
I did a new compilation of the wheel in order to get it to run.

But disregards my input. I do t really know what Im doing. Just googling and following tips on the internet

Skickat från min iPhone

WouterJD, 2022-02-24:
> I think it would be the easiest way to make one good self-expanding installation shrinked to 2GB. In my opinion this make sense because the RPI installation runs on RPI3 without any changes. It is a [...]

I understand the point.
What self-installers would you like to have:

OS: rpi0/1/2/3/4, windows 10/11, macOS, Linux
and then customized for what Tacx-model?

Professionally spoken, yes.
But how to create, where to store, who maintains and especially: who pays for the work?

decodeais, 2022-02-24:
If you really near to get the bleak/bless solution it is better to focus on this. I new this kind of solution from another project. Aim would have been  in installation ready to parameterize  and start. The image for RPI 2/3/4  are the same. 
I think RPI 0 is similar.
 I searched 1 month to get my new installation working.
After this it was clear to me why nobody had trouble with the grade scaling bug.
But I wish you fast success for the BT improvement . But there are many problems, what kind of development system, how to set up, how to debug. I' am fraid that you are ready when I have understood the Python IDE.

WouterJD, 2022-02-24:
Hi @decodeais when I found FortiusAnt predecessor antifier, I did not know python, github or visual source code and had no clue what a raspberry would be.
So... don't feel intimidated and start off!

decodeais, 2022-02-24:
My first Kontakt with python was with RPI LEDs for your program. 
Do you use a special tool like Py

MeanHat, 2022-02-24:
hi @decodeais  - the RPi0 is different with armv6l architecture which requires a different install. @WouterJD 's shell script [https://github.com/WouterJD/FortiusANT/blob/master/raspberry/3_InstallNodeJs.sh](url) checks this and alters the install to suit the architecture.

decodeais, 2022-02-24:
> hi @decodeais - the RPi0 is different with armv6l architecture which requires a different install. @WouterJD 's shell script [https://github.com/WouterJD/FortiusANT/blob/master/raspberry/3_InstallNodeJs.sh](url) checks this and alters the install to suit the architecture.

I can not test it because I have no RPI 0 and they are crazy expensive at the moment. I think I do not compile, I am only asking the nodemanager `n` to find it.  What happens when you install `npm install -g n` and change the version, for example to latest version with `n latest`  or `n 14.15.3`.

[13 more replies]

closed 2022-08-09 by WouterJD