WouterJD/FortiusANT #373: V6.3 will not boot with a custom fortiusantsetting.JSON file created with the console

jujroy, 2022-02-07:
Hello,  not sure if this problem is unique to me.
Using 2 Windows 10 laptops
T1932
Motor Brake
Never seen this issue with V6.2 or prior versions.
I normally start FortiusAnt by double clicking the Fortius.exe file found in the WindowsExecutable folder.  Program start normally.
I like starting the program with my HRM number (17844) thus I save the Json file.
With the Json file in the WindowsExecutable folder,  the program will NOT boot and I am getting errors as per the picture enclosed.
If you need a copy of the faulty JSON file, I can email seperately given that Github wont let me attach this type of file.
I also tried various json settings such as start without calibration or start without PSA.  Same same.
Thanks  @WouterJD 
[image]

WouterJD, 2022-02-07:
As an alternative, you can start exe with -H 17844

Toattach JSON, add .txt to full filename and then attach

jujroy, 2022-02-07:
Thansk @WouterJD for your prompt replys.  I forgot to mention that as a workaround, I simply delete the file and restart.  It works everytime as long as I do not create a new json file.

As requested find attached the faulty file.

Cheers

[FortiusAntSettings.json.txt](https://github.com/WouterJD/FortiusANT/files/8018841/FortiusAntSettings.json.txt)

WouterJD, 2022-02-07:
I will check next week

(although it should work, I never use it and use the command line)

WouterJD, 2022-02-07:
It may well be introduced in 6.3 (see wiki)

WouterJD, 2022-04-11:
Should be solved with #366 in branch https://github.com/WouterJD/FortiusANT/tree/Implement-BLE-using-bless
Please verify and let me know

WouterJD, 2022-04-12:
Please share the JSON file

jujroy, 2022-04-13:
[FortiusAntSettings.json.docx](https://github.com/WouterJD/FortiusANT/files/8484454/FortiusAntSettings.json.docx)
As requested.  JSON file created today.  I tried to remove PSA and saved the file.  Once the file is created in the windows executable folder, the GUI does not come up.  Version 6.2 is working fine with a custom JSON file.  Tried different parameters (custom HRM, no calibration, etc) , it did not work.  You may be able to recreate this issue on your own system.  Many thanks.  Cheers.  No pressure, I am using v6.2.

WouterJD, 2022-04-14:
Hi @jujroy 

The development branch https://github.com/WouterJD/FortiusANT/tree/Implement-BLE-using-bless
is version 6.3 like the master branch. I will change that

Also, I did not yet create a .exe in that branch, so the problem is solved only in the python-version.

I have tested with your .json and the problem can be reproduced with .exe and not with .py; so I guess it's solved.
I will create .exe for you asap.

WouterJD, 2022-04-14:
.exe published

jujroy, 2022-04-14:
[FortiusAntSettingsjson.docx](https://github.com/WouterJD/FortiusANT/files/8490743/FortiusAntSettingsjson.docx)
Hello Wouter,  I tried your new .exe file this morning.  Everything is booting properly now and GUI shows V6.4.
The resistance provided by the motorbrake is unrealistic/too weak.  Power displays in excess of 1000W and speed in excess of 60km/h.  JSON attached.  Cheers

WouterJD, 2022-04-15:
> The resistance provided by the motorbrake is unrealistic/too weak. Power displays in excess of 1000W and speed in excess of 60km/h. JSON attached

Sorry to hear that. Please raise a separate issue including specifics of your environment.

closed 2022-04-15 by jujroy