WouterJD/FortiusANT #376: Genius T2021 no connect

Lobito1965, 2022-02-09:
Hello Wouter, I'm Wolfgang from Germany! I have a Tacx i-genius T2021 multiplayer and would like to use it for Zwift and the like!
I downloaded FortiusAnt, but the trainer doesn't connect!
I know there are already many explanations here, sorry, but I think I'm too stupid to implement this. Please can you tell me step by step what I have to install to make it work!? I have windows10 and want to use the windows quick start link!?
I have 2 Ant+dongles and the Tacx antenna T2028 is available, do I have to use the T2022 interface?
Black track T2420 is also available.
Thanks for your answer for me who has no brain for computers!

WouterJD, 2022-02-15:
Hi Wolfgang, please do not underestimate yourself. You found github and got the software on your system.

Basically the manual contains instructions what to do, so that's the place to start. Check section 5.14.

For Genius:
- start FortiusAnt from startup folder, using "FortiusAnt (gui, Genius, autostart).bat"; the essential is the -t parameter.
- If you prefer, you may also use the executable version: "`..\WindowsExecutable\FortiusAnt.exe  -a -g -H0 -t Genius`"

- Start FortiusAnt, press Locate HW and Start and wait som e minutes so that the Genius is found.

When that's all working, start Zwift and Zwift should find FortiusAnt 

I hope you manage...

WouterJD, 2022-02-23:
Since there is no communication here, I assume can be closed.
If not happy to reopen

closed 2022-02-23 by WouterJD

Lobito1965, 2022-02-23:
.fr-emoticon.fr-emoticon-img {
                                background-repeat: no-repeat !important;
                                font-size: inherit;
                                height: 1em;
                                width: 1em;
                                min-height: 20px;
                                min-width: 20px;
                                display: inline-block;
                                margin: -0.1em 0.1em 0.1em;
                                line-height: 1;
                                vertical-align: middle;
                            }
                            .fr-emoticon {
                                font-weight: normal;
                                font-family: "Apple Color Emoji","Segoe UI Emoji","NotoColorEmoji","Segoe UI Symbol","Android Emoji","EmojiSymbols";
                                display: inline;
                                line-height: 0;
                            }
                        Hello Wouter ,thank you for your answer and for the Programm FortiusAnt.My brother help me for instal the rest of the programm,he is an IT person.Now the programm runs and all is perfekt ,thank you and have a good time Wolfgang
-----urspr&uuml;ngliche Nachricht-----