WouterJD/FortiusANT #378: vortex T 1961

goupet, 2022-02-15:
Hallo

Bestaat er een uitleg hoe ik een vortex T 1961 kan koppelen met de fortiusant. De head unit heeft een foutmelding en is niet meer bruikbaar. Kan u mij helpen.

Verder heb ik ook nog een Bushido T1981 staan van 2008. de head unit werkt. Doch als ik fiets koppelt hij de bushido tacx (lampje brandt), maar gebeurt er niets op het scherm.

Ik heb de fortiusant gedownload.

is er ergens een duidelijk stappenplan.

mvg

Peter

WouterJD, 2022-02-17:
@goupet Welkom in de FortiusANT community
- - -
Ik ben altijd nieuwsgierig met wie ik communiceer, waar FortiusANT wordt gebruikt en in welke configuratie.
Welke bundle heb je gekocht, en welke rem en head-unit gebruik je?
Vertel meer over jezelf, eventueel onder issue #14.
- - -

> is er ergens een duidelijk stappenplan.

De beste start is het manual, zie 5,15 voor Bushido. Voor de Vortex en Genius zijn de headunits optioneel omdat alleen de knoppen van de headunit worden gebruikt; voor de Bushido is de headunit noodzakelijk omdat de trainer via de headunit wordt aangestuurd.

goupet, 2022-02-17:
Zelf 1 fietstrainer gekocht in 2008. Wil nu kijken om toch iets uitdagender te gaan met online training. Via wat doorzoeken ben ik plots bij jou terecht gekomen. Ik probeer veel te fietsen doch wil ik wat uitdagender fietsen.
Na wat mails tussen garmin/tacx en mezelf wouden ze mij een nieuwe trainer verkopen. Ze zeiden vlakaf dat er geen ondersteuning was en dat dit ook niet mogelijk was.

goupet, 2022-02-17:
Ik heb voor de bushido het stappenplan 5.15 gevolgd doch blijven de wijzers beneden staan.
Zelf ben ik geen echte computerexpert in bepaalde vlakken en heb ik wat meer ondersteuning nodig.
Zo stel ik mij de vraag of ik dit kan doen met 1 pc of ik er toch 2 nodig heb?

goupet, 2022-02-17:
de 2e fietstrainer is een Vortex van een collega die de Vortex wil koppelen met een zwift of een ander trainingsprogamma. Doch is de headunit gecrasht (:(). en zou ik haar graag willen helpen.

WouterJD, 2022-02-18:
[image]

Je moet wel op **Locate HW** en **Start** drukken!

WouterJD, 2022-02-18:
Ik zie in het zwarte scherm (console) wel een hoop berichten, dat is ongebruikelijk. Even opletten daar!

goupet, 2022-02-19:
Dat zwarte scherm komt automatisch op

Ik heb reeds een ant dongle van Garmin gekocht? Weet je toevallig of dit dezelfde moet zijn of toch eentje van een ander merk mag zijn.
Het zwarte scherm ontstaat automatisch bij de opstart van de Fortius ant.

dank je wel.

closed 2022-02-19 by goupet

WouterJD, 2022-02-19:
Het zwarte scherm is een python dingetje en altijd aanwezig.
In principe zijn alle ANTdongles compatible, er zijn problemen bekend met cyckplus

WouterJD, 2022-02-19:
Ik begrijp dat het nu werkt (issue closed)

goupet, 2022-02-19:
Mijn fout was dus dat ik nog een tweede antdongle moet gebruiken. Ik ga deze dan ook kopen.

Dank je wel voor de support. en ik zal er reclame voor maken...

Peter