WouterJD/FortiusANT #380: connect bushido "No tacx trainer found"

graafro, 2022-02-24:
I hope I have declared it according to the rules of the game. I try to connect a bushido to the PC but get the message no tacx trainer found. On request another ANT dongle bought from the supplier XAND. Unfortunately no success

graafro, 2022-02-24:
communication with the bushido is in the air. However, I do not receive any (speed) values. Garmin sensor with new 2032 battery. Any idea what's causing that?
[image]

WouterJD, 2022-02-25:
The bushido is not yet paired; FortiusAnt is waiting: "can take a minute".

graafro, 2022-02-25:
The tacx is next to the PC and is found by the tacx app. Is there any way I can continue troubleshooting? So that I can find out what is causing the problem. Something like "ping"

graafro, 2022-02-25:
Tried to create the log file, but get the message that writing is not possible. Access denied.

If I only call -t Bushido from the prompt, the communication starts but the values are not retrieved

[image]
[image]

WouterJD, 2022-02-25:
Fortius wird in the current directory which is not suitable ; start from another directory:

cd /temp
/... /FortiusAnt/python code/FortiusAntpy... 

Then logfile is created in temp

graafro, 2022-02-25:
Hi Wouter, 

This is the file that was just logged.

Do you want to see if you can discover a cause why it does not function.  

[FortiusAnt.2022-02-25 13-17-53.log](https://github.com/WouterJD/FortiusANT/files/8141118/FortiusAnt.2022-02-25.13-17-53.log)

graafro, 2022-02-25:
[Uploading FortiusAnt.2022-02-25 13-17-53.log…]()

WouterJD, 2022-02-25:
The bushido pairs with the headunit, is that switched on?
Note that I do not have a Bushido so cannot reproduce.

ANT communication is through channel 5; [see here](https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/antDongle.py#L148) but I do not see much traffic going on:
```
---------- FORTIUSANT.2022-02-25.13-17-53.LOG
13:18:13,139: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:18:23,147: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:18:33,391: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:18:43,399: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:18:47,993: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xb9, info="05 4e 15" (ch=5, msg=0x4e)
13:18:47,993: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xb9, info="05 4e 15" (ch=5, msg=0x4e)
13:18:47,993: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xb9, info="05 4e 15" (ch=5, msg=0x4e)
13:18:47,993: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xb9, info="05 4e 15" (ch=5, msg=0x4e)
13:18:48,624: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xa0, info="05 42 00" (ch=5, msg=0x42)
13:18:48,655: Dongle    send   : synch=0xa4, len= 5, id=0x51 Channel ID           , check=0xa6, info="05 00 00 52 01" (ch=5, nr=0, ID=82, tt=1)
13:18:48,655: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xb3, info="05 51 00" (ch=5, msg=0x51)
13:18:48,687: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xa7, info="05 45 00" (ch=5, msg=0x45)
13:18:48,718: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xa1, info="05 43 00" (ch=5, msg=0x43)
13:18:48,749: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0x82, info="05 60 00" (ch=5, msg=0x60)
13:18:48,780: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xa6, info="05 44 00" (ch=5, msg=0x44)
13:18:48,812: Dongle    drop   : synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xa9, info="05 4b 00" (ch=5, msg=0x4b)
13:18:48,843: Dongle    send   : synch=0xa4, len= 2, id=0x4d Request Message      , check=0xbf, info="05 51" (ch=5, msg=0x51)
13:18:48,843: Dongle    drop   : synch=0xa4, len= 5, id=0x51 Channel ID           , check=0xa6, info="05 00 00 52 01" (ch=5, nr=0, ID=82, tt=1)
13:18:53,634: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:19:03,688: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:19:13,741: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:19:23,791: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:19:33,844: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:19:43,901: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:19:53,964: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:20:03,984: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:20:13,991: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:20:24,010: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:20:34,253: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:20:44,256: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:20:54,504: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:21:04,751: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:21:14,996: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:21:24,997: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
[... cut ...]
13:28:37,988: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:28:47,994: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:28:57,995: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))
13:29:08,256: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xe6, info="05 00 00 00 00 00 00 00 00" (ch=5 p=0(Default data page))

```

graafro, 2022-02-25:
it is possible that the head unit was active. This one was in the room. The display is black but it had new batteries in it. Batteries removed but otherwise no success. Isn't it expected that there will be no data traffic if the bushido is not paired? If you don't know a solution direction, I'm afraid the end is practice.

WouterJD, 2022-02-25:
Hmmm.... Is the headunit dead?

WouterJD, 2022-02-27:
Perhaps other Bushido users have an idea?

@mac2444, @Muleguy-Security, @Marmi86, @TinusVee, @philhaupt, @david-evanochko, @rjon65, @Kiredeid, @Arie-new, @wybrenj, @Ferfra, @slilja, @AdyGB, @plucky00, @Yvank7, @ede911, @MiroslavPlamenovNikolov, @Roguish000, @kbruneel, @StijnVerm, @vaneeDutch, @goupet

graafro, 2022-02-28:
The display of the head unit does indeed show nothing. Is the head unit not superfluous if you operate the bushido via the app on your phone? Or is it possible that the bushido is connected to the head unit and therefore communication via the fortiusANT is not possible?

WouterJD, 2022-03-01:
FortiusAnt communicates with (through) the bushido headunit.

graafro, 2022-03-01:
[image]
when the head unit is needed for communication then i have a problem. The connections of the flat cable are oxidized. I tried to clean it but to no avail.

WouterJD, 2022-03-02:
Confirmed: FortiusAnt communicates with the Bushido headunit.
I do not know the reasons why Vortex/Bushido/Genius are different, but this is how developer has built it.

goupet, 2022-03-05:
I have the same problem with the bushido.
I bicycle for 5 minutes , headunit dispays values but the screen gves nothing. I note that ANT CP becomes no signal. Only the tacx light gives a signal.

WouterJD, 2022-03-05:
Be careful that ant communication should be within 1.5meters. Keep ANT transmitters close to each other. I hope this helps

goupet, 2022-03-09:
Hello

I have done different combinations of putting the ant dongles and the head unit.  He gives the signal that the Tacx Bushido paired: 1005 - configuring head unit. But after 4 minutes is this the same signal.

With the parameters that are written in the manual.

WouterJD, 2022-03-23:
Hi Peter, sorry for late response. Being busy with BLE lately :-(

What is the status of your bushido?

goupet, 2022-03-26:
Hello Wouter

I tried it three times but i still get the same message as above... (i cycled for 10 minutes, but nothing gives a signal). The second trainer that i had to pair works perfectly.

mvg

Peter

WouterJD, 2022-03-28:
Hi Peter, sorry to hear you did not get any further.

If a second trainer works and the first does not (assuming they have the same model), the problem seems trainer#2 related.

If you start FortiusAnt with debugging ON (-d4 for ANT-logging, see manual 4.2 command line) you can see what data is received on the Bushido channel 5, see antDongle.py)

Succes!

goupet, 2022-03-28:
Hello Wouter

I will try that.

The first trainer was a vortex. The second trainer was a bushido.

thnx for the tip.

Peter

WouterJD, 2022-03-28:
> I have done different combinations of putting the ant dongles and the head unit. He gives the signal that the Tacx Bushido paired: 1005 - configuring head unit. But after 4 minutes is this the same signal. Both antdongles are left and right in the PC.

Hmmm... if it says "paired" it means FortiusAnt has done the initial handshake and then waits for further info.
I do not have a bushido, so cannot test for you.

WouterJD, 2022-08-17:
Since there is no communication here, I assume can be closed.
If not happy to reopen

closed 2022-08-17 by WouterJD

Arie-new, 2022-08-17:
Wouter,

I have Bushido brake and broken display for sale € 25,- or less. Are you interested?

Thx.
Regards,
Arie