WouterJD/FortiusANT #381: Control Gearbox settings by keyboard?

ChrisLambert1, 2022-02-27:
Hi, I wonder if it's possible to control the gearbox by keyboard?
Formerly I used a Tacx fortius headunit, that worked like a charm, but upgraded to a Genius with a T2022 headunit.
This headunit is not recognized nor pressing buttons on the unit changes the gearbox settings. So I wonder if it's possible to change the gearbox settings by the PC keyboard?

Regards,
Chris

WouterJD, 2022-02-27:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

The Genius is controlled directly (FortiusAnt <--> Brake) and the headunit is not used.
The computer is not used to control, because it is assumed the keayboard is not close to the bike.

In the current implementation, you have two alternatives:
- use a remote control, see manual 4.1.7 to change gears
- use a raspberry with buttons, which can control the gearbox

WouterJD, 2022-03-23:
Keyboard switching could be implemented... feature to be implemented

ChrisLambert1, 2022-03-24:
Perfect!

krusty82, 2024-01-03:
> Welcome to the FortiusANT community
> 
> I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used. Please tell me what bundle did you buy, and what brake and what head unit do you use? I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14. You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
> 
> The Genius is controlled directly (FortiusAnt <--> Brake) and the headunit is not used. The computer is not used to control, because it is assumed the keayboard is not close to the bike.
> 
> In the current implementation, you have two alternatives:
> 
> * use a remote control, see manual 4.1.7 to change gears (e.g. from O-Synce)
> * use a raspberry with buttons, which can control the gearbox

Hello, I have a Genius and would like to use virtual shifting.
The manual states, that it is not implemented (yet).
Are there plans to implement it?
With a ANT+ remote i managed to change virtual gears. The shown speed in FortiusAnt is changed, but the power/resistance from the Genius is unchanged.

Greetings from Germany,
Thomas

WouterJD, 2024-01-03:
> The manual states, that it is not implemented (yet).

What do you mean with "manual states"?

> ANT+ remote i managed to change virtual gears.

Nice

> The shown speed in FortiusAnt is changed, but the power/resistance from the Genius is unchanged.

As described in the manual, the buttons on the Remote work exactly like the buttons of a tacx head unit; it changes the virtual gears (I understand you see that happening) which in the end means that the required resistance changes and the speed.

I checked the implementation:
https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L550
differs from
https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L569

Which suggests that the virtual gears would only work in PowerMode and not in GradeMode.
PowerMode is when the CTP requires to produce a certain power (Watts) where
GradeMode is when the CTP tells FortiusAnt to simulate a hill (slope with certain degree)

PowerMode is usually used during structured training and GradeMode is used for virtual rides.

As a first step, please explain a bit more what training you do, including what CTP is used. A screen short while testing the Shifting helps.

WouterJD, 2024-01-03:
Reopened to investigate whether Virtual Shifting would not work in GradeMode...
@BikeBeppe64 I believe you tested this with Rouvy...

WouterJD, 2024-01-03:
And @krusty82 
Welcome to the FortiusANT community
Dit you visit [the sponsor page](https://github.com/sponsors/WouterJD) already?
- - -

WouterJD, 2024-01-03:
@pquentin

krusty82, 2024-01-03:
Okay,
maybe I should mention, that I also have a Fortius Unit and with this Virtual shifting works perfectly.
Now I also got the Genius Unit and with this the Virtual Shifting does not work in Grade Mode.
I used Rouvy and Zwift to test it.
It seems like the Genius only uses the Grade to determine the needed resitance and ignoring the VirtualGear

With normal (Virtual)Gear everything is fine:
[image]

But with a high Gear the expected and the reported Power don't match:
[image]

WouterJD, 2024-01-03:
> Fortius Unit and with this Virtual shifting works perfectly

Also in GradeMode?

WouterJD, 2024-01-03:
According to the screen prints, FortiusAnt is not paired with the Genius.
=== There is no communication between FortiusAnt and trainer.

krusty82, 2024-01-03:
> > Fortius Unit and with this Virtual shifting works perfectly
> 
> Also in GradeMode?

Yes. So the problem must be with the implementation of the Genius.
I think I read somewhere, that the Genius is not controlled by resistance, but only by grade

krusty82, 2024-01-03:
> > With normal (Virtual)Gear everything is fine:
> 
> According to the screen prints, FortiusAnt is not paired with the Genius. === There is no communication between FortiusAnt and trainer.

No, it is connected. After starting FortiusAnt the message to Nudge the wheel to start Calibrating appears.
Genius gets calibrated. Then the Pairing message appears again. But the trainer is connected and works with CTP.

WouterJD, 2024-01-03:
> Then the Pairing message appears again.

That's a bug in itself (please raise an issue); which is hard for me to investigate because I have no Genius.

WouterJD, 2024-01-03:
> Yes. So the problem must be with the implementation of the Genius.
> I think I read somewhere, that the Genius is not controlled by resistance, but only by grade

**Fortius implementation:**

The brake requires a TargetResistance.
If a TargetPower is given, it's calculated to a TargetResistance
If a TargetGrade is given, it's calculated to a TargetPower then to a TargetResistance

**Genius implementation:**

The brake requires a TargetPower or a TargetGrade and further conversions required.

The VirtualGearbox modified the TargetPower [550](https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L550) but not the TargetGrade [569](https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L569)

Both Fortius AND Genius use the same TargetPower (modified) and TargetGrade.
So how can the Fortius then work well with the virtual gear IN GRADE MODE and Genius does not?
That's an interestinmg question...

WouterJD, 2024-01-03:
So @krusty82 your assessment is important here:
Please confirm that VirtualGearBox works in GradeMode for Tacx Fortius, please provide some visuals to illustrate.

krusty82, 2024-01-03:
I am not good at programming, but if I understand that correctly:
https://github.com/WouterJD/FortiusANT/blob/5a89179f456e84e48d88c0b5200b650a7ca28b89/pythoncode/usbTrainer.py#L1609

for mode_Grade the Genius is used in mode_slope and only the grade is send, without conversion. So the VirtualGearRatio has no effect.

pquentin, 2024-01-04:
> Hi @pquentin I see you follow this issue, Do you use FortiusAnt - all the way in La Reunion?

Hello! What makes you think I am following this issue? I have never heard about FortiusAnt, sorry, and I don't understand what it does from the README.

WouterJD, 2024-01-04:
> Hello! What makes you think I am following this issue?

When typing @, github provides your name @pquentin as suggestion ... which put me on your track.
The previous post is also marked by github as contributor
And at https://github.com/pquentin?tab=repositories&q=FortiusAnt I saw that you forked the repository.
So that's why...

[image]

Funny that github shows this and you have no clue.
But hey, no harm done :-)

krusty82, 2024-01-04:
So, I finally got Python running on my maschine and could play a bit with the code.
Also I tried to understand the code some more.
I think the difference with the Genius and the other trainers is, that for the other trainers the grade is converted to a resistance which then is send to the trainer. In this conversion the virtual shifting comes in to play. 
But for the Genius the grade is send directly to the trainer without conversion.
So I don't think it's a bug, but a feature of the Genius that it has the slope mode.

Now that I can mess with the code, I can adjust the grade that is send to the Genius, because my problem with it is, that on flat parts I have to use a very high gear and the genius becomes quite loud.

Maybe with some more coding it is possible to implement a virtual shifting for the Genius that works and feels right.
I just played around a bit with power mode, but didn't work well.

pquentin, 2024-01-04:
Actually I remember now! I forked to contribute https://github.com/WouterJD/FortiusANT/pull/433 because I was trying to get urllib3 2.0 adopted as an urllib3 maintainer.

WouterJD, 2024-01-06:
I have created a new branch **#381-Virtual-Gear-in-GradeMode** and a new usbTRainer/py

https://github.com/WouterJD/FortiusANT/blob/%23381-Virtual-Gear-in-GradeMode/pythoncode/usbTrainer.py

Please test and let me know

krusty82, 2024-01-06:
Thanks, will test tomorrow.
In the meantime I changed the code myself.
In usbTrainer.py I changed lines 1608ff to:

`Weight = max(0x0a, int(self.GearboxReduction * self.UserAndBikeWeight))
                        if self.TargetResistance < 0:
                           abspower = 0
                        else: 
                           abspower = self.TargetResistance 
                        info = ant.msgPage220_01_TacxGeniusSetTarget(self.Channel, ant.GNS_Mode_Power,
                                                                     abspower, Weight)           `

Weight to change the flywheel accordingly.
The abspower is needed, because if power is negative in descent, then the brake blocks.
With this change, the genius is handled the same as the other trainers, especially the Vortex. 
The only thing missing is the speeding up of the wheel on steep descents, but I don't miss this.
The cycling feels good.

WouterJD, 2024-01-06:
OK @krusty82 it may work.
But: it works for Genius only
And: instead of modifying the targetResistance, you modify the userAndBikeWeight.

Anyway, it's great that you have made it work.

> Thanks, will test tomorrow.

This would be nice. If that provides the desired result it may imply that it works for all trainers.

WouterJD, 2024-01-06:
Please note the concerns (in the code) regarding the effect when Grade=0

krusty82, 2024-01-06:
> Please note the concerns (in the code) regarding the effect when Grade=0

Yeah, thats one concern, because I use the virtual gears mostly on flat parts.
Also I am curious what happens in the descents? 
I guess the gears get inverted.

krusty82, 2024-01-06:
> OK @krusty82 it may work. But: it works for Genius only And: instead of modifying the targetResistance, you modify the userAndBikeWeight.

No, I used the same method, that is already used for the other trainers. 
targetResistance gets modified by GearboxReduction in the grade2power formula.
But in the original code the Genius didn't use this formula but got send the grade directly.
Also for the Fortius the flywhel weight gets adjusted, so I did this also for the Genius. I think the other trainers can't adjust the flywheel anyways.
> 
> Anyway, it's great that you have made it work.
> 
> > Thanks, will test tomorrow.
> 
> This would be nice. If that provides the desired result it may imply that it works for all trainers.

WouterJD, 2024-01-06:
On the Tacx Fortius, when Grade get's negative, the brake becomes an engine and produces a negative resistance.
I **think** that the Genius cannot do that, but as said I don't have a Genius so cannot test that.

Gears will not invert, but in a low gear with no resistance, your cadence wil go high. (cycling with no resistance).

> The abspower is needed, because if power is negative in descent, then the brake blocks.

Apparently the provided Terget must be a positive value
Hence; Genius does not support decents.

krusty82, 2024-01-06:
> Apparently the provided Terget must be a positive value Hence; Genius does not support decents.

Genius does support it, but i guess only in slope_mode, in which it is not possible (or ot that easy) to use virtual gears

WouterJD, 2024-01-06:
Sorry, I use Vortex and Genius as synonyms.

I will study all the previous next week.
Please provide all data you have gathered and I will merge....

Thanks for cooperation.

krusty82, 2024-01-10:
So i uploaded my changes to a fork here: [https://github.com/krusty82/FortiusANT](url)
In addidtion to the virtual shifting for the Genius, I also encountered a bug with the ANT remote.
Every press of a button was executed twice. 
I solved this by reseting the TacxTrainer.Buttons = 0 , but I don't know if this interferes with anything else.

Another thing I noticed is, that the reported Power from the Genius is much slower in change compared to the Fortius. 
It also seems to be more averaged. But I don't know if it is in the code of FortiusANT or if it is just the Genius.

krusty82, 2024-01-10:
> Another thing I noticed is, that the reported Power from the Genius is much slower in change compared to the Fortius. It also seems to be more averaged. But I don't know if it is in the code of FortiusANT or if it is just the Genius.

Never mind, it seemed to be a connection issue. I moved the ANT receiver closer and it worked flawlessly.

WouterJD, 2024-01-19:
> So i uploaded my changes to a fork here: [https://github.com/krusty82/FortiusANT](url)

Hi @krusty82 just got time to study your work right now.
Great solution, I have adopted in usbTrainer.py. Just a few editorial changes.

> I solved this by reseting the TacxTrainer.Buttons = 0 , but I don't know if this interferes with anything else.

I've adopted this too

My ANT-remote has three buttons, hence
```
                    if   ctrl_CommandNr == ctrl.MenuUp:     TacxTrainer.Buttons = usbTrainer.UpButton
                    elif ctrl_CommandNr == ctrl.MenuDown:   TacxTrainer.Buttons = usbTrainer.DownButton
                    elif ctrl_CommandNr == ctrl.MenuSelect: TacxTrainer.Buttons = usbTrainer.OKButton
```
apparently yours has four buttons:
```
                    if   ctrl_CommandNr == ctrl.MenuUp:     TacxTrainer.Buttons = usbTrainer.UpButton
                    elif ctrl_CommandNr == ctrl.MenuDown:   TacxTrainer.Buttons = usbTrainer.DownButton
                    elif ctrl_CommandNr == ctrl.MenuSelect: TacxTrainer.Buttons = usbTrainer.CancelButton
                    elif ctrl_CommandNr == ctrl.Start:      TacxTrainer.Buttons = usbTrainer.OKButton
```

If I would adopt this code, the function of the select button (middle one) would change...

krusty82, 2024-01-19:
> apparently yours has four buttons:
>
> ```
>                     if   ctrl_CommandNr == ctrl.MenuUp:     TacxTrainer.Buttons = usbTrainer.UpButton
>                     elif ctrl_CommandNr == [...]

The control has 3 Buttons, but you can long press everyone for a different function.
I changed the code, so the middle button will change the front gear ring for short press, and reset the gearing with long press.

WouterJD, 2024-01-19:
Oops, forgot one line...

you changed
                        if DataPageNumber & 0x7f in (0,1,2,3,4,5,6,7,89,95):
into
                        if DataPageNumber & 0x7f in (0,1,2,3,4,5,6,7,**64**,89,95):

please explain :-)

krusty82, 2024-01-19:
> Oops, forgot one line...
> 
> you changed ` if DataPageNumber & 0x7f in (0,1,2,3,4,5,6,7,89,95):` into ` if DataPageNumber & 0x7f in (0,1,2,3,4,5,6,7,64,89,95):`
> 
> please explain :-)

I have an ANT+ HRM from Ciclosport. But I could not connect it through Rouvy ant FortiusANT. But in Zwift it was working.
By using ExplorANT I could see that the HRM was connecting and sending data. But because it was on page 64, the data was ignored. So by adding 64 to the pages Fortiusant is listening to, I could get the HRM into FortiusANT.
But to get the data to Rouvy was a different problem.
Rouvy can't read the HRM out of the FE-C datastream for some reason. And Fortiusant doesn't establish the HRM channel for an ANT HRM, because you should be able to connect the HRM directly to the CTP.
To get a workaround I established the HRM outgoing channel also for ANT HRM.

But instead of this workaround it would be cleaner to get the HRM signal in the FE-C channel.

> 
> I have a special situation;
> 
> * If I start FortiusAnt and then wear my HRM, FortiusAnt does not find it
> * If I wear the HRM and then start FortiusAnt, it is found
> 
> I get the impression, you know more on programming just a bit; do you see where the problem is? I would be most obliged when you find the bug :-)

I also noticed that. And I don't know much on programming. It's mostly just trial and error until it's working ;)
Maybe FortiusANT is just scanning for HRM for some time at startup?

WouterJD, 2024-01-19:
CTP's like Rouvy do not use the HRM info in the data from FE-C, because ANT/BLE devices are expected.

The old tacx trainers connected to a uncoded HRM and provided the HRMdata through USB to the PC where the proprietary tacx software was running. 
From this historical background, the HRM data has been added to the FE-C spec, but it's not used by any software as far as I know. 

I connect HRM to display the heart rate on the console. Implementing an HRM broadcast of the received data requires a second device, like the SCS. This will be quite some work. 

Buying a "standard" HRM is out-of-pocket money, but "cheaper" than all the work to implement

WouterJD, 2024-01-19:
> Maybe FortiusANT is just scanning for HRM for some time at startup?

I think so and will do some testing

krusty82, 2024-01-20:
> CTP's like Rouvy do not use the HRM info in the data from FE-C, because ANT/BLE devices are expected.
>
> The old tacx trainers connected to a uncoded HRM and provided the HRMdata through USB to the PC [...]

I made a branch here: [https://github.com/krusty82/FortiusANT/tree/Testing](url)
I changed a bit of code and now the heartrate (from my ANT HRM) is broadcast over FE-C and picked up by Rouvy.

WouterJD, 2024-01-20:
Nice, I will check asap

krusty82, 2024-01-21:
> > Maybe FortiusANT is just scanning for HRM for some time at startup?
> 
> I think so and will do some testing

In my fork I set the SearchTimeout to infinity. Maybe this helps.

https://github.com/WouterJD/FortiusANT/commit/13e7ac4ac39e443b4d2698b4cd1a6e43e3605e7a

WouterJD, 2024-01-21:
I recall there is some sort of restart and will look into it as well. I just realized recently the sequence issue re HRM. 

Thanks for joining the quality improvement activities, it encourages me to dive into it as wellagain.

WouterJD, 2024-01-23:
I have adopted the updates

antDongle.py:
```
# 2024-01-23    #381 Weight should be positive
#               #381 HRM is searched for infinitely
#               #381 HRM is transmitted through FE-C

```
FortiusAntBody.py:
```
# 2024-01-19    #381/1  ANT/Remote buttons are processed twice
#               #381/2  ANT/Remote has four buttons, but 3 are implemented
#               #381/3  Additional datapage implemented for HRM
#               #381/4  HRM set to zero when no signal for 5 seconds
#               #381/5  HRM transmitted to FE-C

```
usbTrainer.py:
```
# 2024-01-19    In GradeMode virtual gearbox does not work (#381) for antTrainers,
#               like Genius and Vortex.

```
Task to be done is the Remote, I'll do that asap.

### Concluding

Regarding the HRM, the updates you suggested are now implemented.
Also, if no HeartRate is received for 5 seconds, it's reset to zero (so that an old value does not remain forever).
I have tested (alu-foil around the device is an easy switch on/off) and it now works as one would expect.

PS "And I don't know much on programming. It's mostly just trial and error until it's working ;)" seems a little humble; the modifications in the antDongle module show experience. Thanks. Good work!

**Regression-test from your side is highly appreciated!**

krusty82, 2024-01-24:
> I have adopted the updates
> 
> antDongle.py:
> 
> ```
> # 2024-01-23    #381 Weight should be positive
small correction: weight should be positive and below 255kg

> PS "And I don't know much on programming. It's mostly just trial and error until it's working ;)" seems a little humble; the modifications in the antDongle module show experience. Thanks. Good work!

okay, maybe I do know something about programming, but I don't know python and its syntax too well.

> 
> **Regression-test from your side is highly appreciated!**

Will do some testing later. (Additional motivation for training :) ) 
In the meantime I did some programming on the GUI-side (see pull request)
> 
> PS. Are you also on Strava, publishing your training?

Yes, my profile is: https://www.strava.com/athletes/106755454

WouterJD, 2024-02-17:
Hay @krusty82 apart form issue #454 which is split off from this usse; can this one be closed?
I monitor the open issues at irregular basis, so like to keep the list short.

krusty82, 2024-02-18:
> Hay @krusty82 apart form issue #454 which is split off from this usse; can this one be closed? I monitor the open issues at irregular basis, so like to keep the list short.

yes, it can be closed

closed 2024-02-18 by WouterJD