WouterJD/FortiusANT #385: Trainer is a bit dancer (Raspberry Pi4 buster)

BikeBeppe64, 2022-03-04:
Hi any of you have a situation like mine using Bluetooth?

https://user-images.githubusercontent.com/64853498/156856100-3df8032a-0238-4c61-af0a-901f6e1c51ca.mp4

decodeais, 2022-03-12:
I had this effect on my RP too. The sensors wete flashing, connection was passibke, but controle of trainet wad not possible. My new installation script was the solution. 
[https://github.com/decodeais/FortiusANT/blob/master/raspberry/PiBulleye_Install.sh](https://github.com/decodeais/FortiusANT/blob/master/raspberry/PiBulleye_Install.sh) .

decodeais, 2022-03-12:
It is the same problen like issue  #372.  The install script fixes the node installation.

BikeBeppe64, 2022-03-12:
> I had this effect on my RP too. The sensors were flashing, connection was possible, but control of trainet was not possible. My new installation script solved the problem.
> [https://github.com/decodeais/FortiusANT/blob/master/raspberry/PiBulleye_Install.sh](https://github.com/decodeais/FortiusANT/blob/master/raspberry/PiBulleye_Install.sh) .  

I'll try your solution. Thanks in advance

decodeais, 2022-03-15:
> > I had this effect on my RP too. The sensors were flashing, connection was possible, but control of trainet was not possible. My new installation script solved the problem.
> > https://github.com/decodeais/FortiusANT/blob/master/raspberry/PiBulleye_Install.sh .
> 
> I'll try your solution. Thanks in advance

Please  tell me wether it worked for you or what problem appeared.

WouterJD, 2022-03-23:
@BikeBeppe64 nice script; all-in-one :-)
What is the major change, so that it can be added to the current scripts.

I did not create a big script to keep the scripts simple; but it requires to start them one-after-the-other.
In short future we will have the option to use NodeJs and/or bless, which would require parameterization...

decodeais, 2022-03-24:
In this case it was the problem from #372 package-lock.json.
The second is to erase the old installation complete at the beginning (a relict of the crazy package-lock effects). For OS Bullseye there is needed  another wxpython version. I found it at the Ubuntu distribution ( already compiled). I tried to reduce the the amount of questions I'm am asked by the script with default yes, for example.
But this script is a product of try and error, until it worked from a fresh OS install.
But your new bleak/bless version makes the node part obsolete, I think.

BikeBeppe64, 2022-03-25:
> Please  tell me wether it worked for you or what problem appeared.

@decodeais installed your script and at the moment works fine

< VID20220325104001.mp4 -->

BikeBeppe64, 2022-03-25:
> Please  tell me wether it worked for you or what problem appeared

@decodeais installed your script and everything works fine (RGT = CPT)

[image]

WouterJD, 2022-04-11:
Please recheck with bless version, then the node.js issues should be gone.

WouterJD, 2022-08-17:
Hi @BikeBeppe64 ble/bless is now released and the master branch.
Is this issue still happening?

closed 2022-08-17 by WouterJD