WouterJD/FortiusANT #386: 64bit Bullseye: GUI not displaying with Bless/bleak version of FortiusANT

Toetta, 2022-03-29:
Hi!

Im not getting the gui to display with the -g parameter.

It likely that I did something wrong due to quite extensive install procedure and I'm all new to this and don't know anything about programming.

Im not receiving any error messages.

I installed the latest versions of everything and compiled a 64bit wxpython pypi and a new wheel along with all other stuff that I found during the way it complained about.

Could someone assist what to do next when I don't see any error messages?

//
I did not install using the all steps of the steps included in the raspberry directory and did not install "node" stuff. Just ran the Fortius straight off except granting accesses. 
//

Cheers

Malkus

WouterJD, 2022-03-29:
Hi malkus.

The GUI is disabled for raspberry, because wxpython is not available in standard environment without compilation etc.

Enabling is not difficult; check UseGui:
https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/constants.py#L25

decodeais, 2022-03-29:
Here is a location for a compiled wxpython wheel for python 3.8:
[//extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-20.04/wxPython-4.1.1-cp38-cp38-linux_x86_64.whl ](https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-20.04/wxPython-4.1.1-cp38-cp38-linux_x86_64.whl)
Is it not working anymore?

Toetta, 2022-03-29:
In my constants.py file its already set to true. Do I need rebuild FortiusANT.sh file in the raspberry folder that I use to start for it to use the info in the constants.py file in python folder? Im not sure how the files are linked / connected...

Toetta, 2022-04-09:
HI @WouterJD !

Tried your latest build on github. I now get the gui! 

Having new issues though. Power output is about 4x normal to zwift. Connection is now stable. during restarts I get error messeges often when trying to autostart FortiusAnt that the bluetooth name is already in use and BLE can not be startes. 

(I did run through the installation steps to double check that missing) Dont know if it was a good idea or not to install the nodejs stuff as well but I did..

WouterJD, 2022-04-09:
Node.js does not conflict

WouterJD, 2022-04-09:
> during restarts

Why do you restart?
I can start / stop as many times as I like.

Toetta, 2022-04-09:
If it autostarts during bootup it never works.

I have manually close the app and start it again.  Then ble ant and
everything connects.

I will provide better feedback / screendumps / logs

Cheers

WouterJD, 2022-04-09:
As if FortiusAnt starts too early...
Try to delay startup, to check whether first start works when not done too fast

Toetta, 2022-04-09:
These settings wont store after saving / restarting. Always reset. Target powet value restes to 20. I reenter 100 after every restart.

[image]

Toetta, 2022-04-09:
Heartrate values are recieved by Fortiusant through Ant+ but not transmitted over BLE.

[image]

[image]

Toetta, 2022-04-09:
Heart rate with direct connection over bluetooth works well as a work around not passing theought FortiusAnd Bless

[image]

Toetta, 2022-04-09:
Power values (W) are now back to normal. Nothing changed, several restarts only

WouterJD, 2022-04-09:
> These settings wont store after saving / restarting. Always reset. Target powet value restes to 20. I reenter 100 after every restart

See #373 to be investigated and solved.

WouterJD, 2022-04-09:
> Heartrate values are recieved by Fortiusant through Ant+ but not transmitted over BLE.

Although it should be...

WouterJD, 2022-04-09:
> The Vortex controll display powers down after some minutes.

https://github.com/WouterJD/FortiusANT/blob/Implement-BLE-using-bless/pythoncode/usbTrainer.py#L1187
I see the code contains a mechanism to avoid this

Toetta, 2022-04-10:
Thanks for all your responses @WouterJD !

WouterJD, 2022-04-11:
Please test with current version; JSON should be improved now as well.

https://github.com/WouterJD/FortiusANT/tree/Implement-BLE-using-bless

Toetta, 2022-04-16:
GUI works now! Many thanks

closed 2022-04-16 by WouterJD