WouterJD/FortiusANT #389: FortiusANT Installed and Working With ANT.  Instructions for BLE?

Dmlrmed1, 2022-08-09:
I apologize if this is already answered but I can't seem to find specific instructions for BLE.  When I plug my BLE into the USB drive (like I would with my ANT) and launch FortiusANT the BLE device is not recognized.  Seems to work ok for my ANT.  Is there something else I need to install?  I plan to use Zwift on iPad and need BLE.

Thanks

WouterJD, 2022-08-09:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
You could also consider to visit [the sponsor page](https://github.com/sponsors/WouterJD)
- - -

With the currently published FortiusAnt software (master branch), node.js must be installed as is described in the manual.
The version that is to be published (Implement-ble-using-bless branch) is more likely to discover your BLE-dongle, note the version of the BLE-dongle to be used (also described in the manual).

Dmlrmed1, 2022-08-10:
Thanks for the quick reply.  I see the instructions on how to install node.js and zadig,  Thanks.  I plan to bring my Tacx I-Magic back to life as a second trainer.  Still great riding outside in Northeast Ohio right now but will be riding indoors in a few months.  I am using a Windows 10 computer.

WouterJD, 2022-08-10:
Hi @Dmlrmed1 yesterday I promoted the latest software to thge master branch. Installing that version allows to use BLE without instyalling nodejs. Just requires a modern BLE dongle under windows.

Dmlrmed1, 2022-08-12:
I am trying to use an Insignia Bluetooth 4.0 USB dongle.  Is this compatible?  It looks like I need to get a 5.0?

WouterJD, 2022-08-12:
Yes, note that on Windows a BLE-5 dongle must be present BUT the native windows driver is OK (no zadig required)
When BLE-4 is available, Node.js must be used

WouterJD, 2022-08-17:
Hi @Dmlrmed1 anything I can assist with, or can issue be closed?

Dmlrmed1, 2022-08-17:
Thanks.  Out of town but will try with BLE-5 when I get back.

WouterJD, 2022-08-17:
See you soon

Dmlrmed1, 2022-09-03:
So I'm back at it again trying to get bluetooth to work.  I decided to use my Insignia Bluetooth 4.0 USB and installed Node JS.  When I launch FortiusANT.exe the trainer is detected but not the Bluetooth.  I tried to check the box in settings about using Bluetooth with Node JS but the restart just hangs.  What do I need to do and in what order.  For example, just because I installed Node JS do I need to start it somehow.  Not a techy.  Thanks.

WouterJD, 2022-09-20:
Hi @Dmlrmed1 

I just tried to modify settings to activate the BLE interface using node.js and that works (on my system).
Why your system hangs I cannot tell right away.

It produces a json file containing, amongst the other settings.
```
{
    "bluetooth": {
        "nodejs": true
    },
}
```
-----
I usually use the scripts (the .bat files in startup) to start FortiusAnt, because that's the easy way to start with different sets of parameters. You could try that as follows:

You would use **FortiusAnt (exe).bat** (instead of double-clicking the executable itself from explorer).
That would start FortiusAnt with the default set of parameters.

You can modify that file, using notepad.
The file contains
`..\WindowsExecutable\FortiusAnt.exe`
which can be changed into
`..\WindowsExecutable\FortiusAnt.exe -a -g -b`
which means that it starts automatically, uses the GUI and communicates through bluetooth.

As soon as you are succesful with this, you can modify the parameters at will.

WouterJD, 2022-10-11:
Wel @Dmlrmed1 at first follow the manual and then pose a question which is a bit more specific 😀

WouterJD, 2022-10-11:
Hi @Dmlrmed1 replacing is not described specifically, good idea to add that to the manual. I will provide instructions later

Dmlrmed1, 2022-10-11:
Hi again.  I decided to purchase a Bluetooth 5.0 dongle.  When I launch Fortiusant with the dongle inserted I do see both the tacx light and the Bluetooth light on the GUI.  However, when I launch Zwift on my iPhone and search for the trainer I do not see it showing up.  Is there something else I need to do?

WouterJD, 2022-10-11:
How to upgrade? Manual page 22 describes how to download and install the FortiusAnt code. An upgrade can be done in a similar fashion, overwriting the existing code.
Of course, also git can be used. I suppose following the previous instructions will be easier.

When you start FortiusAnt, you have to press [Locate HW] to find the BlueTooth dongle and then [Start] to actually start the "bridge" between your Tacx and BLE. When that is operational, FortiusAnt should be found on your iPhone. See page 28 section 3.8.5.

Sorry that the sequence in the manual is not 100% logical. Perhaps it's caused by the multitude of trainers and interfaces that are supported.

If you have more questions, please let me know.

WouterJD, 2022-10-11:
Git instructions

Installation using git can be done with `git clone https://github.com/WouterJD/FortiusANT.git`
To upgrade an installed version, the command `git pull origin master` can be used.

For more information, check https://git-scm.com/book/nl/v2

Dmlrmed1, 2022-10-16:
Hello again.  I’ve followed your instructions.  Ran Fortiusant.exe, locate HW, Start.  GUI says Bluetooth interface open.  Both the Tacx and BLE ctp lights are blinking.  However when I run Zwift on iPhone the trainer does not show up.

WouterJD, 2022-10-17:
Hi @Dmlrmed1 it's a stubborn environment you have.

_PS. do you use -b (nodejs) or -bb (bless)? As you will see, I use the latter, if you get no error-messages there should be no differrences._

Let's see.

You should have console + UI interface as below, always mind messages in the console (black window).
[image]

Now you can use Zwift to detect a trainer, but for BLE-debugging also the LightBlue app can be used on Android or iPhone.

The app comes up with the following window:
[image]

And when opening the LT-ENTERTAIN (name of Windows PC where FortiusAnt is running):
[image]
which shows the **Fitness Machine Feature** of FortiusAnt, indicating it's a trainer your could connect to.

This is the same process Zwift would do, then presenting the trainer as available to use.
Please let me know what your systems tell you.

Dmlrmed1, 2022-10-22:
Again, thanks for trying to help.  Still having issues.  After installing driver for TP-Link Bluetooth and starting up with -bb I get the following:
[image]
Then when I START I see the following:

[image]

Very frustrating.  Understand if you can't help.  Appreciate all that you've done up to this point.

WouterJD, 2022-10-22:
Software seems ok

BUT what type is your BLE-dongle, it must be ble-5!

Dmlrmed1, 2022-10-22:
Yes it’s 5

WouterJD, 2022-10-22:
Hmmmm. The error returned by the bless library - until now - was caused by a BLE4 dongle. I remember that there are ble5 dongles around that are not.

I can post a question for the bless developer.

You could try another make of ble5 dongle
OR try - b and see to get node.js working, which does not require the ble5 features

WouterJD, 2022-10-22:
Please provide all details of your ble5 dongle.

Dmlrmed1, 2022-10-23:
I rolled back the driver, plugged in both an ANT and Bluetooth USB dongles and ran software.  Here's what I get:

[image]

It appears to be working but I don't see the computer in either Zwift or Light Blue.  I feel I am very close.

WouterJD, 2022-10-23:
Back to this: https://github.com/WouterJD/FortiusANT/issues/389#issuecomment-1280593341

Dmlrmed1, 2022-10-29:
I’m happy to report that I can connect to Zwift by updating the Bluetooth driver.  Now my next issue is that my watts are way too high?  I am showing 500+ watts when it should be closer to 150 watts,  I did a run off but that didn’t fix it.  Even Zwift is giving me a message that my equipment isn’t calibrated correctly.

WouterJD, 2022-10-29:
Hey, this is good and again a step further.
What did you change since:

> It appears to be working but I don't see the computer in either Zwift or Light Blue. I feel I am very close.

I'll get back to you shortly

Dmlrmed1, 2022-10-29:
By updating the driver I was able to see and connect to Zwift

WouterJD, 2022-10-30:
> Now my next issue is that my watts are way too high? I am showing 500+ watts when it should be closer to 150 watts

As I understand from the picture from last week, you have a T1932 headunit, hence a USB-connected Tacx trainer.

The trainer returns the actual resistance and using formula's (resistance and wheelspeed) the power is calculated.
The approach can be found in the manual and the exact formula's in the code: (https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py, functions CurrentResistance2Power)

It may be that the calculated power does not match with a real power (measured witha power meter) and then the -p parameter can be used to adjust. A range of 10% is expected; defining -p90 or -p110 to adjust for that.

The difference you report is quite high; please provide more info to be able to thinbk about the cause.
Please study the manual, see sections 2.5 and 6.

WouterJD, 2023-03-07:
Since there is no communication here, I assume can be closed.
If not happy to reopen

closed 2023-03-07 by WouterJD

jr111, 2023-04-07:
> By updating the driver I was able to see and connect to Zwift

Hello, I have same chipset, wondering what driver worked for you. I have tried various drivers, at one point LightBlue saw devices and changing values for steering, with another Zwift would occasionaly show a power meter with No Connection. Mostly want this for steering, works great for years with Ant. Wouter I commend you, my rusty Fortius still gets many km not possible without your work.